espressif/arduino-esp32 #1345: More problem, giving up on ESP32

laercionit, 2018-04-22:
If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

Decoding stack results
0x4008ebdc: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 139
0x4008edb7: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 148
0x4008209f: lock_init_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 81
0x400820c0: lock_acquire_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 132
0x40082205: _lock_acquire_recursive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 169
0x40110b17: _vfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 860
0x4010b43d: fiprintf at ../../../.././newlib/libc/stdio/fiprintf.c line 50
0x40104019: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 59
0x4013d28e: sys_timeout at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/timers.c line 353
0x4013d3ea: nd6_timer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/timers.c line 244
0x4013d4a1: sys_timeouts_mbox_fetch at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/timers.c line 575
0x40135ae0: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 102

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 WOOR-ROOM
Core Installation/update date:			(stickbreaker)
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Debug Messages:
[E][esp32-hal-i2c.c:945] i2cProcQueue(): install interrupt handler Failed=261

[E][WString.cpp:185] changeBuffer(): realloc failed! Buffer unchanged
abort() was called at PC 0x4008209f on core 0

Backtrace: 0x4008ebdc:0x3ffdc620 0x4008edb7:0x3ffdc640 0x4008209f:0x3ffdc660 0x400820c0:0x3ffdc680 0x40082205:0x3ffdc6b0 0x40110b17:0x3ffdc6d0 0x4010b43d:0x3ffdc990 0x40104019:0x3ffdc9e0 0x4013d28e:0x3ffdca10 0x4013d3ea:0x3ffdca30 0x4013d4a1:0x3ffdca50 0x40135ae0:0x3ffdca70

Rebooting...

laercionit, 2018-04-23:
PC: 0x4008dd89: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 218
EXCVADDR: 0x00000004

Decoding stack results
0x4008dd89: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 218
0x4008d25c: xTaskRemoveFromUnorderedEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3205
0x4008b3b2: xEventGroupSetBits at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./event_groups.c line 626
0x4008b42d: vEventGroupSetBitsCallback at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./event_groups.c line 696
0x4008d86a: prvProcessReceivedCommands at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 716
0x4008d943: prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 546

beegee-tokyo, 2018-04-23:
`[E][WString.cpp:185] changeBuffer(): realloc failed! Buffer unchanged`
Just a guess, but I had similar problem before. It was because I run out of heap memory. Do you have big buffers or other large areas of memory that you reserve globally? Try to reduce your memory consumption.

smacyas, 2018-04-23:
I have the same problem.
Try to change in WString.cpp in function unsigned char String::changeBuffer(unsigned int maxStrLen) {...
size_t newSize = ((maxStrLen + 16) & (~0xf)) - 1;

to
 size_t newSize = ((maxStrLen + 32) & (~0xf)) - 1;
or
size_t newSize = ((maxStrLen + 64) & (~0xf)) - 1;

laercionit, 2018-04-23:
Hello @beegee-tokyo,

My project uses large memory structures for some Structs, I use 16% of global memory.

Sketch uses 1009946 bytes (77%) of program storage space. Maximum is 1310720 bytes.
Global variables use 47256 bytes (16%) of dynamic memory, leaving 247656 bytes for local variables. Maximum is 294912 bytes.

I intended to use ESP32 to launch an automation product on the market, but it has presented many problems and intermittency, I am very afraid to use ESP32 as a product development platform.

beegee-tokyo, 2018-04-23:
Well, Arduino framework (imho) is no good for production. You might want to either learn how to use esp-idf and FreeRTOS directly or look into [Moongoose](https://mongoose-os.com/).

tablatronix, 2018-04-23:
You do realize that there has not been a single release of esp arduinio yet? Meaning it is pre alpha, meaning very unstable..

everslick, 2018-05-03:
Seems as the original issue was about memory running out and the string buffer could not grow any further.

closed 2018-05-03 by everslick