espressif/arduino-esp32 #1346: MAC Wi-Fi == MAC ETH (?)

smacyas, 2018-04-23:
How can i do the same mac in wifi and ETH?
I have an auto-connection from wifi to eth and vice versa, and I would like the ip address remains the same after switch, and this requires the same mac address.
Can i do this in Arduino-IDE? I know, that in IDF there is such opportunity.

lbernstone, 2018-04-23:
I think that is a very niche need, and as a systems admin that often has to track down spanning tree loops my answer would be an emphatic no.  The idf functions will work in arduino, and provide the flexibility you need.

copercini, 2018-07-08:
is this a good idea? no (as @lbernstone explained)

is this possible? yes, using something like `esp_wifi_set_mac(WIFI_IF_AP, eth_mac);`

closed 2018-07-08 by copercini