espressif/arduino-esp32 #1347: (Possibly NVS related) crash on boot

jamesbaber, 2018-04-23:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			10/apr/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
The chip was running my project code which makes use of the Preferences library and it seemed to work fine but after I came back to it at a later date, it didn't work... After checking the serial console I found that it is now reboot looping with the output shown below. I tried replacing the application code with the "Blink" and "BareMinimum" examples but got the same result. I tried esptool erase_flash which gives no output so I can't tell if it did anything, but it certainly didn't fix it.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  pinMode(22, OUTPUT);
}

void loop() {
  digitalWrite(22, HIGH);
  delay(1000);
  digitalWrite(22, LOW);
  delay(1000);
}
```

### Debug Messages:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
assertion "entry < ENTRY_COUNT" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.hpp", line 188, function: uint32_t nvs::Page::getEntryAddress(size_t) const
abort() was called at PC 0x400d2a6f on core 0

Backtrace: 0x4008755c:0x3ffc6510 0x40087737:0x3ffc6530 0x400d2a6f:0x3ffc6550 0x400d5197:0x3ffc6580 0x400d5483:0x3ffc65a0 0x400d5fa5:0x3ffc65f0 0x400d4d92:0x3ffc6650 0x400d4c16:0x3ffc66a0 0x400d4caf:0x3ffc66c0 0x400d4cfa:0x3ffc66e0 0x400d08e8:0x3ffc6700 0x400e42db:0x3ffc6720 0x400d394a:0x3ffc6750

Rebooting...
```

lbernstone, 2018-04-24:
You don't get any errors from erase_flash or uploading firmware?  It sounds like your spi_flash may be damaged. The initialization code will open nvs to try to get your wifi config.

jamesbaber, 2018-04-24:
@lbernstone That could explain it, but what would cause it to be damaged? Hardware issue or corrupt data being pushed to it?
erase_flash completes with no output whatsoever regardless of if the ESP32 is even plugged in :confused: so I could be doing something wrong there!
While uploading code from the arduino IDE, the console output is fairly benign.
```
esptool.py v2.0-beta3
Connecting....
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash params set to 0x022f
Compressed 11120 bytes to 7191...

Writing at 0x00001000... (100 %)
Wrote 11120 bytes (7191 compressed) at 0x00001000 in 0.1 seconds (effective 828.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 105...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.0 seconds (effective 9621.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 40530.5 kbit/s)...
Hash of data verified.
Compressed 202560 bytes to 66351...

Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 202560 bytes (66351 compressed) at 0x00010000 in 1.8 seconds (effective 900.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

lbernstone, 2018-04-24:
erase_flash should definitely have some output and take some time.  It is the best way to ensure that your nvs gets cleared if corrupt.  Not sure exactly what it looks like in Windows CMD (you may need to explicitly call python esptool.py), but in linux I get the following:
```
larryb@larryb-EZpad:~/arduino-1.8.5/hardware/espressif/esp32/tools$ ./esptool.py erase_flash
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.3s
Hard resetting via RTS pin...
```

jamesbaber, 2018-04-24:
@lbernstone Progress! Turns out I had mistakenly installed esptool from the Ubuntu APT repositories, without realising that version doesn't work very well... So reinstalled using PIP and now I'm getting erase_flash output. Erased successfully, re-flashed and its all working. Thanks for your help, glad its not a bigger issue.

lbernstone, 2018-04-24:
Glas to help. Please close the issue!

closed 2018-04-24 by jamesbaber