espressif/arduino-esp32 #1349: Compilation without Arduio IDE

sanu-krishnan, 2018-04-24:
### Hardware:
Board:							Sparkfun ESP32 thing
Core Installation/update date:		11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
Is there any makefile to compile esp32-arduino-core without using Arduino IDE?
If not please guide me to make one.

What is the compiler used in Arduino to compile esp32 projects in windows platform ?

I checked the boards.txt and platform.txt files and found the recipes used by the compiler there, but couldn't figure out the exact method or compiler used.

lbernstone, 2018-04-24:
This took exactly 4 seconds of googling. https://playground.arduino.cc/Learning/CommandLine

sanu-krishnan, 2018-04-25:
Thank you for your help. 
Actually, I checked those makefiles. They are too old and have support only for AVR microcontrollers. 

I checked https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc as well. But it seems like it needs Arduino software.

What exactly I was trying is to find/make a lightweight stand-alone set of files that can be used to make esp32 projects.

My first sample project compiles and gets downloaded well on both Ubuntu and windows with MSYS2.
But "esp32_win32_msys2_environment_and_toolchain-20180110.zip" with size 500MB seems too much for me. Is there any alternative for this? Any method, that compiles esp32 projects on windows without this heavy file?

I tried cygwin, mingw32, ESPArduinoMakefile etc.. But didn't work for me.

But Arduino IDE compiles esp32 projects without MSYS2. How does this work? I checked the boards.txt and platform.txt files and found the recipes used by the compiler there, but couldn't figure out the exact method or compiler used.

lbernstone, 2018-04-25:
It uses the tools/xtensa-esp32-elf/bin executables to compile and link.  Install platformio.  It is maintained for esp32 and will compile a project from the command line.  What the IDE compiler is doing in the background is not trivial, and you will spend far more time trying to figure that out than figuring out platformio.

unixway, 2018-04-26:
Hi, I used this this doc:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

Generally it works, however component behaves somehow differently

sanu-krishnan, 2018-04-26:
@lbernstone . That means xtensa-esp32-elf/bin can compile esp32 projects without msys in windows?

I can find the recipe files inside platform.txt

Will it be too hard to write a makefile based on this platform.txt?

closed 2019-08-15 by stale[bot]