espressif/arduino-esp32 #1350: Support for vTaskList and vTaskGetRunTimeStats

trlafleur, 2018-04-24:
Will these FreeRTOS function be exposed in a future release of the Arduino code build?

vTaskList
vTaskGetRunTimeStats

wiseda51, 2018-04-24:
I've built the blink example with esp-idf and the appropriate options in sdkconfig (via make menuconfig/Component config/FreeRTOS) and the FreeRTOS options are built.  But not entirely clear to me how to merge my build into the arduino-esp32 directories.  Do I need to take the entire batch of archives from my project build and just overlay the archives in ../arduino-esp32/tools/sdk/lib?  I tried just overlaying the libfreertos.a and get a couple of undefined symbols (_thread_local_start and _thread_local_end); not clear which archive they are in.  Are there any other changes or considerations needed?

lbernstone, 2018-04-24:
Follow the directions in issue 1142.  You must have your esp-idf repo synched to the same point that arduino-esp32 was last built at (currently f586f5e) or you may have mismatched symbols/functions.  It should then just be a matter of copying the libfreertos.a archive to the lib directory.  I haven't ever done this for libfreertos.a, but have replaced libesp32 and liblwip.a before.

trlafleur, 2018-04-24:
It would be nice if it just worked without having to rebuild it all when a new release is out...

lbernstone, 2018-04-25:
arduino-esp32 is set up for simplicity.  If you are micromanaging your tasks, you should probably be using esp-idf or arduino as a component.  Please close the issue if this is now working for you.

wiseda51, 2018-04-25:
I finally got this working, there was one little trick that took a while for me to figure out, the arduino-esp32/tools/sdk/include/config/sdkconfig.h needs to be updated with the appropriate defines in the Blink/build/include/sdkconfig.h which is generated by make menuconfig.  I did a diff on the files between my esp-idf Blink example build and the arduino-esp32 version and merged the appropriate debug settings:
#define CONFIG_FREERTOS_USE_TRACE_FACILITY 1
#define CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS 1
#define CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS 1
#define CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER 1
Only the Blink/build/freertos/libfreertos.a needed to be copied to arduino-esp32/tools/sdk/lib.
vTaskList and vTaskRunTimeStats work now.

trlafleur, 2018-04-27:
It appears that in the ESP-IDF Release v3.0, release notes we see that Runtime stats are available in the base code. --> FreeRTOS Runtime statistics support (configurable via menuconfig).... 

So the work is done, just need to have a way to enable in the Arduino port with out a re-compile of the base code...

themindfactory, 2018-09-12:
Would be awesome for these to be added, more and more people spawning tasks and this info would be nice to have access to.

trlafleur, 2019-08-08:
is it possible to get these function exposed so one can use them under the Arduino platform...

Thanks

lbernstone, 2019-08-08:
Will it make more sense to you if I just say no?

trlafleur, 2019-08-09:
No, it's a very useful features of freertos when developing code.  Ive been
using freertos for 8 years now.  Lots of experience with it...

It's a great tool

atanisoft, 2019-08-09:
@trlafleur while it is a useful tool for some it isn't highly useful for the majority of Arduino users. I'd suggest shift to Arduino as an IDF component instead.

trlafleur, 2019-08-09:
Enabling the features in the Arduino ide will not effect most users, but will allow those who need it to have it.....

I have many students in my classes, and would love to access it with out having to move to the IDF....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

lbernstone, 2019-08-09:
Profiling features require memory and cpu cycles.  Most people are not willing to pay that penalty, essentially having the device in a debug mode all the time.  What I can offer, is that @atanisoft has made it significantly easier to compile your own libraries (or compile from scratch every time), by making a simplified method to compile [arduino as an esp-idf component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) in platform.io.  Watch that page in the next couple months for instructions on how to do this.

trlafleur, 2019-08-09:
Thanks, that would be great to find an easy way to build the code... PS:
ESP32 processor have a lot of CPU power, adding any debug code in most
cases make very little performance hit...

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

trlafleur, 2019-10-03:
as a minimum would consider just adding vTaskList to the Arduino release, minimal code increase...
and it's no more complicated to use as any other FreeRTOS function.

davepl, 2019-12-12:
Expose it!

closed 2020-02-24 by stale[bot]