espressif/arduino-esp32 #1356: OTA with interrupts

trlafleur, 2018-04-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:	       23/Apr/2018
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:				        115200

### Description:
Calling ArduinoOTA.handle();  in loop with lots of interrupt traffic, caused a crash when I try to update code...  My ISR has IRAM_ATTR set...

If I remove the active interrupt source, and do an update, all is well...

If you need some test code, I should be able to build a small sample...
~~~
Progress: 0%
Progress: 0%
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400d1ea0  PS      : 0x00060034  A0      : 0x400816c8  A1      : 0x3ffc0bf0  
A2      : 0x0000000c  A3      : 0x3ffc4610  A4      : 0x00000000  A5      : 0x00001000  
A6      : 0x00000000  A7      : 0x1300005c  A8      : 0x80080ea8  A9      : 0x80000020  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc5498  A13     : 0x00000000  
A14     : 0x3ffc5494  A15     : 0xffffffff  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082035  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d1ea0

Backtrace: 0x400d1ea0:0x3ffc0bf0 0x400816c5:0x3ffc0c10 0x40082032:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57
~~~

lbernstone, 2018-04-26:
Just need to see what the ISR looks like.  ISRs should complete quickly or they can trigger watchdogs.  If you need more processing than that, turn on a flag variable that can be processed in the main loop.

everslick, 2018-05-03:
Is suggest turning off as much as possible prior to OTA. while this may not fix the root cause of the problem, i'd consider it good practice anyway.

AFAIK, there are plans to make ArduinoOTA completely async, which might improve things on your front as well.

jnicolson, 2018-12-29:
Adding a comment to this because I had a similar problem but with a single timer interrupt.

My ISR was identical to the one from the example code (work all handled from reading Semaphore flag in loop() ):

https://github.com/espressif/arduino-esp32/blob/6dd8be32629a3d9cd6610da93224f3559f63c6ee/libraries/ESP32/examples/Timer/RepeatTimer/RepeatTimer.ino#L21-L30

It looks like the culprit was the millis() call as when I replaced it with the FreeRTOS xTaskGetTickCount(), OTA started working again.

I'm not sure why as millis (and micros) are both defined with IRAM_ATTR and then from what I could see in ESP-IDF all the timer related functions were too, but the instant i swapped it back to millis OTA would fail again.

closed 2019-08-15 by stale[bot]