espressif/arduino-esp32 #1358: ESP Specific APIs - Missing API's

timpur, 2018-04-26:
There are some APIs from ESP8266 lib ESP missing like:
```
ESP.getSketchSize()
ESP.getFreeSketchSpace()
ESP.getSketchMD5()
```
Most interested in getSketchMD5. Working to implement these, but new to esp32 and need to understand the IDF more, so if anyone has hints, these are appreciate. Trying to understand how partitions work, since i think ill have to hash the current working partition, similar to how the esp8266 does it.

lbernstone, 2018-04-26:
If you are up to the C coding, the relevant headers are esp_image_format.h and esp_ota_ops.h.  I took a quick look, and didn't see an easy way to stream the data through md5builder, but those headers are the place to start.  Look through the Update library as well.

timpur, 2018-04-29:
@lbernstone, question , why does `Update` not the the `Over The Air Updates (OTA) APIs` from the `ESP IDF` ?

http://esp-idf.readthedocs.io/en/latest/api-reference/system/ota.html

lbernstone, 2018-04-30:
Not a question for me.  I just loiter here.  You'll need to have @me-no-dev answer that.  He probably wanted more control than just handing off the read and write to the C code.

timpur, 2018-04-30:
Okay thanks :)

closed 2019-08-15 by stale[bot]

sillyfrog, 2019-09-27:
Fixed in #2176