espressif/arduino-esp32 #1361: WifiClientSecure::connect does not give any certificate-fail clarification

Rob58329, 2018-04-27:
### Hardware:
Board:							ESP32 "NodeMCU-32S"
Core Installation/update date:			GitHub as at 26Apr18
IDE name:							Arduino IDE

### Description:
WiFi_Client_secure.connect [WifiClientSecure::connect] does not distinguish between a [fail due to no-connection-to-server/no-reply-from-server] and a [fail due to bad security-certificate/"test_root_ca"]

### Details:
I note that using the current GitHub (as at 26Apr18), when the command "WiFi_Client_secure.connect(address, port)" fails for any reason, it returns zero.  IE. it appears not possible to determine if the failure was due to (a) no reply from the server, or (b) a "test_root_ca" or other security-certificate-failure (due to expiry or other reason).

In my application, if the failure is due to a "security-certificate failure", I would like to be able to note this and then continue (or re-start) the connection ignoring the security-certificate-failure (which is possible by resetting the "*test_root_ca=NULL;").  However to do this I need know if the "WiFi_Client_secure.connect(address, port)" failure was due to the security-certificate or due to a connection failure (ie. the server not replying).

This was possible for the ESP8266 (as the [WiFi_Client_secure.verify] was a separate command).  But for the ESP32 (which does the verify within .connect) there currently seems to be no way to distinguish these 2 different failure reasons (as the "WiFi_Client_secure.connect" just returns zero for any type of failure).

Is there any way to determine/identify these 2 different failure reasons?

[PS. I note that the "espressif erno.h" has an error message "#define EKEYEXPIRED 127 /* Key has expired */", but don't think this is currently made use of in the above?]

Any info would be appreciated!

copercini, 2018-04-27:
If you are just debugging, enable the verbose logs of arduino IDE and it will give you the exactly error, but if you want to catch it in the code, `WifiClientSecure::lastError()` can be used like this:
```cpp
  if (WiFi_Client_secure.connected()) {
      Serial.println("Connected!");
  } else {
      Serial.println("Connection failed!");
      char err_buf[100];
      if (WiFi_Client_secure.lastError(err_buf, 100) < 0) {
          Serial.println(err_buf);
      } else {
          Serial.println("Connection error");
      }
  }
```

closed 2018-04-29 by Rob58329

iamneo2416, 2018-06-15:
ME TOO, I meet the same problem !!

### Hardware:
Board:							ESP8266 NodeMCU V2
Core Installation/update date:		2018/03/21
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:

I installed ubuntu 16.04 apache web server with self-signed certificate in LAN environment and use browser to access **https://192.168.1.210/** web page always successfully.

I even use **dos command line** to **telnet 192.168.1.210 443** to check port listening. It's working perfectly.
**C:\> telnet 192.168.1.210 443**

But when I use **WiFiClientSecure client.connect(ip, 443)**. It always get fail return. Is this a bug for ipv4 connect in connect function ?

Here is the exact step I followed to build this virtual machine (192.168.1.210) w/self-signed certificate.
[https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04)

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char * host = "192.168.1.210";
const int port = 443;

static WiFiClientSecure g_client;
String wifi = "ssid";
String pass = "pass";

unsigned char root_ca[] = {
  0x30, 0x82, 0x04, 0x02, 0x30, 0x82, 0x02, 0xea, 0xa0, 0x03, 0x02, 0x01,
  0x02, 0x02, 0x09, 0x00, 0xc5, 0xb6, 0xe0, 0xe5, 0x41, 0x5f, 0x08, 0xa6,
  0x30, 0x0d, 0x06, 0x09, 0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x01,
  0x0b, 0x05, 0x00, 0x30, 0x81, 0x94, 0x31, 0x0b, 0x30, 0x09, 0x06, 0x03,
  0x55, 0x04, 0x06, 0x13, 0x02, 0x54, 0x57, 0x31, 0x0f, 0x30, 0x0d, 0x06,
  0x03, 0x55, 0x04, 0x08, 0x0c, 0x06, 0x54, 0x61, 0x69, 0x77, 0x61, 0x6e,
  0x31, 0x0f, 0x30, 0x0d, 0x06, 0x03, 0x55, 0x04, 0x07, 0x0c, 0x06, 0x54,
  0x61, 0x69, 0x70, 0x65, 0x69, 0x31, 0x15, 0x30, 0x13, 0x06, 0x03, 0x55,
  0x04, 0x0a, 0x0c, 0x0c, 0x42, 0x61, 0x79, 0x65, 0x73, 0x74, 0x65, 0x6b,
  0x20, 0x43, 0x4f, 0x2e, 0x31, 0x0c, 0x30, 0x0a, 0x06, 0x03, 0x55, 0x04,
  0x0b, 0x0c, 0x03, 0x4d, 0x49, 0x53, 0x31, 0x16, 0x30, 0x14, 0x06, 0x03,
  0x55, 0x04, 0x03, 0x0c, 0x0d, 0x31, 0x39, 0x32, 0x2e, 0x31, 0x36, 0x38,
  0x2e, 0x31, 0x2e, 0x32, 0x31, 0x30, 0x31, 0x26, 0x30, 0x24, 0x06, 0x09,
  0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x09, 0x01, 0x16, 0x17, 0x74,
  0x6f, 0x6d, 0x6d, 0x79, 0x2e, 0x79, 0x61, 0x6e, 0x67, 0x40, 0x62, 0x61,
  0x79, 0x65, 0x73, 0x74, 0x65, 0x6b, 0x2e, 0x63, 0x6f, 0x6d, 0x30, 0x20,
  0x17, 0x0d, 0x31, 0x38, 0x30, 0x36, 0x31, 0x31, 0x31, 0x30, 0x33, 0x30,
  0x33, 0x37, 0x5a, 0x18, 0x0f, 0x32, 0x31, 0x31, 0x38, 0x30, 0x35, 0x31,
  0x38, 0x31, 0x30, 0x33, 0x30, 0x33, 0x37, 0x5a, 0x30, 0x81, 0x94, 0x31,
  0x0b, 0x30, 0x09, 0x06, 0x03, 0x55, 0x04, 0x06, 0x13, 0x02, 0x54, 0x57,
  0x31, 0x0f, 0x30, 0x0d, 0x06, 0x03, 0x55, 0x04, 0x08, 0x0c, 0x06, 0x54,
  0x61, 0x69, 0x77, 0x61, 0x6e, 0x31, 0x0f, 0x30, 0x0d, 0x06, 0x03, 0x55,
  0x04, 0x07, 0x0c, 0x06, 0x54, 0x61, 0x69, 0x70, 0x65, 0x69, 0x31, 0x15,
  0x30, 0x13, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x0c, 0x0c, 0x42, 0x61, 0x79,
  0x65, 0x73, 0x74, 0x65, 0x6b, 0x20, 0x43, 0x4f, 0x2e, 0x31, 0x0c, 0x30,
  0x0a, 0x06, 0x03, 0x55, 0x04, 0x0b, 0x0c, 0x03, 0x4d, 0x49, 0x53, 0x31,
  0x16, 0x30, 0x14, 0x06, 0x03, 0x55, 0x04, 0x03, 0x0c, 0x0d, 0x31, 0x39,
  0x32, 0x2e, 0x31, 0x36, 0x38, 0x2e, 0x31, 0x2e, 0x32, 0x31, 0x30, 0x31,
  0x26, 0x30, 0x24, 0x06, 0x09, 0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01,
  0x09, 0x01, 0x16, 0x17, 0x74, 0x6f, 0x6d, 0x6d, 0x79, 0x2e, 0x79, 0x61,
  0x6e, 0x67, 0x40, 0x62, 0x61, 0x79, 0x65, 0x73, 0x74, 0x65, 0x6b, 0x2e,
  0x63, 0x6f, 0x6d, 0x30, 0x82, 0x01, 0x22, 0x30, 0x0d, 0x06, 0x09, 0x2a,
  0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x01, 0x01, 0x05, 0x00, 0x03, 0x82,
[... 124 lines cut ...]
=== CERTIFICATE ISSUED BY ===
Common Name (CN):	192.168.1.210
Organization (O):		BestMe CO.
Organizational Unit (OU):MIS
Location (L):			Tokyo
Country (C):			JPN
State (ST):			Japan
Not Before:			Mon Jun 11 10:30:37 2018
Not After:			Sun Apr 11 04:02:21 1982
RSA bitsize:			2048
Sig Type:				SHA256
CA Cert set
Connecting to 192.168.1.210
State:	sending Client Hello (1)
Alert: handshake failure **<== why failed, but browser always sucess ?**
Error: SSL error 40
Alert: unexpected message **<== why strange message here?**
Error: SSL error 40
Alert: close notify
Connection failed
pm open,type:2 0

```

```

Rob58329, 2018-06-15:
@iamneo2416 

I have a couple of comments on your above code:
(1) If you are using an ESP32, then I believe that you should use "#include <WiFi.h>" and not "#include <ESP8266WiFi.h>"
(2) If you are using an ESP8266, then I believe it does not have the ability to use a certificate as above, the only thing you can do with an ESP8266 is to check the FINGERPRINT.
(3) If you are getting the message "'class WiFiClientSecure' has no member named 'lastError'" then I suspect you are not using the "github.com/espressif/arduino-esp32" version of the software (current last commit was 31May18).

iamneo2416, 2018-06-19:
1. YES, I use **ESP8266** chip.
2. from debug message I think CA was set and loaded successfully. That's why CA was printed on the debug log. Check debug log and you will see CA was loaded w/succeed.
**=== CERTIFICATE ISSUED TO ===**   and 
**=== CERTIFICATE ISSUED BY ===**

3. when it runs to line **if(!g_client.connect(host, port)),** it always get fail and return. I use wireshark to listen to network packets, and it always show SSL error 40 with unexpected message sent from client. There must be some bugs inside SSL handshake sequence. I don't think SSL provider will do thing wrong. I installed it from public domain and it runs for a very lon time wo/errors. Server receive some message unexpected that was send by client which I use WiFiClientSecure client.connect(). That means SSL handshake meet some unexpected error sequence.

Rob58329, 2018-06-19:
@iamneo2416 

This is the "ESP32-issues" section of github, and not for "ESP8266-issues".
(The ESP32 is a much more powerful chip than the ESP8266.)

As I said, "If you are using an ESP8266, then I believe it does not have the ability to use a certificate as above, the only thing you can do with an ESP8266 is to check the FINGERPRINT."

I think this is because the ESP8266 does not have sufficient RAM to hold and process the full certificate as well as run the "WiFi_Client_secure." routines.  Instead, you have to only store the "Certificate-Fingerprint" on the ESP8266 (ie. NOT the full certificate as in your example code), and then use the ESP8266's "WiFi_Client_secure.verify()" command to check this Fingerprint.  Your errors are probably because the ESP8266 is running out of RAM.

ESP8266 code examples can be found at:
"https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/client-secure-examples.rst"