espressif/arduino-esp32 #1362: WifiClient connect unstable: lwip_connect_r:113

robertcasanova, 2018-04-27:
### Hardware:
Board:							Adafruit Huzzah32
Core Installation/update date:			26/04/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

#180 This issue is still present.
The connection seems very unstable and sometimes works well and some others i have this error:

`[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113`

I'm using a Adafruit Huzzah32 but should not be board related.
To check this behaviour I'm using the Example sketch: Wifi -> WifiClient without modifications.

Any ideas? At least if there's a workaround because at the moment Wifi connection is not usable at all for a project.

Thx for help!

fanhuafeng, 2019-02-13:
I have same problem.[E][WiFiClient.cpp:213] connect(): lwip_connect_r: 113.
I reconnect esp32 to  AP,but can't resolve the problem, must reset the device.

TheNetStriker, 2019-02-25:
I also have this problem with my Adafruit Huzzah32, but it seams as it only occurs when connecting to one of my two wireless access points. On my Fritz!Box 7490 the connection works for days without problems. But when I connect the Huzzah32 to my older D-LINK DIR-635 access point the connection only works for about 3 to 5 Minutes until the connection drops and I get those debug messages.

Could this be a problem with older access points or is my older access point simply broken?

closed 2019-08-15 by stale[bot]