espressif/arduino-esp32 #1363: Documentation?

tarmogr, 2018-04-28:
Sorry for stupid question but where can I find documentation? I mean yes there are Arduino install/setup instructions but is there a description of what commands to use if you want to for example: connect to wifi, pull some output pins high, read inputs, send GET, POST etc. requests?

Or this type of info only exists in examples/youtube tutorials etc?

lbernstone, 2018-04-28:
Specifics for esp32 can be found in the example sketches included with the various libraries (<esp32>/libraries/*/examples/*), but general tutorials for the arduino and esp8266 should apply fairly well to esp32.  Take a look at [arduino.cc](https://www.arduino.cc/en/Guide/HomePage), or google "getting started esp8266".  I found it easiest to figure out what you want to automate, and then look around for examples of people doing that which you can adapt.

everslick, 2018-05-03:
Please do not use the issue tracker for support questions. The prefered medium for general questions is the [esp32-arduino](https://gitter.im/espressif/arduino-esp32) channel on gitter. **This issue tracker is for bugreports regarding the Arduino Core for ESP32.**

We are swamped with issues, that do not belong here. Thanks for your understanding. Please close the issue, if you agree.

skyladle, 2018-05-07:
ya, should open a branch for general questions.  so many good knowledge and skill man here, so people love raise question here. hope a new branch for general question as hot as here. and we never embarrass raise question does not matter with arduino core for eSP32

closed 2019-08-15 by stale[bot]