espressif/arduino-esp32 #1364: Missing files in tools directory of Arduino-ESP32

alacivita, 2018-04-28:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			28/04/2018
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I'm trying to reproduce the example mqtt_esp8266 but using a esp32.
When I press Verify Arduino IDE returns: 
fork/exec /home/pi/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory
Error compiling for board ESP32 Dev Module.
Effectively the xtensa-esp32-elf directory doesn't exist.

I checked issue  #581 and redo all the process including the last line in different formats:
python2 get.py
python get.py
sudo python get.py

Allways with the same result:
pi@raspberrypi:~/Arduino/hardware/espressif/esp32/tools $ sudo python get.py
System: Linux, Info: Linux-4.9.80-v7+-armv7l-with-debian-9.4
Platform: arm-linux-gnueabihf
Tool mkspiffs-0.2.2-arduino-esp32-linux-armhf.tar.gz already downloaded
Extracting mkspiffs-0.2.2-arduino-esp32-linux-armhf.tar.gz
Renaming mkspiffs-0.2.2-arduino-esp32-linux-armhf to mkspiffs
Done

What am i doing wrong?

lbernstone, 2018-04-29:
The toolchain does not come compiled for ARM. https://desire.giesecke.tk/index.php/2018/01/30/building-esp32-toolchain-on-arm-armbian-raspbian/

alacivita, 2018-05-01:
Thanks you very much for your answer.
Do you know if there is any way to do the same for a Raspberry Pi with Raspbian Stretch?

alacivita, 2018-05-01:
The instructions on the link gave me some error. I suposse it's not valid for Raspbian Stretch.
But I have a toolchain for ARM created in a previous installation of my RPI based on these instructions:
[https://esp32.com/viewtopic.php?t=256](url)
that are almost equal to those you recomended.
So I copied the xtensa-esp32-elf directory of that installation to home/pi/Arduino/hardware/espressif/esp32/tools and the Verify process seems to work up to a point where a new error message apears:
as: unrecognized option '--longcalls'
Any idea?

BudBundi, 2018-05-02:
I compiled the tool chain last night with the instructions from the ebook "Kolbans's Bokk on ESP32"

The CPU on the Raspberry Pi is ARM based which means that we need a cross
compiler to build executables for an ESP32. At the time of writing, there is not a known
location for a binary download of the tool suite for ESP32 so our choice is to build it from
scratch. Fortunately there are instructions for this in the "linux-setup.rst" document
found as part of the ESP-IDF project. In summary the steps are:

$ sudo apt-get install gawk gperf grep gettext libcurses5-dev python python-dev automake bison flex texinfo help2man libtool libtool-bin
$ git clone -b xtensa-1.22.x https://github.com/espressif/crosstool-NG.git
$ cd crosstool-NG
$ ./bootstrap && ./configure --prefix=$PWD && make install
$ ./ct-ng xtensa-esp32-elf
$ ./ct-ng build
$ chmod -R u+w builds/xtensa-esp32-elf

Be aware that the build of the tool chain requires about 4GBytes of storage in order to
complete. As such, you will likely need a much larger micro SD card than the minimum
of 8GBytes that is normally used with a Pi. The resulting binaries for the tool chain end
up at about 100MBytes. You might want to download the files pre-compiled. A version
of the files can be downloaded from my web site from:
http://www.neilkolban.com/esp32/downloads/xtensa-esp32-elf.tar.gz

Once downloaded, I suggest extracting the content into /opt using:
$ sudo tar --extract --directory /opt --ungzip --file xtensa-esp32-elf.tar.gz

alacivita, 2018-05-03:
Kolban's recipe is the same one I used with an earlier version of Raspbian
and it worked well. But it did not work with Stretch. For example, just the
first line throws many errors:

pi@raspberrypi:~/esp32 $ sudo apt-get install gawk gperf grep gettext
libcurses5-dev python python-dev automake bison flex texinfo help2man
libtool libtool-bin
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package texinfo is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  install-info info

Package automake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libtool is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package flex is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package bison is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package gawk
E: Unable to locate package gperf
E: Unable to locate package gettext
E: Unable to locate package libcurses5-dev
E: Package 'automake' has no installation candidate
E: Package 'bison' has no installation candidate
E: Package 'flex' has no installation candidate
E: Package 'texinfo' has no installation candidate
E: Unable to locate package help2man
E: Package 'libtool' has no installation candidate
E: Unable to locate package libtool-bin

That's why i copied that previous genereted toolchain below the tools
directory and it seems to work until the point where the it returns the
error
as: unrecognized option '--longcalls'
The problem is that I have no clue of where that option comes from.

BudBundi, 2018-05-03:
Strange the packages are available for stretch eg https://packages.debian.org/stretch/texinfo
I think I must not ask, but have you apt-get update and dist-upgrade before?

Or have you try to download Kolban's version and install it in /opt?

alacivita, 2018-05-06:
Good for asking!!! I use to run apt-get update and apt-get upgrade, but I
never run apt-get dist-upgrade.
After that Kolban's recipe worked fine.
Thanks you very much.

alacivita, 2018-05-15:
Just one comment, the command:

sudo apt-get install gawk gperf grep gettext libcurses5-dev python python-dev automake bison flex texinfo help2man libtool libtool-bin

gave me an error:

E: Unable to locate package libcurses5-dev

I had to replace it by 

sudo apt-get install gawk gperf grep gettext libncurses5-dev python python-dev automake bison flex texinfo help2man libtool libtool-bin

("lib**n**curses5-dev" in place of "libcurses5-dev") to make it work

closed 2018-05-20 by alacivita