espressif/arduino-esp32 #1367: Unexpected bin filesize increase (+300kB) after core upgrade

mvdbro, 2018-04-29:
### Hardware:
Board:							ESP32 Lolin Lite
Core Installation/update date:		16/apr/2018
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:					256000

### Description:
Have been running a large sketch on Arduino ESP32 core, dated 2017-12-19. Output bin filesize: 866 kB
Upgraded Arduino IDE to  use Arduino ESP32 core, dated 2018-04-16. Output bin filesize: 1169 kB
Without changing a single line within the sketch code, bin filesize has increased 303 kB (!)
Was this expected due to a newer SDK or something?

reaper7, 2018-05-02:
You have to accept it...

similar: 
https://github.com/espressif/arduino-esp32/issues/1194
https://github.com/espressif/arduino-esp32/issues/1282

https://github.com/espressif/arduino-esp32/pull/1302

CRprog, 2018-05-03:
Board: ESP32 DoIt V1
Core Installation/update date: 16/Apr/2018  (Version used for comparing: Dez/2018)
IDE name: Arduino IDE 1.9.0 beta 44

I have so far no Problems with the .bin-size-increase .. but I'm interested. For what are these many bytes used for? Code optimization, speedups or similar?
My tests:
“BareMinimum.ino” is growing from 135724 Bytes (10%) to 159060 Bytes (12%).

And some Code from me, using the following parts ….
#include <WiFi.h> #include <FS.h> #include <AsyncTCP.h> #include <ESPAsyncWebServer.h>
#include <ESPmDNS.h> #include <SPIFFS.h> #include <OneWire.h>
... is growing 557kb -> 703kb    +146kb !  ... although without any BLE-Stuff.

lbernstone, 2018-05-04:
The size of the esp-idf libraries has gone up.  I'm guessing that is the root cause here.

mvdbro, 2018-05-04:
Despite the fact were not using BLE...

I still guess that lots of new (but unreferenced) code is now included by design. I think I've seen similar issues on the ESP8266 project and those were fixed later on.

The exact same source code builds into almost twice the size compared between ESP8266 and ESP32. 

But yes, looks like we have to live with it.

closed 2018-05-04 by mvdbro