espressif/arduino-esp32 #1370: Logging incorrect for i2c i2cProcQueue() function

echoGee, 2018-04-30:
### Hardware:
Board:							ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:						Platform.io
Flash Frequency:					80Mhz
Upload Speed:					?

### Description:
Logging for I2C incorrect. tBefore and tAfter are defined at same location 
In file esp32-hal-i2c.c: ``` i2cProcQueue() ```
### Sketch:
```cpp

   i2c->dev->ctr.trans_start=1; // go for it

    uint32_t eBits = xEventGroupWaitBits(i2c->i2c_event,EVENT_DONE,pdFALSE,pdTRUE,ticksTimeOut);

    //log_e("after WaitBits=%x @tick=%d",eBits,xTaskGetTickCount());

#if ARDUHAL_LOG_LEVEL >= ARDUHAL_LOG_LEVEL_ERROR
    portTickType tBefore=xTaskGetTickCount();
    portTickType tAfter=xTaskGetTickCount();
#endif
```
should be 
```cpp
   i2c->dev->ctr.trans_start=1; // go for it

#if ARDUHAL_LOG_LEVEL >= ARDUHAL_LOG_LEVEL_ERROR
    portTickType tBefore=xTaskGetTickCount();
#endif

    uint32_t eBits = xEventGroupWaitBits(i2c->i2c_event,EVENT_DONE,pdFALSE,pdTRUE,ticksTimeOut);

    //log_e("after WaitBits=%x @tick=%d",eBits,xTaskGetTickCount());

#if ARDUHAL_LOG_LEVEL >= ARDUHAL_LOG_LEVEL_ERROR
    portTickType tAfter=xTaskGetTickCount();
#endif
```
### Debug Messages:
See https://github.com/espressif/arduino-esp32/issues/1352#issuecomment-385220706 for more details

stickbreaker, 2018-06-27:
@echoGee You can close this now, me-no-dev pulled.

Chuck.

echoGee, 2018-07-02:
@stickbreaker Feel free to close it. Could you add the commit for the pull here for reference ?

stickbreaker, 2018-07-02:
@echoGee the pr happened in the strickbreak-i2c branch, me-no-dev cleaned up(deleted) all development branches when he merged stickbreaker-i2c.  When he was preparing for the RC1.

I cannot close this issue, I do not have edit writes for this repo.  You have the right to close it because you opened it.

Chuck.

closed 2018-07-03 by echoGee