espressif/arduino-esp32 #1372: Handing out NO gateway via DHCP?

schweini, 2018-05-02:
I am trying to get my Android phone connected to my ESP32, while at the same time maintaining my mobile connection.

The problem seems to be that Android simply routes all traffic over the mobile data connection, until it is sure that the wifi network provides internet connectivity.

(strange side node: the WebView inside the 'sign in to network' pop-up can see the ESP32 - no other browser can. Only if I disable mobile data, can I use a browser to visit the ESP32's IP)

I suspect (but am not sure) that Android doesn't trust an AP that defines a default gateway, but offers no internet connectivity.

I would like to, therefore, NOT hand out a default gateway address over DHCP when my phone connects. 

Is there a way to do this? Or some line I could quickly comment out for testing if this actually works?

lbernstone, 2018-05-02:
This doesn't have anything to do with ESP32, but simple TCP/IP behavior.  The stack will use whichever route it thinks will get it to the destination address.  If your ESP is using a private address (eg, the default softAP address of 192.168.4.1), then your phone will route requests to that network through the softAP.  All other traffic will go through the mobile network.  Are you saying that if you have mobile on that you cannot get to a webserver running on the ESP, but as soon as you turn it off you can get there?  The "sign in to network" captive portal from WiFiManager is just a redirect to that address.

schweini, 2018-05-02:
Sadly, that's not how it seems to work on Android, even though that should
be the standard behaviour, and my computers do it that way.

Android does seem to block routing to a network that doesn't have internet
access, probably as an anti-spoofing measure.

schweini, 2018-05-02:
Here's someone with the same problem on ESP8266:

https://github.com/esp8266/Arduino/issues/1956

And someone recommended trying
````wifi_softap_set_dhcps_offer_option(OFFER_ROUTER, 0)```
but google can't tell me if that function exists in ESP32/Arduino or how to set that.

schweini, 2018-05-02:
I commented out 
`	info.gw.addr = static_cast<uint32_t>(gateway);`
in WiFiAP.cpp (hoping that this would cause the gateway not to be set), but this causes my Android phone to take on a gateway address of 235.103.13.128, which is in the IANA multicast IP block,

everslick, 2018-05-03:
@schweini I use the ESP soft AP with Android and I **cannot** confirm your problem. when i connect to the softap, my phone tells me that, this AP does not seem to provide an internet connection and if you want to connect anyway. after clicking yes, everything works as expected. I have mobile data ON all the time. this is on a Nexus from LG with Android 8.0

schweini, 2018-05-03:
This is potentially very good news!

Can you confirm that:
A) you can browse to the ESP32's IP (while it is in AP mode)
B) at the same time navigate the general internet
?

If yes, which sketch are you running? Because the standard Captive Portal
example does cause the problem for me.

Does you program cause the "sign in to network" popup?

I'm sorry that I'm being nosy, but I don't seem to be the only person
encountering this problem - but on the other hand, it doesn't seem to be as
common a problem (there are even links to Android java functions to remedy
this via connection manager) as I'd think.

everslick, 2018-05-03:
> A) you can browse to the ESP32's IP (while it is in AP mode)

yes.

> B) at the same time navigate the general internet

no.

But I did not expect B) to work. I obviously misunderstood your issue. I read it as 'you cannot reach the ESP AP as long as you have mobile data enabled'. but you have a point. if the default gateway still points to the mobile data interface, it might be possible to have both.

nevercast, 2018-07-25:
I'm looking for this functionality on a different downstream project from the ESP-IDF (Pycom), is this functionality already available in the IDF and this issue is regarding Arduino implementation?

closed 2019-08-15 by stale[bot]

no-trick-pony, 2021-06-17:
Did anyone ever find a solution to this. I am struggling with that too ATM. I can either connect to the ESP32 AP and access the webserver it provides but can NOT browse the internet on the phone or the the other way around. This seems to be an Android issue though.

ammgws, 2021-07-27:
@no-trick-pony Did you figure anything out this past month?

chrismerck, 2021-09-30:
FWIW, I was able to disable the offering of a gateway by changing `tcpip_adapter_init()` in `esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c`:

        IP4_ADDR(&esp_ip[TCPIP_ADAPTER_IF_AP].ip, 10, 7, 42, 1); // we override the default AP IP because Espressif's default conflicts with some customer networks. This IP is less likely to conflict unless the user is a big Douglas Adams fan.
        IP4_ADDR(&esp_ip[TCPIP_ADAPTER_IF_AP].gw, 0, 0, 0, 0);  // making this 0.0.0.0 disables gateway offer
        IP4_ADDR(&esp_ip[TCPIP_ADAPTER_IF_AP].netmask, 255, 255 , 255, 0);

I've only tested on iOS, but I find that it causes the phone to be able to communicate on the Config AP *and* on the internet via cellular. It does have the unfortunate side-effect of preventing the iOS app from viewing the current SSID (something Apple surprisingly allows the app to do), however the app can still attempt to communicate with the device via its IP address.

tjunussov, 2023-10-12:
> [chrismerck](/chrismerck)

Hi! here you mentioned filepath to `esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c` but Arduino-ESP32 doen't have such folder ? is it refered to ESP-IDF not  Arduino-ESP32?