espressif/arduino-esp32 #1375: HardwareSerial brick my ESP32 Pls Help !

Fbo06, 2018-05-03:
Hello, i use this script and it seems that it brick my ESP32 WEMOS LOLIN32 V1.0.0
now when i try to upload any script, i have a ....----....---- until the message : A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
How can i erase these bad script ?

Please Help if you can. thank you.

#include <HardwareSerial.h>
HardwareSerial MySerial(0);
void setup() {
    MySerial.begin(115200, SERIAL_8N1, 16, 17);
}
void loop() {
    while (MySerial.available() > 0) {
        uint8_t byteFromSerial = MySerial.read();
        // Do something
    }
    
    //Write something like that
    MySerial.write(rand());
}

lbernstone, 2018-05-03:
Two options:
1) If you have buttons for reset and gpio0, you can hold down gpio0 and then press reset.  This will put the esp32 in a "waiting for download" state and you can then upload good code.
2) run `python esptool.py erase_flash` from the command line.  This will wipe all code from the device (including the nvs partition).

beegee-tokyo, 2018-05-04:
Good point, everyone should now that when using an ESP32. I added it to the [Wiki](https://desire.giesecke.tk/index.php/2018/05/04/upload-of-code-over-usb-serial-fails-because-of-erroneous-code/)

Fbo06, 2018-05-04:
Hello lbernstone, thank you for your response, for the first solution, I only have one button called reset, but nothing but reset happen when i push on it. i have a pin called IO0 do i manual connect connect it to the GRD to put the ESP32 in a "waiting for download" mode ?
- for the second solution, what is the incidence to erase the cvs partition ?
thank you for your response again.

beegee-tokyo, 2018-05-04:
Erasing the NVS partition clears all stored WiFi credentials you used before and as well all Bluetooth paring informations of Bluetooth devices you connected to before.
I used it several times and after flashing a new application everything worked without problems.

Fbo06, 2018-05-04:
thank you beegee-tokyo i 'll try that solution. I am on OS X do i only type "python esptool.py erase_flash" on a command line ?

Fbo06, 2018-05-04:
i found this link who can help : https://github.com/espressif/esptool

Fbo06, 2018-05-04:
when i try this python code : python esptool.py erase_flash
 i get this message : 
Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
    import serial
ImportError: No module named serial

beegee-tokyo, 2018-05-04:
Actually on a second thought that cannot work if you rerouted the Serial0 pins to other GPIOs than the one connected to the Serial/USB chip. Because esptool communicates over the USB with the device. Try the first solution with pulling down GPIO0 (I would use a 4.7k or similar resistor to ground) then push reset release reset then remove the pulldown and try to flash a simple app like the blink example.

Fbo06, 2018-05-04:
hello beegee-tokyo your last solution work 👍 thank you very much. i retrieve my ESP32 like new. You are the boss

beegee-tokyo, 2018-05-04:
You're welcome. Please close the issue if the problem is solved.

lbernstone, 2018-05-04:
@beegee-tokyo: esptool should still work.  The bootloader tx/rx has not changed, so as long as there is a USB->UART adapter, it is going to physically pull down gpio0 and get into the bootloader, where it uses the hardware pins to program it.  I do not know how to run esptool on a Mac, since I would never spend $1500 on a laptop...

closed 2018-05-04 by Fbo06

oparex, 2019-11-05:
Hello,

I got a never used esp32_devkitc_v4 from a friend and tried to flash it with a simple led sketch. I got the same connection error as mentioned above. I tried a lot of things including what was suggested here.

1. I pulled down gpio0 and pressed reset, still could not upload.
2. I ran the tool command and got the following
```
$ python esptool.py erase_flash
esptool.py v2.9-dev
Found 7 serial ports
Serial port /dev/cu.usbserial-1440
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.usbserial-1440 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header```

I even connected a capacitor to en and gnd as proposed by someone to fix the connection issue.

I am still not able to upload to ESP32. Please help.

lbernstone, 2019-11-05:
Don't hijack issues.
Check [mac osx specific issues](https://github.com/espressif/arduino-esp32/issues?utf8=%E2%9C%93&q=is%3Aissue+mac+osx) esp #1084

oparex, 2019-11-06:
sorry for doing that.
I checked the issue and solved the problem.
USB to UART driver was missing.
Thanks!