espressif/arduino-esp32 #1376: Cannot find esp32 with MacBook Pro High Sierra.

AsafFisher, 2018-05-03:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE - Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Cannot find board at all. Although all drivers for Mac were downloaded. I tried with windows and it works. 
It stopped detecting it after I took my Mac for a repair in apple store, everything was restored and is the same, but after that i couldn't find it anymore... (On windows it works perfectly...)

### Debug Messages:
```
Arduino: 1.8.4 (Mac OS X), Board: "NodeMCU-32S, 80MHz, 115200"

Board at /dev/cu.SLAB_USBtoUART is not available

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

lbernstone, 2018-05-03:
Please google usb-prober.app and close this issue.  This is a Mac issue, not an issue with arduino-esp32.

AsafFisher, 2018-05-03:
Don't you think it should be mentioned somewhere just to avoid confusion?

AsafFisher, 2018-05-03:
Ill close it...

closed 2018-05-03 by AsafFisher