espressif/arduino-esp32 #1377: 3.7V Lithium battery as power supply for the DOIT ESP32 DevKit-v1

volkerpetersen, 2018-05-03:
### Hardware:
Board:							DOIT ESP32 DevKit-v1
Core Installation/update date:			?11/jul/2017?
IDE name:					        ?MicroPython
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Has anyone experience powering the DOIT ESP32 with a 3.7V Lithium battery? I've tried putting the 3.7V on Vcc - failed. Putting it on the USB port fails too. I even used an external low voltage-drop regulator (0.4V at 400mA) and put it on 3V3 - failed. Last and final test was putting the regulated 3.3V onto Pin 2 (Vout) of the onboard voltage regulator - failed.
In all cases the board doesn't boot and the current draw cycles between 50 and 100mA with about a 1 second frequency. Thus, it looks as if the unit starts up, voltage drops, causing the device to reset, current drops and the whole cycle start all over again.

Any idea / suggestions?

lbernstone, 2018-05-04:
This is a basic electronics question, not esp32.  Please close this issue.  If your board does not have a header for a battery, then it likely cannot directly use one.  Many boards have a CP2102 on board which has a regulator to charge and power 3.3V.

WereCatf, 2018-05-04:
@volkerpetersen It's quite simple: you connect the battery to a proper 3.3V LDO and the output from that to the 3.3V - pin.

@lbernstone The CP2102 is a USB-to-serial adapter, not a regulator of any kind, and has nothing to do with li-ion cells or providing power.

beegee-tokyo, 2018-05-04:
@WereCatf 
Check the CP2102 datasheet, section 2.1:
[image]
But you are right, it still needs an external battery charger module. But in section 2.2 you can see that the CP2102N can detect the presence of such a module and charge the battery over USB:
[image]

stickbreaker, 2018-05-04:
The builtin regulator of the cp2102 is only rated for 100mA ouput, minimum Vin of 4.0.  Both of these limits are violated when connected to a 3.7v battery and trying to power a ESP32.  

[image]

The [ESP32 Datasheet](https://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) 
lists 500mA as minimum power supply capacity.
[image]
 
By its self, the cp2102 cannot support a ESP32 from USB or 5v power.

and directly connecting a 3.7 v battery is overloading the ESP32.  It is expecting and requiring a powersupply to provide 3.3v and 500mA of power.

If the OP actually had a single cell Lithium Ion Polymer (liPo) battery his battery voltage was not 3.7v.  It should be between 4.2v and 3.6v.  Either way, Direct connect is a fail, and connection through a cp2102 won't work.

Chuck.

lyusupov, 2018-07-04:
@volkerpetersen : see if [this workaround](https://github.com/lyusupov/ESP32-NODEMCU-ADAPTER/issues/3) can help you.

closed 2019-08-15 by stale[bot]