espressif/arduino-esp32 #1380: esp_log_set_vprintf does not work ?

psykokwak-com, 2018-05-04:
## Environment

- Development Kit: ESP32-Wrover
- Core (if using chip or module): ESP32-Wrover-I
- IDF version : 1424b6d
- Development Env:		Visual Studio Micro (Arduino environment)
- Operating System:		Windows
- Power Supply:         External 3.3v

## Problem Description

I try to redirect logging text but it does not work.

### Expected Behavior

"After" should not be displayed and replaced by "Nop nop nop".

### Code to reproduce this issue

```cpp
static int debugOutputHandler(const char *v, va_list l)
{
  return Serial.println("Nop nop nop");
}

void setup() {
  Serial.begin(115200);
  Serial.println("OK, lets see :)");
  log_e("Before");
  esp_log_set_vprintf(debugOutputHandler);
  log_e("After");

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13268
entry 0x40078a94
OK, lets see :)
[E][sketch_may04a.ino:9] setup(): Before
[E][sketch_may04a.ino:11] setup(): After

```

psykokwak-com, 2018-05-04:
These Macro are redefined in esp32-hal-log.h and they redirect to log_x.
So, with these ESP_LOGx the result is the same. (I just tried it to confirm)

everslick, 2018-05-05:
esp_log_level_set("*", ESP_LOG_VERBOSE);

psykokwak-com, 2018-05-07:
Change nothing.

```cpp
static int debugOutputHandler(const char *v, va_list l)
{
  return printf("coool");
}

esp_log_level_set("*", ESP_LOG_VERBOSE);
ESP_LOGI("my", "Before redirect");
esp_log_set_vprintf(debugOutputHandler);
ESP_LOGI("my", "after redirect");
```

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13268
entry 0x40078a94
[I][myfile.cpp:529] setup(): Before redirect
[I][myfile.cpp:531] setup(): after redirect
```

charlie71, 2018-10-15:
Any news on this issue?
It still exists.
thx

az-z, 2020-08-07:
@psykokwak-com  - did you figure out how to use it or is it indeed broken ?

zekageri, 2020-09-17:
I still can't use it. :(

stevenlayne12, 2021-10-30:
I am still seeing this on my side. Cant seem to find anything around it :(

VojtechBartoska, 2022-04-06:
Hello guys, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-21:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable.

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-21 by VojtechBartoska

letmeknowFR, 2023-09-03:
I had the same issue. I have changed all my command ESP_LOGI by esp_log_write and it works.

Maaajaaa, 2023-11-23:
I would like to add to this that the  `-DUSE_ESP_IDF_LOG` buildflag must be set