espressif/arduino-esp32 #1385: state of yield()

trycoon, 2018-05-06:
### Hardware:
Board:							NodeMCU ESP32
Core Installation/update date:			xxxx
IDE name:						Platform.io
Flash Frequency:					xxxx
Upload Speed:					xxxx

Hi!
When I started to develop on a Arduino everything was singlethreaded, and I was in full control of the CPU-cycles (as far as I know). Then I moved up to the ESP8266, with more of everything, including WiFi. The ESP8266 was still singlethreaded and I was required to call upon the yield()-function when doing heavy computations to allow the system to do some housekeeping (including keeping the WiFi connection going). Failing to call upon yeild() or sleep() in a long running loop() would result in a unstable system and flaky WiFi.

I'm now moving on to the ESP32 platform and my question is if I'm still required to call upon yield() when doing long running computations? Since the ESP32 platform is running a real OS (FreeRTOS) with multithread support on a dual core CPU, maybe all timecritical houskepping is already built into the system using some kind of maintenance thread?

Thanks for any clarification, and keep up the good work!
// Henrik

closed 2018-05-11 by trycoon

pipi61, 2018-05-11:
[http://lmgtfy.com/?q=esp32+yield](url)
[https://hackaday.com/2016/10/31/whats-new-esp-32-testing-the-arduino-esp32-library/](url)
igrr says:
October 31, 2016 at 9:20 am
In ESP32 Arduino core we run the “sketch” (setup and loop and everything they call) on CPU1, while the WiFi stack is running on CPU0. This allows messing with interrupts at will, and running more or less timing critical code. No more yield, yay! Interrupts attached through Arduino APIs also run on CPU1.

If you want to run something on CPU0 from Arduino, you have to use FreeRTOS function to start a new task, with task affinity set to CPU0. That’s it! Then use some FreeRTOS synchronization primitives to communicate with the new task.

trycoon, 2018-05-12:
Thanks @pipi61, that was good news. I can finally remove all bloated "yields" from my code.

lbernstone, 2018-05-12:
Please close the issue!