espressif/arduino-esp32 #1389: bluetooth serial crash

moijerem, 2018-05-06:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			07/05/2018?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Hello, sorry for my English I go through an online translator.
our project and consists of:
  ds3231
lm75a
pca9685

I will try to be as clear as possible; our code works perfectly when communicating via ftdi.
On the other hand when one communicates via the BT (classical)
it also works but after a while the esp crash and reboot without any apparent reason.
I mail you the integrality of the code, as well as a rar archive containing the whole;
I hope this bug is found.
Thank you for your precious time

https://jeremc.freeboxos.fr:41532/share/vQ5MVFE-7AVu0QyV/RECIFALX_ESP32_PCA_V1.002bt.rar

### Sketch:
RECIFALX_ESP32_PCA_V1.002bt.ino
`/* PAD RECIFALX */
#include "BluetoothSerial.h"
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif
BluetoothSerial SerialBT;

#include "Wire.h"
#include "EEPROM.h"
#include "ROM.h"
#include "VAR.h"
#include "MACROS.h"
#include "INA.h"
#include "MOON.h"
#include "DS3231.h"
#include "DRIVER.h"
#include "LM75.h"
#include "SYSTEM.h"
#include "REGISTER.h"
#include "FAN.h"
#include "LEDS.h"
#include "COM.h"

void setup() {
  INIT_REGISTRE();
  Serial.begin(115200);
  SerialBT.begin("ESP32_RECIFALX"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
  delay(1000);
  Serial.println(F("\r\n.: INITIALISATION PAD :."));
  delay(1000);
  eeprom.begin();
  Wire.begin();
  DRV.begin();
  RTC.begin();
  LM75.begin();
  Serial.print(F("> Test PERIFERIQUES ... "));
  if(!RTC.isPresent() || !LM75.isPresent() || !DRV.isPresent()) {
[... 2502 lines cut ...]
                "var xhr = new window.XMLHttpRequest();"
                "xhr.upload.addEventListener('progress', function(evt) {"
                    "if (evt.lengthComputable) {"
                        "var per = evt.loaded / evt.total;"
                        "$('#prg').html('progress: ' + Math.round(per*100) + '%');"
                    "}"
               "}, false);"
               "return xhr;"
            "},"                                
            "success:function(d, s) {"    
                "console.log('success!')"
           "},"
            "error: function (a, b, c) {"
            "}"
          "});"
"});"
"</script>";

#endif
`

### Debug Messages:
```

```

copercini, 2018-05-07:
Well, you can start enabling the logs on arduino IDE:
[image]

moijerem, 2018-05-07:
Hello, thank you for this info. So I have to try.
So my place came the tram for several minutes then suddenly

`[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:83] esp_spp_cb(): ESP_SPP_WRITE_EVT
ASSERT_PARAM(32768 0), in rwbt.c at line 265
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400845ad  PS      : 0x00060034  A0      : 0x800879f8  A1      : 0x3ffc05b0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x800845ad  A9      : 0x3ffc0590  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400844e5  LEND    : 0x400844ec  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4015762a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400845ad

Backtrace: 0x400845ad:0x3ffc05b0 0x400879f5:0x3ffc05d0 0x40087c8f:0x3ffc05f0 0x400816cd:0x3ffc0610 0x40157627:0x00000000

Core 1 register dump:
PC      : 0x4015762a  PS      : 0x00060b34  A0      : 0x8008b820  A1      : 0x3ffcf550  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffcf080  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc67cc  A9      : 0x3ffc6790  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4015762a:0x3ffcf550 0x4008b81d:0x3ffcf570

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
The device started, now you can pair it with bluetooth!
[I][BluetoothSerial.cpp:47] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:62] esp_spp_cb(): ESP_SPP_START_EVT

.: INITIALISATION PAD :.
> Test EEPROM ... OK
> Test PERIFERIQUES ... OK
TERMINE AVEC SUCCES !`

moijerem, 2018-05-09:
Hello, I hope to have posted the requested items. could you confirm me because I can not solve my problem. thank you very much. (sorry I'm going through a translator)

moijerem, 2018-05-12:
Hello, I voluntarily overtaxed identical tram otherwise it would have been long
Thanks again for your analysis

`> Test EEPROM ... OK
> Test PERIFERIQUES ... OK
TERMINE AVEC SUCCES !
free heap: 114124
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
..........................................................................

free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 110148
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
free heap: 110308
[... 267 lines cut ...]
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 112944
free heap: 113556
free heap: 113556
free heap: 113556
free heap: 113556
free heap: 113556
..............................................................................;
free heap: 113556
free heap: 113556
free heap: 113556
free heap: 113556

moijerem, 2018-05-12:
if I do not connect to the BT all rolls the free heap stays a
free heap: 114100
once connected the crash comes, and it can not be reconnected after.
I have to cut off the power

here is an example of the tram that I send in BT
{
<V>ESP-V1.002<V>
<MD>1<MD>
<BD>127<BD>
<H>11:06:36<H>
<A>391<A>
<B>391<B>
<C>391<C>
<D>391<D>
<T>20,37<T>
<F>0<F>
<R>0<R>
<AL>0<AL>
<M>0<M>
<N>0<N>
<AC>0<AC>
<DA>12/05/18<DA>
<PBA>255<PBA>
<PBB>255<PBB>
<PBC>255<PBC>
<PBD>255<PBD>
<PA>0,0<PA>
<PV>2,6<PV>
<PW>0<PW>
<MP>7<MP>
}

moijerem, 2018-05-12:
I add your code to my version without the BT l esp does not crash, the free heap remains permanently at 141424

moijerem, 2018-05-13:
hello, it worked for a while and then it crashed
since a simple reconnection do redoot the esp
As we can see in the example if you like

`TERMINE AVEC SUCCES !
free heap: 114128
free heap: 114104
free heap: 114104
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
The device started, now you can pair it with bluetooth!

.: INITIALISATION PAD :.
> Test EEPROM ... OK
> Test PERIFERIQUES ... OK
TERMINE AVEC SUCCES !
free heap: 114128
free heap: 114104
free heap: 114104
free heap: 114104
E (17799) BT: l2cu_adjust_out_mps bad packet size: 0  will use MPS: 0
free heap: 109892
free heap: 109892
ASSERT_PARAM(32768 0), in rwbt.c at line 265
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400845b2  PS      : 0x00060034  A0      : 0x800879f8  A1      : 0x3ffc05b0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x800845ad  A9      : 0x3ffc0590  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400844e5  LEND    : 0x400844ec  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x401570ba  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400845b2

Backtrace: 0x400845b2:0x3ffc05b0 0x400879f5:0x3ffc05d0 0x40087c8f:0x3ffc05f0 0x400816cd:0x3ffc0610 0x401570b7:0x00000000

Core 1 register dump:
PC      : 0x401570ba  PS      : 0x00060b34  A0      : 0x8008b820  A1      : 0x3ffcf510  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffcf040  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc678c  A9      : 0x3ffc6750  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008c8a8  A13     : 0x3ffcd060  
A14     : 0x00000000  A15     : 0x3ffcf1c0  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x401570ba:0x3ffcf510 0x4008b81d:0x3ffcf530

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
The device started, now you can pair it with bluetooth!

.: INITIALISATION PAD :.
> Test EEPROM ... OK
> Test PERIFERIQUES ... OK
TERMINE AVEC SUCCES !
free heap: 114128
free heap: 114100
free heap: 114100
free heap: 114100
free heap: 114100
`

moijerem, 2018-05-15:
I get the impression that the library serialtoserialBt sometimes monopolizes too much the esp which makes it reboot. is this related to the yield () function?

closed 2019-08-15 by stale[bot]