espressif/arduino-esp32 #1390: IntegerDivideByZero Error in map() function

wilhelmzeuschner, 2018-05-07:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation/update date:			10th Dec. 2017
IDE name:							Arduino IDE v.1.8.5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:

> Guru Meditation Error of type IntegerDivideByZero occurred on core  1. Exception was unhandled.

The problem occurs, whenever  (in_max - in_min) is equal to 0 or in other words, when in_max is equal to in_min in:
```cpp
long map(long x, long in_min, long in_max, long out_min, long out_max)
{
    return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}
```
When running the "exact same" code on an Arduino Nano there is no such error.
Instead, the value is mapped correctly.

Obviously one cannot divide any number by zero but this issue doesn't seem to exist on the AtMega328p. 
I was able to avoid the error by checking whether (in_max - in_min) is equal to 0 and adjusting one of the values if necessary.

### Sketch:
```cpp
void setup() {
  Serial.begin(115200);
  Serial.println("Divide by Zero Error in map-function:");
}

void loop() {
  float value_to_be_mapped = 9119;
  value_to_be_mapped = map(value_to_be_mapped, 9119, 9119, -360, 360);
  Serial.println(value_to_be_mapped);
  delay(1000);
}
```

### Debug Messages:
Guru Error: _0x400e5297: map(long, long, long, long, long) at E:\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WMath.cpp line 57_

```
Sketch uses 154806 bytes (11%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11140 bytes (3%) of dynamic memory, leaving 283772 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.1
Connecting........_____....._
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 11120 bytes to 7467...

Writing at 0x00001000... (100 %)
Wrote 11120 bytes (7467 compressed) at 0x00001000 in 0.1 seconds (effective 917.1 kbit/s)...
Hash of data verified.
Compressed 155952 bytes to 86337...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 155952 bytes (86337 compressed) at 0x00010000 in 1.3 seconds (effective 928.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 1638.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting... 
```

[MapDividebyZero.zip](https://github.com/espressif/arduino-esp32/files/1979114/MapDividebyZero.zip)

ARPavel22, 2018-10-17:
I have the same problem

lbernstone, 2018-10-17:
What does arduino return if they are equal?

wilhelmzeuschner, 2019-02-24:
@lbernstone this issue for the ESP8266 explains the reasons behind this behaviour and @drmpf provided a fix: 
```
long div = (in_max - in_min);
if (div == 0) div = 1;
long scale = (out_max - out_min) / div ; // do this first to avoid avoidable overflows
// from (x - in_min) * (out_max - out_min)
return (x - in_min) * scale + out_min;
```

closed 2019-02-24 by wilhelmzeuschner