espressif/arduino-esp32 #1391: HTTPS OTA fails

vshymanskyy, 2018-05-07:
```cpp
  DEBUG_PRINT(String("Firmware update URL: ") + overTheAirURL);

  HTTPClient http;
  http.begin(overTheAirURL);

  int httpCode = http.GET();
  if (httpCode != HTTP_CODE_OK) {
    DEBUG_PRINT("HTTP response should be 200");
    return;
  }
  int contentLength = http.getSize();
  if (contentLength <= 0) {
    DEBUG_PRINT("Content-Length not defined");
    return;
  }

  bool canBegin = Update.begin(contentLength);
  if (!canBegin) {
    DEBUG_PRINT("Not enough space to begin OTA");
    return;
  }

  Client& client = http.getStream();
  int written = Update.writeStream(client);
  if (written != contentLength) {
    DEBUG_PRINT(String("OTA written ") + written + " / " + contentLength + " bytes");
    return;
  }

  if (!Update.end()) {
    DEBUG_PRINT("Error #" + String(Update.getError()));
    return;
  }

  if (!Update.isFinished()) {
    DEBUG_PRINT("Update failed.");
    return;
  }

  DEBUG_PRINT("Update successfully completed. Rebooting.");
  ESP.restart();
```

When `overTheAirURL` is plain HTTP URL, the update works. When it is a HTTPS endpoint, it fails with `OTA written 0 / 786272 bytes` message. `contentLength` is correct, which hints that SSL auth and HTTP headers are processed correctly.

lbernstone, 2018-05-07:
Can you set core debug level to verbose and see if it gives you any indication of the problem?

vshymanskyy, 2018-05-08:
@lbernstone sorry completely forgot about this log. Here it is:

```
[9579] Firmware update URL: https://test.blynk.cc/static/ota/FUp_4910308060665836127_upload.bin
[V][HTTPClient.cpp:140] beginInternal(): url: https://test.blynk.cc/static/ota/FUp_4910308060665836127_upload.bin
[W][HTTPClient.cpp:152] beginInternal(): unexpected protocol: https, expected http
[V][HTTPClient.cpp:140] beginInternal(): url: https://test.blynk.cc/static/ota/FUp_4910308060665836127_upload.bin
[D][HTTPClient.cpp:181] beginInternal(): host: test.blynk.cc port: 443 url: /static/ota/FUp_4910308060665836127_upload.bin
[V][ssl_client.cpp:48] start_ssl_client(): Free heap before TLS 150524
[V][ssl_client.cpp:50] start_ssl_client(): Starting socket
[V][ssl_client.cpp:86] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:95] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:119] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:143] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:158] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:177] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:186] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:201] start_ssl_client(): Free heap after TLS 109816
[D][HTTPClient.cpp:850] connect():  connected to test.blynk.cc:443
[V][ssl_client.cpp:240] send_ssl_data(): Writing HTTP request...
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'content-length: 797792'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'content-type: application/octet-stream'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'access-control-allow-origin: *'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'date: Tue, 08 May 2018 10:45:31 GMT'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'expires: Tue, 08 May 2018 10:46:31 GMT'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'cache-control: private, max-age=60'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'last-modified: Tue, 08 May 2018 10:45:30 GMT'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'connection: close'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:968] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:971] handleHeaderResponse(): size: 797792
[D][Updater.cpp:129] begin(): OTA Partition: app1
[W][HTTPClient.cpp:568] getStream(): getStream: not connected
[10734] OTA written 0 / 797792 bytes

```

For some reason, **getStream(): getStream: not connected** ;)

copercini, 2018-05-09:
This problem is caused because `getStream ` return a WiFiClient object and not a WiFiClientSecure as expected, here is some more detail of this problem: https://github.com/espressif/arduino-esp32/issues/566

closed 2019-08-15 by stale[bot]