espressif/arduino-esp32 #1392: Read and customize TCP_WND, TCP_MSS

starlino, 2018-05-07:
I am porting some code from ESP8266 to ESP32 core. 
What is the proper way to read/change  TCP_WND, TCP_MSS from the code ?
The constants TCP_WND, TCP_MSS no longer work on ESP32, but worked fine on ESP8266.
It would be nice if these can be brought back for compatibility.
Thank you.

lbernstone, 2018-05-07:
If you put lwip/opt.h in your sketch folder, you should be able to set the TCP_WND in that file.

starlino, 2018-05-08:
Looking at opt.h I see TCP_WND are conditioned by not being defined already
#ifndef TCP_WND
#define TCP_WND(pcb)                         (4 * TCP_MSS)
#endif

TCP_WND is now a function of 'pcb' , while TCP_MSS is not , what is the logic here ???
why not just make it 
#define TCP_WND   4*TCP_MSS   ??

like it was before in esp8266 core

starlino, 2018-05-08:
@lbernstone  sorry , but your solution does not work:
```
#include "lwip/opt.h"
#include <WiFi.h>
//static_assert(  536 == TCP_MSS   ,"536 != TCP_MSS ");
void setup() {
  printf("TCP_MSS:%u\n",TCP_MSS);
}
void loop() {
  
  
}

```

outputs:

`TCP_MSS:1436`

lbernstone, 2018-05-08:
Not include.  Physically copy the opt.h into your sketch directory and modify it there.

starlino, 2018-05-08:
@lbernstone are you sure you don't mean **lwipopts.h**  ? Because this code worked:

> #include "lwipopts.h"
> #include <WiFi.h>
> static_assert(  536 == TCP_MSS   ,"536 != TCP_MSS ");
> void setup() {
> 
>   printf("TCP_MSS:%u\n",TCP_MSS);
>   
> }
> void loop() {
>   
>   
> }
>

lbernstone, 2018-05-14:
If you have a solution, please close the issue.

starlino, 2018-05-14:
This is not 100% solved, can we hold this for couple of days so I can do more tests with Wireshark.  I suspect even if I change TCP_MSS define, the real value is hard-coded in the pre-compiled lib,   I have to figure out how to change  TCP_WND as well.

On a related question - if I customized esp-idf , how do a make Arduino IDE to use my esp-idf and not the default supplied one ? (This way I can just use  sdkconfig , to set all these settings) ?

copercini, 2018-05-14:
You can use [Arduino as IDF component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) and set these options on IDF `make menuconfig`

copercini, 2018-05-14:
ok, got it =)

You can set up Arduino as IDF component just for you, then configure the TCP things on `make menuconfig` , doing a `make build` it will compile all the core with your configs and generate a file with the new TCP settings on `project_name\build\lwip\liblwip.a` get it and distribute in your github project. 

To work on everyone's computer with their settings, just tell the people to grab your version of `liblwip.a` and replace it here: `https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/lib` in their Arduino installation

lbernstone, 2018-05-14:
I think I can do even better.  It is not the headers you need to drop in to overrride, but the c files themselves.  Download the esp-idf (https://github.com/espressif/esp-idf) and find the .c file(s) where those defines are used.  Place THOSE in the sketch folder, and add defines to them for the values you want.  I just tested this with sntp.c (which is much easier to see your changes), and got it to refresh every minute instead of every hour.  Sorry for forgetting how that override works...

closed 2018-08-15 by starlino