espressif/arduino-esp32 #1395: SHT31 not working over I2C with ESP32 and Stickbreaker's library.

JacoFourie, 2018-05-08:
I got this RH probe

https://www.aliexpress.com/item/SHT30-SHT31-AM23XX-single-bus-protection-temperature-and-humidity-sensor-probe-soil-moisture-sensor-5m-cable/32850764307.html

I can not get it to work with the ESP32. It works with the Arduino Mega on 5V not 3V. The supplier says because the Arduino is 5V logic it wil only work on 5v with the Arduino. As the ESP32 is 3v it should work on 3v.

Here is the test code.

````
/*************************************************** 
  This is an example for the SHT31-D Humidity & Temp Sensor

  Designed specifically to work with the SHT31-D sensor from Adafruit
  ----> https://www.adafruit.com/products/2857

  These sensors use I2C to communicate, 2 pins are required to  
  interface
 ****************************************************/
 
#include <Arduino.h>
#include <Wire.h>
#include "Adafruit_SHT31.h"

Adafruit_SHT31 sht31 = Adafruit_SHT31();

void setup() {
  Serial.begin(115200);

  Serial.println(SDA);
  Serial.println(SCL);
  
  while (!Serial)
    delay(10);     // will pause Zero, Leonardo, etc until serial console opens

  Serial.println("SHT31 test");
  if (! sht31.begin(0x44)) {   // Set to 0x45 for alternate i2c addr
    Serial.println("Couldn't find SHT31");
    //while (1) delay(1); }
  }
  else {
    Serial.println("Found SHT31 !!"); 
  }
}

void loop() {
  float t = sht31.readTemperature();
  float h = sht31.readHumidity();

  if (! isnan(t)) {  // check if 'is not a number'
    Serial.print("Temp *C = "); Serial.println(t);
  } else { 
    Serial.println("Failed to read temperature");
  }
  
  if (! isnan(h)) {  // check if 'is not a number'
    Serial.print("Hum. % = "); Serial.println(h);
  } else { 
    Serial.println("Failed to read humidity");
  }
  Serial.println();
  //delay(1000);
}
````

I get a [E][esp32-hal-i2c.c:1127] i2cProcQueue(): I2C exitCode=0x112 error from time to time on the output.

````
[E][esp32-hal-i2c.c:1127] i2cProcQueue(): I2C exitCode=0x112
Failed to read temperature
Failed to read humidity

Failed to read temperature
Failed to read humidity

Failed to read temperature
Failed to read humidity

````

stickbreaker, 2018-05-09:
@JacoFourie The SHT3x sensors are slow, they can sample at a maximum of 10hz, depending on how you have it programmed, it can pause communications by stretching SCL while it is processing a reading.  My Wire library has a default timeout of 50ms. So you are seeing timeouts.  Increase the default timeout with `Wire.setTimeOut(1000);` This will set the timeout to 1 second, instead of 50ms.

Chuck.

JacoFourie, 2018-05-09:
@stickbreaker Chuck tanks for the reply. I added the timeout but I still see the same. Here is the test code.

````
/*************************************************** 
  This is an example for the SHT31-D Humidity & Temp Sensor

  Designed specifically to work with the SHT31-D sensor from Adafruit
  ----> https://www.adafruit.com/products/2857

  These sensors use I2C to communicate, 2 pins are required to  
  interface
 ****************************************************/
 
#include <Arduino.h>
#include <Wire.h>
#include "Adafruit_SHT31.h"

Adafruit_SHT31 sht31 = Adafruit_SHT31();

void setup() {
  Serial.begin(115200);

  Serial.println(SDA);
  Serial.println(SCL);
  
  while (!Serial)
    delay(10);     // will pause Zero, Leonardo, etc until serial console opens

  Serial.println("SHT31 test");
  if (! sht31.begin(0x44)) {   // Set to 0x45 for alternate i2c addr
    Serial.println("Couldn't find SHT31");
    //while (1) delay(1); }
  }
  else {
    Serial.println("Found SHT31 !!"); 
  }
  
  Wire.setTimeOut(1000);
}

void loop() {
  float t = sht31.readTemperature();
  float h = sht31.readHumidity();

  if (! isnan(t)) {  // check if 'is not a number'
    Serial.print("Temp *C = "); Serial.println(t);
  } else { 
    Serial.println("Failed to read temperature");
  }
  
  if (! isnan(h)) {  // check if 'is not a number'
    Serial.print("Hum. % = "); Serial.println(h);
  } else { 
    Serial.println("Failed to read humidity");
  }
  Serial.println();
  //delay(1000);
}
````

I also found this Wemos library for the SHT31 

https://github.com/wemos/WEMOS_SHT3x_Arduino_Library

This code also works on the Arduino but not on the ESP32. 

````
#include <WEMOS_SHT3X.h>

SHT3X sht30(0x44);

void setup() {

  Serial.begin(115200);
  Wire.setTimeOut(1000);
}

void loop() {

  if(sht30.get()==0){
    Serial.print("Temperature in Celsius : ");
    Serial.println(sht30.cTemp);
    Serial.print("Temperature in Fahrenheit : ");
    Serial.println(sht30.fTemp);
    Serial.print("Relative Humidity : ");
    Serial.println(sht30.humidity);
    Serial.println();
  }
  else
  {
    Serial.println("Error!");
  }
  delay(1000);

}
````

JacoFourie, 2018-05-09:
I also ran this I@C scanner. 

````
#include <Wire.h>

void setup()
{
  //Wire.setTimeOut(1000);
  Wire.begin();

  Serial.begin(115200);
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
````

This is the output with just the sensor connected. It changes all the time.

````
Scanning...
I2C device found at address 0x06  !
I2C device found at address 0x11  !
I2C device found at address 0x21  !
I2C device found at address 0x3E  !
I2C device found at address 0x40  !
I2C device found at address 0x47  !
I2C device found at address 0x61  !
I2C device found at address 0x69  !
I2C device found at address 0x73  !
I2C device found at address 0x7C  !
done

Scanning...
I2C device found at address 0x04  !
I2C device found at address 0x0A  !
I2C device found at address 0x17  !
I2C device found at address 0x25  !
I2C device found at address 0x26  !
I2C device found at address 0x39  !
I2C device found at address 0x44  !
I2C device found at address 0x55  !
I2C device found at address 0x60  !
I2C device found at address 0x61  !
I2C device found at address 0x66  !
I2C device found at address 0x68  !
I2C device found at address 0x74  !
I2C device found at address 0x79  !
I2C device found at address 0x7A  !
done

Scanning...
I2C device found at address 0x08  !
I2C device found at address 0x30  !
I2C device found at address 0x39  !
I2C device found at address 0x44  !
I2C device found at address 0x55  !
I2C device found at address 0x6E  !
I2C device found at address 0x79  !
I2C device found at address 0x7A  !
done

And if I unplug the sensor I get this.

````
Scanning...
No I2C devices found
````

So not sure why it is getting all those addresses if only the one sensor is hooked up
````

stickbreaker, 2018-05-09:
@JacoFourie What is you electrical circuit?  Do you have pullups from SDA and SCL to 3.3V?  The Arduino's have strong enough pullups built in to the AVR chips that the I2C bus can usually function without additional pullups, BUT, the ESP32's pullups are too weak by them selves.  So you need to add pullups to SDA and SCL I would recommend you start with 2.4k Ohm.

Add these pullups and rerun the Scan.

Chuck.

JacoFourie, 2018-05-09:
DUDE !!!!!! You are a rock star. I have been batteling with the RH sensors for months now. The sensors I have stated they have resistors already in them, and the fact that they worked on the Arduino I did not bother with external resistors.

So it now works sort off. I used 2K resistors as that is what I have. Not 2.4K. If I use 10K or 4.7K then I get no comms. Should I use less to get rid of those last errors ?

````
Temp *C = 26.44
Failed to read humidity

Failed to read temperature
Hum. % = 42.76

Temp *C = 26.43
Hum. % = 42.76

Temp *C = 26.47
Hum. % = 42.77

Temp *C = 26.47
Hum. % = 42.78

[E][esp32-hal-i2c.c:1127] i2cProcQueue(): I2C exitCode=0x112
Temp *C = 26.47
Hum. % = 42.78

Temp *C = 26.47
Hum. % = 42.78

Temp *C = 26.47
Hum. % = 42.76

Failed to read temperature
Hum. % = 42.75

Failed to read temperature
Failed to read humidity

Failed to read temperature
Hum. % = 42.73

Temp *C = 26.46
Failed to read humidity

Temp *C = 26.48
Hum. % = 42.74

Failed to read temperature
````

stickbreaker, 2018-05-10:
@JacoFourie how long are your i2c wires? i2c was designed to work between sensors on the same pcb, so the max length was spec'd at 1m.  If your bus is longer it increases the capacitance which decreases the maximum frequency the bus can operate.  

I compiled a WiKi entry about pullups here: [Pullup Value selection](https://github.com/stickbreaker/arduino-esp32/wiki/Pullup-Values)

You can decrease the pullup values down to about 1k Ohm, depending on the max sink current your sensor can provide.  

You might also try adjusting the bus clock rate.  You sensor is slow, so 400khz bus rate is kind of silly.  Use `Wire.setClock(50000);` for 50khz.  I have used 10k for long lines 15m.  I think the minimum supported by hardware is about 3khz 

Chuck.

JacoFourie, 2018-05-10:
@stickbreaker are you telling me that I can not have long wires when using I2C ? I need about 13 meter cable. I am busy converting a control unit I build using Arduino and Raspberry Pi to run everything on the ESP32 https://www.youtube.com/watch?v=jYnhPLxoM1k  

But I need the RH probe cable to be around 13 meters. The SHT10 sensors I use I have 30 meter cables and they work fine.  I would still use those sensors but the have been discontinued and the I2C will be the only ones you get in future.

stickbreaker, 2018-05-10:
@JacoFourie To get reliable communication at that long of a range you are going to have to do some work.
Here is a good discussion on long I2C busses [long i2c Bus](https://www.hackster.io/chipmc/extend-the-reach-of-your-i2c-sensor-simply-and-inexpensively-ea6b53)

In this article, he uses dedicated bus drivers and can remote sensors up to 30m.  You might be able to adjust the pullup and clock rate to compensate for your long wires,  I would have to measure the wave forms before I would guarantee it. 

Chuck.

JacoFourie, 2018-05-12:
@stickbreaker I am using 1K resistors now and it seems a lot better. I still get an error here or there but my code is handling it.

What does this mean ?

````
[E][esp32-hal-i2c.c:1127] i2cProcQueue(): I2C exitCode=0x112
````

Thanks for all your help. My electronics side is not that good but I write mean code on the functional side lol.

JacoFourie, 2018-05-13:
@stickbreaker if I slow down the bus how will that effect other items on the bus like the OLED screen that is also on the same bus ?

stickbreaker, 2018-05-13:
@JacoFourie All of the devices are effected by the long bus, the clock rate is limited by the capacitance of of bus.  The capacitance does not change when communicating with different slave devices.  Different slaves may handle the extra capacitance better, but the electrical signal will still be degraded.  
Since you have multiple devices, some near, some far.  Why don't you split them up, use both hardware I2C busses?  That way all of your near devices could operate at full speed, and you could slow down your far devices until they work properly.

To use the second I2C hardware:
```c++
TwoWire Wire1(1);

void setup(){
Wire1.begin(SDA_PIN,SCL_PIN,25000); // configure second i2c bus on pins(SDA_PIN,SCL_PIN) at 25khz
}
```
 
Chuck.

stickbreaker, 2018-05-13:
only two, to connect multiple sensors with the same address you need an i2c hub check out [PCA9548A](http://www.ti.com/product/tca9548a) 1:8 switch, with this chip you can break your i2c bus into multiple segments, each segment can be connected/disconnected by software, and if a sensor hangs the bus you and use a pin to reset the switch to release all segments and recover.  I use them.  They also do automatic 3.3v to 5v conversions.

Chuck.

closed 2018-06-18 by JacoFourie