espressif/arduino-esp32 #1398: How do I decrease the size of my firmware?

rammak, 2018-05-10:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		10 May 2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600

### Description:
I'm working on a project where I want the smallest firmware size possible for ESP32. The sketch uses BLE so the firmware size goes to about 1.2MB but even with just a simple blink sketch the firmware size is at  161565 bytes. Is there any way I can reduce the filesize from Arduino IDE? I don't know if ESP-IDF produces smaller outputs but since the whole project is written in Arduino I would prefer not to switch and take the minimum hit in terms of extra size.
 
I tried opening the output Blink.ino.bin file and I see that a large part of the file is made up of ASCII data like

```
   system_api  E (%d) %s: Base MAC address is NULL
    E (%d) %s: Base MAC address from BLK3 of EFUSE version error, version = %d
 E (%d) %s: Base MAC address from BLK3 of EFUSE CRC error, efuse_crc = 0x%02x; calc_crc = 0x%02x
    E (%d) %s: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0x%02x; calc_crc = 0x%02x
    E (%d) %s: mac address param is NULL
   E (%d) %s: mac type is incorrect
   master  v3.1-dev-661-gf586f5e6  uart_tx_wait_idle   Task watchdog got triggered. The following tasks did not reset the watchdog in time:
    - %s (%s)
 CPU %d: %s
 Aborting.
  esp_intr_alloc(ETS_TG0_WDT_LEVEL_INTR_SOURCE, 0, task_wdt_isr, NULL, &twdt_config->intr_handle) /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./task_wdt.c  esp_intr_free(twdt_config->intr_handle) esp_register_freertos_idle_hook_for_cpu(idle_hook_cb, i)    esp_task_wdt_add    esp_task_wdt_init   
Brownout detector was triggered
```

On the other sketch where I use BLE, almost 1/3rd of the file consists of such data. I wonder if this data is actually used by the sketch. Any recommendations on how to lower the size of the firmware?

### Sketch:
```cpp
void setup() {
  pinMode(21, OUTPUT);
}

void loop() {
  digitalWrite(21, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(21, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90

```

ArTyGill, 2018-05-18:
1. You can change the partition table for your device.
2. Discard the use of "C ++" in favor of the pure language "C"

beegee-tokyo, 2018-05-19:
Disabling debug complete should remove most of the ASCII text you see in the .bin file. They are debug messages.
However using BLE or Bluetooth significantly increases app size. I guess we have to wait for more optimizations in the ESP-IDF to make it smaller. I don't think this is an issue of arduino-esp32.

rammak, 2018-05-25:
1. I don't know how changing the partition table can decrease the app size.
2. If I use pure C then I guess I have to rewrite most libraries because they are Arduino specific so in that case I guess it would be better to use ESP-IDF directly.
3. Debug was already disabled in my case. 

I guess there's no way to decrease the size until they optimize ESP-IDF.

closed 2018-05-25 by rammak

ArTyGill, 2018-05-25:
Yes, changing the partition table will not change the size of your app, but the standard partition table limits the application size to 1.3 MB.
In my project, under the app, I made a section of 3 MB, and a section with an OTA update of 0.5 MB.
But after I ran into the problem of lack of RAM. Wi-Fi and bluetooth use pretty much all of the free heap, but I suspect that both C ++ and Arduino modules use no less the a free heap.
The solution to the problem is two, use external SRAM, switch to eclipse.
Yes, it is undeniable on C to write hard, but if you have code in arduino, you can use the source library files to understand the functioning of the code.

I use Google translator, so write, if something is not clear.

ArTyGill, 2018-05-28:
There is an option to break the firmware into several parts, and write them gradually and then check on the CRC, but maybe then the encryption will not work.

ArTyGill, 2018-05-28:
You can try to download only changes in the firmware, as in the git.

ArTyGill, 2018-05-28:
NRF52 is also a very cool product, for him too there is arduino. But it's hard to encode, and the j-link programmer costs $100 for training purposes.