espressif/arduino-esp32 #1400: xtensa-esp32-elf-g++ file does not exist error

ThortheM, 2018-05-10:
### Hardware:
Board:	Wemos Lolin					
Core Installation/update date:		10.5.2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I simply cannot get the esp32 support on arduino IDE to work. I keep getting this error:

exec: "C:\\Users\\[myname]\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist
Error compiling for board WEMOS LOLIN32.
 
I believe I have done everything right, even did the procedure 3 times, just to avoid mistakes, but now, I just don't know what could've gone wrong.

Thank you for your help.

pgrlopes, 2018-05-10:
Hey man.

I'm not very experienced with this but I reinstalled the whole thing 3 times today until I got it to work so I might be able to help you out. Did you follow all the procedures of the tutorial (here for windows https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md)? Did you install the arduino ide in a directory other than the sketchbook directory (usually the sketchbook directory is in users\your_name\...)? I was having issues because I was installing the esp32 libraries on the arduino ide directory directly.

ThortheM, 2018-05-10:
Well my sketchbook is located at C:\Users\[myname]\Documents\Arduino and the arduino files are at C:\Program Files (x86)\Arduino. I installed the esp32 package in C:\Program Files (x86)\Arduino\hardware\espressif\esp32. I'm sure I followed all the steps. The update, running get.exe and so on. I also have all the files(even the C:\Users\[myname]\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ one) but I keep getting errors.

sansillusion, 2018-05-11:
It has to be in ...\Documents\Arduino\hardware NOT c:\Program Files\...

pgrlopes, 2018-05-11:
The esp32 package has to be in User\Documents\Arduino\hardware...

ThortheM, 2018-05-13:
Oh right, my mistake. Thank you, that fixed it!

lbernstone, 2018-05-14:
If this is solved, then please close the issue.

closed 2018-05-15 by ThortheM