espressif/arduino-esp32 #1401: Generate two PWM signals with 90 degree phase shift

Jeard, 2018-05-11:
### Hardware:
Board:							WEMOS LOLIN32
IDE name:							Arduino IDE

### Description:
I need to generate two PWM signals with 90 degree phase shift at 20KHz for controlling a switching power supply. I have looked in to this matter but wasn't able to figure it out. 

Thank you.

stickbreaker, 2018-05-11:
The only way I see is to set a 20kHz timer interrupt then count clock cycles for 10us in a critical section.
You can control the 90 degree phase, but I don't know how regular the 20kHz period would be.
Sounds like a hardware problem, not a software solution.

Chuck.

lbernstone, 2018-05-14:
Not only do I agree with @stickbreaker , but see that this is clearly not an issue with this code base.  You can ask on https://esp32.com, or just spend $0.50 on a PWM controller/comparator.  Please close the issue.

closed 2018-05-15 by Jeard