espressif/arduino-esp32 #1403: Analog read generates spurious interrup using GPIO35..31

rrobinet, 2018-05-11:
### Hardware:
Board:							WMOS ESP32 (WROOM32)
Core Installation/update date:			March 2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Hi,
Can someone explain me this behavior.
I am using physical switches to generate interrupts (actually pin GPIO25 & 26 for UP and DOWN counter)
Concurrently I am measuring an analog pin GPIO34 for instance (same issue with .. 33, 32 and 31).
Without the analog read, everything works find but once I start to read the analog signal, interrupts (UP and DOWN) are spontaneously generated.
The current solution is use another analog pin such as GPIO4 to read the analog signal.

Is there an explanation why GPIO36 to 31  generates such issue?

The following script is a simplified version of my working environment.
Tests can be simulated by changing the pin allocation of the BATPIN definition or to suppress the analog read in the loop function.

### Sketch:
```
#define TEMPDWNPIN  25            // Temperature DOWN push button pin
#define TEMPUPPIN   26            // Temperature UP push button pin
/* GPIO34 as Analog pin *
#define BATPIN      34             // Analog Pin for RTC Battery Level Monitoring*/
/*GPIO 4 as Analog pin */
#define BATPIN      4             // Analog Pin for RTC Battery Level Monitoring */
uint16_t BATLVL;                  // Used to save the Battery Level value for DS3231 RTC

/* ESP32 FREEOS INTERRUPT PROCEDURE HANDLING
********************************************/

boolean  TempUpFlag = false;             // Used to indicate that a push button interrupt temperature target UP is detected
boolean  TempDownFlag = false;           // Used to indicate that a push button interrupt temperature DOWN UP is detected

portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR TempMUpInterrupt() 
{
  portENTER_CRITICAL_ISR(&mux);
  TempUpFlag = true;
  portEXIT_CRITICAL_ISR(&mux);
}
void IRAM_ATTR TempMDownInterrupt() 
{
  portENTER_CRITICAL_ISR(&mux);
  TempDownFlag = true;
  portEXIT_CRITICAL_ISR(&mux);
}
/********************************************/

void setup() 
{
  Serial.begin (115200);
  /* Activate IO pins */
  pinMode(TEMPUPPIN, INPUT_PULLUP);        // Set pin Temperature UP as input
  pinMode(TEMPDWNPIN, INPUT_PULLUP);       // Set pin Temperature DOWN as input 
  attachInterrupt(digitalPinToInterrupt(TEMPUPPIN), TempMUpInterrupt, RISING);  // activate the interrupt procedure 
  attachInterrupt(digitalPinToInterrupt(TEMPDWNPIN), TempMDownInterrupt, RISING);// activate the interrupt procedure 
}
void loop() {

  if (TempUpFlag)                        // Activate push button Temperature target up action
  {
     portENTER_CRITICAL(&mux);
     TempUpFlag=false;                   // Reset the Flag
     portEXIT_CRITICAL(&mux);
     detachInterrupt(digitalPinToInterrupt(TEMPUPPIN));// Detach the interrupt for debouncing
     TempMUpIntCallback();              // Activate the temperature UP routine
  }
  if (TempDownFlag)                      // Activate push button Temperature target down action
  {
     portENTER_CRITICAL(&mux);
     TempDownFlag=false;                  // Reset the Flag
     portEXIT_CRITICAL(&mux);
     detachInterrupt(digitalPinToInterrupt(TEMPDWNPIN)); // Detach the interrupt for debouncing
     TempMDownIntCallback();              // Activate the temperature DOWN routine  
  }
  delay (1000);
/* Remve the following lines for no analog read */
  BATLVL = analogRead(BATPIN);              // Check RTC Battery Level 
  Serial.print("  Battery level is: "), Serial.println (BATLVL); //*/
}

void  TempMUpIntCallback()
{
  Serial.println ("UP interrupt");
  attachInterrupt(digitalPinToInterrupt(TEMPUPPIN), TempMUpInterrupt, RISING);  // Reactivate the interrupt procedure once done 
}

void  TempMDownIntCallback()
{
  Serial.println("DOWN interrupt");
  attachInterrupt(digitalPinToInterrupt(TEMPDWNPIN), TempMDownInterrupt, RISING);  // Reactivate the interrupt procedure once done 
}
```
### Debug Messages:
```
1. GPIO34 + Analog Read (no Interrupt activated) > analog read works fine but spurious interrupts
------------------------------------------------------------
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
  Battery level is: 562
UP interrupt
DOWN interrupt
  Battery level is: 560
UP interrupt
DOWN interrupt
  Battery level is: 552
UP interrupt
DOWN interrupt
.....

2. GPIO4 + Analog Read (no Interrupt activated) > Analog read works fine
------------------------------------------------------------------
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
  Battery level is: 493
  Battery level is: 511
  Battery level is: 526
```
```

3. GPIO4 + Analog Read (+ Down interrupt) Analog read works fine & interrupts works fine
----------------------------------------------------------------------------
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
  Battery level is: 560
  Battery level is: 561
  Battery level is: 560
  Battery level is: 572
  Battery level is: 4095
  Battery level is: 4095
  Battery level is: 562
DOWN interrupt
  Battery level is: 568
  Battery level is: 563
  Battery level is: 566
  Battery level is: 564
  Battery level is: 560
DOWN interrupt

```

rrobinet, 2018-05-12:
Hi,
In fact I have discovered another issue.

With ESP32 and WiFi active, only GPIO32 and above are working for analog read / write.
So using GPIO4 is not a solution, because if interrupts are working, I can't read analog value anymore.
Is it an alternative solution ?

Thanks
Robert

rrobinet, 2018-05-12:
OK
It looks that the workaround is to deactivate the interrupts before an analogue read (using GPIO32 and above)

```
>   detachInterrupt(digitalPinToInterrupt(TEMPDWNPIN)); // Detach the interrupt for debouncing
>   detachInterrupt(digitalPinToInterrupt(TEMPUPPIN));// Detach the interrupt for debouncing
>   BATLVL = analogRead(BATPIN);              // Check RTC Battery Level 
>   attachInterrupt(digitalPinToInterrupt(TEMPUPPIN), TempMUpInterrupt, RISING);  // Reactivate the interrupt procedure once done 
>   attachInterrupt(digitalPinToInterrupt(TEMPDWNPIN), TempMDownInterrupt, RISING);// activate the interrupt procedure *
```

lbernstone, 2018-05-12:
If you have found a solution, please close the issue.

rrobinet, 2018-05-14:
A workaround is not an explanation, but yes my problem is solved

closed 2018-05-14 by rrobinet

usmanshahid001, 2018-06-05:
hii i have question is falling or rising edge is working fine? i am having issue that it is operating just on change

rrobinet, 2018-06-11:
I am only using rising interrupts which are working fine.
Interrupts on Esp32 (running freeos) are differrent from  single processor MCU, because of the WiFi background activity. So please refrer to my example defining the esp32 freeos interrupt procedure handling (typically by limiting the interrupts activity by setting a flag, that should later be tested for further processing.)
Robert