espressif/arduino-esp32 #1405: RFID read error with RDM 6300 (125Mhz RFID Reader Modun)

hoangthaibmvl, 2018-05-12:
### Hardware:
Board:							ESP32 VN Iot
Core Installation/update date:			15/May/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
I try to read rfid wth rdm 6300 modun but i can't. When i read it have this code then reboot:

Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x40080e8f  PS      : 0x00060031  A0      : 0x80080f40  A1      : 0x3ffc05b0  
A2      : 0x00010000  A3      : 0x00000000  A4      : 0x00000010  A5      : 0x00000000  
A6      : 0x3fefffff  A7      : 0x00000000  A8      : 0x80080e26  A9      : 0x3ffc0590  
A10     : 0x00038d8d  A11     : 0x00000000  A12     : 0x8008578d  A13     : 0x3ffc6db0  
A14     : 0x00000003  A15     : 0x00060323  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40080e8f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d5274

Backtrace: 0x40080e8f:0x3ffc05b0 0x40080f3d:0x3ffc05d0 0x400810ad:0x3ffc05f0 0x400818c1:0x3ffc0610 0x400d480b:0x00000000

Rebooting...

I'm looking for solution. If you know how to fix it please tell me. Thank you so much
### Sketch:
```cpp

//Change the code below by your sketch

#include <SoftwareSerial.h>
#include <SeeedRFID.h>

#define RFID_RX_PIN 16
#define RFID_TX_PIN 17

// #define DEBUG
// #define TEST

SeeedRFID RFID(RFID_RX_PIN, RFID_TX_PIN);
RFIDdata tag;

void setup() {
	Serial.begin(115200);
	Serial.println("RFID Test..");
}

void loop() { 
	if(RFID.isAvailable()){
		tag = RFID.data();
		Serial.print("RFID card number: ");
		Serial.println(RFID.cardNumber());
#ifdef TEST
	Serial.print("RFID raw data: ");
	for(int i=0; i<tag.dataLen; i++){
	    Serial.print(tag.raw[i], HEX);
	    Serial.print('\t');
		}
#endif
	}
}

stickbreaker, 2018-05-12:
@hoangthaibmvl 
>Guru Meditation Error: Core 0 panic'ed (StoreProhibited)
. Exception was unhandled.

`StoreProhibited` is saying that an invalid memory address was specified to store a value.  Usually this means that an uninitialized pointer was used as a storage location.

You need to download and install me-no-dev's backtrace decoder and use it to find where in your code this error is generated. 

Chuck.

hoangthaibmvl, 2018-05-13:
@stickbreaker hi there is the decoder 

PC: 0x40080e8f: SoftwareSerial::rxRead() at C:\Users\hoang\Documents\Arduino\libraries\espsoftwareserial-master\SoftwareSerial.cpp line 265
EXCVADDR: 0x00000000

Decoding stack results
0x40080e8f: SoftwareSerial::rxRead() at C:\Users\hoang\Documents\Arduino\libraries\espsoftwareserial-master\SoftwareSerial.cpp line 265
0x40080f3d: sws_isr_16() at C:\Users\hoang\Documents\Arduino\libraries\espsoftwareserial-master\SoftwareSerial.cpp line 50
0x400810ad: __onPinInterrupt at C:\Users\hoang\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-gpio.c line 211
0x400d480b: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 62

stickbreaker, 2018-05-13:
@hoangthaibmvl Why are you using software serial? The esp32 has 3 hardware serials available?

Chuck.

hoangthaibmvl, 2018-05-13:
@stickbreaker hi i'm a beginner so i read in the internet it say use the software serial so i have no idea about that. You can tell me how to fix that

stickbreaker, 2018-05-13:
@hoangthaibmvl It looks like the Seeed library `#include <SeeedRFID.h>` is based on SoftwareSerial.

Are you using this library? [SeedStudio/RFID_Library](https://github.com/Seeed-Studio/RFID_Library)?

I just read through it, and it is very sloppy code. The `read()` function has this loop in it:
```c++
while (_rfidIO->available())
	{
		_data.raw[_data.dataLen++] = _rfidIO->read();
		delay(10);
	}
```
`_rfidIO()` is the SoftwareSerial object. `_data` is defined:
```+c++
struct RFIDdata
{
	int dataLen;
	byte chk;
	boolean valid;
	unsigned char raw[5];
};
```
So `raw[]` is a five element character array.  The `read()` function has no provision for handling more than 5 characters.  And the while loop will keep reading data from the serial object and adding it to the end of `raw[]`.  Also, if only 1 character is in the serial input buffer when `read()` is called, that single character will be discarded and the rest of the 5 character data packed will still be in the input buffer,  If another packet is received then the input buffer would have 9 characters in it the next time `read()` is called. `read()` then would store those next 4 bytes in [0..3] and the first character of the next packet in [5] then then rest of the second packet past the end of `raw[]`.  That is what is probably causing the explosion.

You need a better library.

Chuck.

hoangthaibmvl, 2018-05-13:
@stickbreaker I use the RDM6300. I search on google many time it just have the library based on SoftwareSerial. So I can find the better library. May be find another way to fix it.

hoangthaibmvl, 2018-05-13:
@stickbreaker Code have this error:  
rfid:63: error: 'MAX_LEN' was not declared in this scope

       if(len<MAX_LEN) {

stickbreaker, 2018-05-13:
Change `MAX_LEN` to `MAX_BUFFER`

Chuck.

hoangthaibmvl, 2018-05-13:
@stickbreaker  Here 

Look for START (0x02) found 0x02
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x44
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x33
 Look for START (0x02) found 0x31
 Look for START (0x02) found 0x39
 Look for START (0x02) found 0x43
 Look for START (0x02) found 0x44
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x37
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x03
 Look for START (0x02) found 0x02
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x44
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x33
 Look for START (0x02) found 0x31
 Look for START (0x02) found 0x39
 Look for START (0x02) found 0x43
 Look for START (0x02) found 0x44
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x37
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x03
 Look for START (0x02) found 0x02
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x44
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x33
 Look for START (0x02) found 0x31
 Look for START (0x02) found 0x39
 Look for START (0x02) found 0x43
 Look for START (0x02) found 0x44
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x37
 Look for START (0x02) found 0x30
 Look for START (0x02) found 0x03

lbernstone, 2018-05-13:
This is now a basic logic control problem.  Please close the issue and either do some troubleshooting on your own or take the conversation over to https://gitter.im .

closed 2018-05-13 by hoangthaibmvl

stickbreaker, 2018-05-13:
@hoangthaibmvl Looks like I forgot to actually accept the start
change:
```c++
  else { // haven't found START yet, so discard
    Serial.printf(" Look for START (0x02) found 0x%02X\n",ch);
	}
```
to:
```c++
  else { 
      if(ch==0x2) started = true;
      else{// haven't found START yet, so discard
          Serial.printf(" Look for START (0x02) found 0x%02X\n",ch);
	  }
       }
 ```

chuck.

hoangthaibmvl, 2018-05-13:
END detected,but packet not Correct Length, expected 10+cksum, actual=12
 Raw Packet = 0D00319CD070

0D00319CD070 this is my RFID right ?