espressif/arduino-esp32 #1409: arduino as a ESP-IDF component

Buffalchill, 2018-05-15:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE / Msys32 ESP-IDF
Flash Frequency:					40Mhz
Upload Speed:						115200

```
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Description:
I need to use the Arduino as an ESP-IDF component. I followed the instructions [Arduino-as-an-idf-component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)
then in `make menuconfig` i changed the sdkconfig => arduino option to run the code with `setup()` and `loop()`.
after `make flash` i got this this error at the end of building the project :

[image]

`undefined reference to Setup()` and  `undefined reference to Loop()`

Can you please tell me what went wrong and how to fix this?
Regards.

lbernstone, 2018-05-15:
Fixing it is easy.  Don't tick that box.  What that option does is compile <arduino-esp32>/cores/esp32/main.cpp into your project.  If you look at that file, it includes an app_main, and references setup() and loop(), which is what Arduino wants.

Buffalchill, 2018-05-16:
thank you!
but i still don't see what is wrong with my sketch. I did everything like in the example. What exactly do i have to change?

Regards

lbernstone, 2018-05-16:
In the future, please post the errors so that I don't have to guess what the compiler is choking on.  I will assume it is function/symbol not defined errors.  Arduino IDE does a lot of magic for you finding libraries.  You will need a lot more includes to do get everything you need.

Buffalchill, 2018-05-16:
C:/msys32/home/esp/hello_world/main/hello_world_main.c:31:2: error: 'Serial' undeclared (first use in this function)
  Serial.begin(115200);

HKGK, 2018-07-25:
@Buffalchill I am seeing the same issue. How did you resolve?

QuadCorei8085, 2018-10-25:
I'm having similar issue. Checked out ESP-IDF v3.1 I get this error:
`hello_world/components/arduino/cores/esp32/esp32-hal-time.c:16:28: fatal error: lwip/apps/sntp.h: No such file or directory`

When using the v3.0 tag:
`make[1]: execvp: xtensa-esp32-elf-gcc: Argument list too long
make[1]: *** [/d/esp32/esp-idf/make/component_wrapper.mk:274: cores/esp32/libb64/cencode.o] Error 127
`

QuadCorei8085, 2018-10-25:
fixed by #1853

Cylindric, 2018-10-27:
Just to add some data, I too am having this problem. I have a simple "main.cpp".
My esp-ldf repo is pulled as of right now, bd90f534fb1cdf7d4406767f9067398d51a12305
My arduino-esp32 is also current, 85032b226c7775bae510e954112823a2ae32181a

In menuconfig I have enabled the Arduino setup and loop processing. 
My sdkconfig and build output are at https://gist.github.com/Cylindric/080a8f4b852574b66bda1f11f578be25

	/home/mark/AquaHub/Firmware/AquaHub/build/arduino/libarduino.a(main.o):(.literal._Z8loopTaskPv+0x0): undefined reference to `setup()'
	/home/mark/AquaHub/Firmware/AquaHub/build/arduino/libarduino.a(main.o):(.literal._Z8loopTaskPv+0x4): undefined reference to `loop()'
	/home/mark/AquaHub/Firmware/AquaHub/build/arduino/libarduino.a(main.o): In function `loopTask(void*)':
	/home/mark/AquaHub/Firmware/AquaHub/components/arduino/cores/esp32/main.cpp:15: undefined reference to `setup()'
	/home/mark/AquaHub/Firmware/AquaHub/components/arduino/cores/esp32/main.cpp:17: undefined reference to `loop()'
	collect2: error: ld returned 1 exit status
	make: *** [/home/mark/AquaHub/Firmware/AquaHub/build/aquahub.elf] Error 1
`

If I turn off the setup() and loop() processing, and instead define it manually in my own app_main(), I get a similar error but complaining about an undefined reference to app_main instead.

Cylindric, 2018-10-27:
In case it matters, my source file is called main.cpp

QuadCorei8085, 2018-10-28:
I'm using arduino as component. I followed the guide and it worked.
0) if using windows download the msys package from the esp docs site (zip file) and add the idf_path setup script under etc/profile.d/ (path is given by step 1))
1) check out the esp-idf, not a released branch but the master. I used the currently latest hash, that is: 822de0d038fd273a731ce9ce76addadfc16c1761 dated to 25th nov 9:14:14
2) place arduino under "components"
3) if you wish to use loop and setup like functions configure your sdk using 'make menuconfig'
4+) all the rest settings can be done using menuconfig

Cylindric, 2018-10-28:
Oddly enough, that's what I've been trying. Do you have your project in git or somewhere I could see it? There must be something different about what the people with problems are trying and what you've done.

I'll try your version, not sure where you got the "latest" as being 25th November from, there have been loads of commits since then. I'll try 822de.

QuadCorei8085, 2018-10-29:
In ESP-IDF you have to have a "main" folder where you have your file(s).
That has to contain a "component.mk".
Above that folder level (project root) you have to have the makefile and the sdkconfig (without .h - that is generated during build time).

Cylindric, 2018-10-29:
Aah, I didn't know that. I grabbed the one from the hello-world example.

So for future reference, I now have:

```
Root
├── Makefile
├── sdkconfig
├── components
│   └── arduino
├── main
    ├── component.mk
    ├── main.cpp
```

QuadCorei8085, 2018-10-30:
Yes but if you follow the instruction (and use the current master) it works. Later you can remove any file one by one and check what is necessary on a trial-and-error basis if you don't want to get familiar with the build system.

Cylindric, 2018-10-30:
I guess I just figure that documentation should include the "why" too, and not require future guesswork to determine what might or might not be an actual requirement, and what is fluff to make a "hello world" app say "hello world". I'm not trying to be difficult, just pointing out that the docs could be a little clearer and not depend on copying examples.
Maybe an empty minimal template would be more suitable to have somewhere in there.

Goldenstarc, 2018-12-06:
> Thank you!
> 
> After some time i got it running now...
> 
> ```
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "Arduino.h"
> 
> #if CONFIG_AUTOSTART_ARDUINO
> 
> #if CONFIG_FREERTOS_UNICORE
> #define ARDUINO_RUNNING_CORE 0
> #else
> #define ARDUINO_RUNNING_CORE 1
> #endif
> 
> void loopTask(void *pvParameters)
> {
>     setup();
>     for(;;) {
>         micros(); //update overflow
>         loop();
>     }
> }
> 
> void app_main()
> {
>     initArduino();
>     xTaskCreatePinnedToCore(loopTask, "loopTask", 8192, NULL, 1, NULL, ARDUINO_RUNNING_CORE);
> }
> 
> #endif
> 
> void setup(){
> 	//Serial.begin(115200);
> 	pinMode(4, OUTPUT);
> }
> 
> void loop(){
> 	//Serial.println("Hello World");
> 	digitalWrite(4, HIGH);
> 	delay(1000);
> 	digitalWrite(4, LOW);
> 	delay(1000);
> } 
> ```
> But one problem that i still have: I get compiler Error messages when I use `Serial.begin()` or `Serial.print()`
> Is it possible to use the complete Arduino sketch with all its functions and librarys in the ESP-IDF ?
> If yes, how ?
> 
> Regards

> Thank you!
> 
> After some time i got it running now...
> 
> ```
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "Arduino.h"
> 
> #if CONFIG_AUTOSTART_ARDUINO
> 
> #if CONFIG_FREERTOS_UNICORE
> #define ARDUINO_RUNNING_CORE 0
> #else
> #define ARDUINO_RUNNING_CORE 1
> #endif
> 
> void loopTask(void *pvParameters)
> {
>     setup();
>     for(;;) {
>         micros(); //update overflow
>         loop();
>     }
> }
> 
> void app_main()
> {
>     initArduino();
>     xTaskCreatePinnedToCore(loopTask, "loopTask", 8192, NULL, 1, NULL, ARDUINO_RUNNING_CORE);
> }
> 
> #endif
> 
> void setup(){
> 	//Serial.begin(115200);
> 	pinMode(4, OUTPUT);
> }
> 
> void loop(){
> 	//Serial.println("Hello World");
> 	digitalWrite(4, HIGH);
> 	delay(1000);
> 	digitalWrite(4, LOW);
> 	delay(1000);
> } 
> ```
> But one problem that i still have: I get compiler Error messages when I use `Serial.begin()` or `Serial.print()`
> Is it possible to use the complete Arduino sketch with all its functions and librarys in the ESP-IDF ?
> If yes, how ?
> 
> Regards

you are not using Arduino runtime, digitalwrite and pin mode exists in ESP library and you think that you are using Arduino interface !

QuadCorei8085, 2018-12-06:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

closed 2019-08-15 by stale[bot]