espressif/arduino-esp32 #1412: ESP32 secure webserver

vishalkothari, 2018-05-16:
## Hardware:
Board:							ESP32 Dev Module WROOM DEVKITV1
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE

I would like to secure communications to ESP32 webserver with authentication and encryption.
I dont see any documentation or examples of how this can be done. All the examples I see are about WifiClientSecure.

It will be great help if the community can point me to an example or documentation about ESP32 secure webserver (with SSL mutual auth or any other methods.)

Best Regards,
Vishal Kothari

martinius96, 2018-06-03:
You cannot use it for that, because you need domain, nad also root CA certificate, without domain, your certificate will be untrusted. It is very complicated. So, you cannot use (right now) this module like you want.

vishalkothari, 2018-06-04:
Thanks @martinius96

closed 2018-06-04 by vishalkothari