espressif/arduino-esp32 #1414: xTaskCreate

Kazuya91, 2018-05-16:
Hello,

I want to use the task functions of FreeRTOS in the Arduino Environment. I also want to use the "ulTaskNotifyTake" so I can synchronize tasks.

The problem is, that I cant create tasks. It results in an error. I googled many examples and even copy-pasted them but still the same...

This is my code:

```
#include <Wire.h>
#include <dummy.h> //ESP32 bibliothek
#include "esp_deep_sleep.h"
#include "driver/rtc_io.h"
#include "freertos/task.h"
#include <WiFi.h>

#define MOSFET_SWITCH_PIN 19

TaskHandle_t turn_on_IR_Task;

void setup() {
  Serial.begin(115200);
  delay(500);
  xTaskCreate(turn_on_IR, "turn_on_IR",10240,NULL,1, NULL);
 //also tried a "&" before turn_on_IR
}

void turn_on_IR(){

  digitalWrite(MOSFET_SWITCH_PIN, HIGH);

  vTaskDelete(NULL);
}
```

The error says:
```
/Users/Kazuya91/Documents/Arduino/test/test.ino: In function 'void setup()':
test:148: error: invalid conversion from 'void (*)()' to 'TaskFunction_t {aka void (*)(void*)}' [-fpermissive]
   xTaskCreate(turn_on_IR, "turn_on_IR",10240,NULL,1, NULL);
                                                          ^
In file included from /Users/Kazuya91/Documents/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:33:0,
                 from sketch/test.ino.cpp:1:
/Users/Kazuya91/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/task.h:422:37: note:   initializing argument 1 of 'BaseType_t xTaskCreate(TaskFunction_t, const char*, uint32_t, void*, UBaseType_t, void**)'
  static inline IRAM_ATTR BaseType_t xTaskCreate(
invalid conversion from 'void (*)()' to 'TaskFunction_t {aka void (*)(void*)}' [-fpermissive]
```

stickbreaker, 2018-05-16:
`void turn_on_IR(void * unused)`

Chuck.

lbernstone, 2018-05-16:
https://desire.giesecke.tk/index.php/2018/01/30/multithreading-support-with-xtaskcreate/

closed 2019-08-15 by stale[bot]