espressif/arduino-esp32 #1420: WIFI, BLE, HTTP, SD_MMS error download file. Small free heap?

ArTyGill, 2018-05-19:
### Hardware:
Board:							ESP32 Dev Module
Chip:							ESP32D0WDQ6 (revision 0)
Core Installation/update date:			11/jul/2017
IDE name:							VisualStudio (Arduino IDE)
Flash Frequency:				                80Mhz
Upload Speed:						921600
esptool.py : 							v2.1

### Description:
I use wifi and https client to download the file to the SD card, everything works fine, but when I initialize the bluetooth le when accessing the server using the get query, it returns an error code (-1). Although the esp itself is connected to the wifi and pings in my local network with bluetooth without problems.

### Sketch:
```cpp

/******************** Includes **********************/
#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>
#include <HTTPClient.h>

#include <FS.h>
#include "SD_MMC.h"

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLEBeacon.h>
#include <BLE2902.h>

/********************** Definitions ***************************/

#define UUID_SERVISE         (BLEUUID("00000000-0000-0000-0000-111111111111"))
#define UUID_CHARACH         (BLEUUID("00000000-0000-0000-0000-222222222222"))

/********************** Variables *****************************/

WiFiMulti Multiwifi;
const char* ssid = "VB";
const char* password = "C039D2E9";

HTTPClient http;

BLEAdvertising *pAdvertising;

uint8_t cardMounted;

/****************** Function prototypes ********************/

void sd_init(void);
void wifi_init(void);
void BLE_init(void);
void ble_adv_start(void);
void writeFile(fs::FS &fs, const char * path, const char * message);

/************************ Functions *************************/
void sd_init(void) {

    gpio_set_pull_mode(GPIO_NUM_2, GPIO_PULLUP_ONLY); //set Pullup for SD_MMC slot
    gpio_set_pull_mode(GPIO_NUM_4, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_12, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_13, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_15, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_14, GPIO_PULLUP_ONLY);
    if (!SD_MMC.begin()) {
	cardMounted = 0;
	Serial.println("Card Mount Failed");
	return;
    }
[... 201 lines cut ...]
[V][ssl_client.cpp:158] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:177] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:186] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:201] start_ssl_client(): Free heap after TLS 76112
[D][HTTPClient.cpp:852] connect():  connected to localhost:8080
[V][ssl_client.cpp:240] send_ssl_data(): Writing HTTP request...
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'X-Powered-By: Express'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Accept-Ranges: bytes'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Cache-Control: public, max-age=0'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Last-Modified: Sat, 19 May 2018 08:19:26 GMT'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'ETag: W/"1-163777b66b0"'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Content-Type: text/plain; charset=UTF-8'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Content-Length: 1'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Date: Sat, 19 May 2018 10:37:58 GMT'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:940] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:970] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:973] handleHeaderResponse(): size: 1
[V][ssl_client.cpp:209] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:209] stop_ssl_socket(): Cleaning SSL connection.
[D][HTTPClient.cpp:1103] writeToStreamDataBlock(): connection closed or file end (written: 1).
[V][HTTPClient.cpp:246] end(): tcp is closed

```

lbernstone, 2018-05-19:
You should try to use streams to download large amounts of data (large is not all that large in esp32).  https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/StreamHttpClient/StreamHttpClient.ino

ArTyGill, 2018-05-19:
Yes, that's right, I know this, you need to use stream to upload large files, BUT this is a text file in which only one character is written, and the error also occurs as with normal print and when recording with stream.

ArTyGill, 2018-05-19:
Is there any reason to use WiFiClientSecure instead of HTTPClient, to save RAM?

lbernstone, 2018-05-19:
I'm sorry.  I didn't really read your post properly the first time through.  Heap errors generally throw kernel exceptions.  This is a connection error, as it says. I don't see where you are setting up the socket on localhost, but if it is handled in loop() this could be an indication that the cpu is busy and doesn't ack before timing out.

ArTyGill, 2018-05-19:
I have a free 50KB of RAM, ssl opens a socket, but still if you look at the logs, you will see there malloc error. 
[E][ssl_client.cpp:28] handle_error(): RSA - The public key operation failed : BIGNUM - Memory allocation failed
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -17040

Why does Bluetooth violate http, maybe someone knows why this is happening?

negativekelvin, 2018-05-20:
Free heap before TLS 123148 - Free heap after TLS 76112 = 47036 which is pretty close to 50524 especially when you consider heap fragmentation

negativekelvin, 2018-05-20:
1. esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT) 
2. Change TLS content length but you have to recompile from idf

ArTyGill, 2018-05-20:
I changed the sdkconfig.h file, and got 64KB of free heap, this allowed me to use the HTTP client in this case, but increasing the project still leads to a decrease in the free heap, which in the future will also represent a problem.
It is worthwhile to see more options for cleansing the heap.

ArTyGill, 2018-05-20:
Is there any reason to use pure "C", and eclipse ide with isp-idf?

twinotter, 2018-07-09:
ArTyGill, can you tell me the steps you used to increase the heap space?

Anyone aware of a way to get the TLS system to use the 32bit word addressable memory? It would be great if we could move it into a less frequently used area..

ArTyGill, 2018-07-11:
In sdkconfig you need to disable all unnecessary and unused parameters, and it is possible to try to reduce CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN.
Files:
c:\Users\"YouUserNAME"\Documents\Arduino\hardware\espressif\esp32\tools\sdk\include\config\sdkconfig.h
C:\Users\"YouUserNAME"\Documents\Arduino\hardware\espressif\esp32\tools\sdk\sdkconfig

ArTyGill, 2018-07-11:
And you can use ESP32 with external RAM.
But I hope the SDK will optimize soon.

ArTyGill, 2018-07-11:
And do not forget to do Backup.

wegunterjr, 2018-07-11:
but making these changes suggested (in the sdk config and the CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN) doesn't work in the arduino IDE...right?

ArTyGill, 2018-07-18:
These are used by the Arduino environment, and therefore work, but do not expect a large increase in the free heap.

kapollo, 2018-07-30:
I'm struggling with the same problem, the bt uses whole lot of heap
Could we recompile the sdk to remove classic bt, there is an option in the sdkconfig for that, how then use that with arduino ide or platform.io?

ArTyGill, 2018-10-04:
After I started using Esp32-Wrover with external RAM, everything works fine.

closed 2018-10-04 by ArTyGill