espressif/arduino-esp32 #1423: Fatal error when trying to include FreeRTOSConfig.h: No such file or directory

PaulChristopher, 2018-05-21:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation/update date:			Manual installation with zip file from github on 23/03/2018
IDE name:							Arduino IDE 1.8.5.
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I want to use the lightweight µGFX library which supports FreeRTOS to build simple GUI applications, see https://ugfx.io/.
Some of the µGFX library files <include> header files such as FreeRTOS.config.h, semphr.de. Compiling the sketch and the library gives a fatal error: "FreeRTOSConfig.h: No such file or directory".

Looking into platform.txt line 25, this path is included: "-I{compiler.sdk.path}/include/freertos". The mentioned directory is empty and has only one subfolder "freertos" with all the necessary file.

I could make the sketch/library get compiled by copying all the files from "{compiler.sdk.path}/include/freertos/freertos" to "{compiler.sdk.path}/include/freertos".

However, the µGFX library isn't outputting anything. So this ticket is starter to figure out what might be the cause for this. Am I using the FreeRTOS system wrong? Or is the µGFX library broken?

Why can't the µGFX library just simply include "FreeRTOSConfig.h"? I read somewhere, that this files get dynamically created? But what about the other includes such as semphr.h?

lbernstone, 2018-05-21:
I don't think you are going to get the performance you want running this through arduino-esp32, which is an abstraction on top of freertos.  I would recommend using arduino-esp32 as an esp-idf component if you really want the additional libraries.  That said, you should be able to `#include <freertos/FreeRTOSConfig.h>` to get it included in your sketch.

closed 2019-08-15 by stale[bot]

ashigupta, 2019-12-18:
> I don't think you are going to get the performance you want running this through arduino-esp32, which is an abstraction on top of freertos. I would recommend using arduino-esp32 as an esp-idf [...]

Hi,

Is there any way with which i can just include FreeRTOSConfig.h rather than freertos/FreeRTOSConfig.h

Regards,
AG

lbernstone, 2019-12-18:
That is the physical location of the file.
https://en.wikipedia.org/wiki/XY_problem

ashigupta, 2019-12-18:
> That is the physical location of the file.
> https://en.wikipedia.org/wiki/XY_problem

Did not understood this response.