espressif/arduino-esp32 #1425: ESPNow example can't work on latest commit

0x1abin, 2018-05-23:
On the latest commit version, I upload the ESPNow Basic example and running master and slave on two devices, the Master can't find the Slave.
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/ESPNow

Serial Log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c
ESPNow/Multi-Slave/Master Example
STA MAC: 30:AE:A4:67:XX:XX
ESPNow Init Success
E (285) wifi: esp_wifi_scan_start 967 wifi not start

Found -2 devices 
No Slave Found, trying again.
E (1290) wifi: esp_wifi_scan_start 967 wifi not start

Found -2 devices 
No Slave Found, trying again.
```

But I checkout to #669 it can work.
```bash
git checkout 4c36e033812a4142ba417565926ff8f919a32646
```
Serial Log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:572
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
ESPNow/Basic/Master Example
STA MAC: 30:AE:A4:67:2F:9C
ESPNow Init Success

Found 46 devices 
Found a Slave.
1: Slave_1 [30:AE:A4:4F:XX:XX] (-30)
Slave Found, processing..
Slave Status: Pair success
Sending: 1
Send Status: Success
Last Packet Sent to: 30:ae:a4:4f:XX:XX
Last Packet Send Status: Delivery Success
```

0x1abin, 2018-05-30:
I erase flash, but it not work, does anyone have the same problem?

eg-astrouka, 2018-05-31:
I have the same problem also. And I'm sure I had Client-Master communication working a few months ago. Now I have Master working only first time I flash new board with working firmware. It stops working after reflash/restart.

eg-astrouka, 2018-05-31:
Update: erase_flash on Master helped

appie2002, 2018-05-31:
I have the same problem. Master only works the first time and after reboot it gives the same error.
E (1290) wifi: esp_wifi_scan_start 967 wifi not start

Found -2 devices 
No Slave Found, trying again.

Mahmoudomar91, 2018-06-05:
you need to add `WiFi.disconnect();` in the first line into the `void InitESPNow() {` then it will work.

eg-astrouka, 2018-06-06:
@Mahmoudomar91 thanks, it works

lbernstone, 2018-06-06:
@0x1abin : Can you confirm the disconnect fixes this for you as well?  I'll put in a pull request with the change if so.

Mahmoudomar91, 2018-06-06:
@lbernstone  yes i tested it on my ESP32 and it worked.

closed 2018-06-11 by 0x1abin