espressif/arduino-esp32 #1426: How to link static libraries into arduino code?

sysizlayan, 2018-05-23:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		Manually updated every day, always using last master
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600

### Description:
Hi, I am curerntly working on IOT project and I need to link BSEC library for BME680 sensor. Library link is given below.
https://www.bosch-sensortec.com/bst/products/all_products/bsec
This library is compiled for many processor architectures and ESP32 is one of them. However Arduino libraries does not contain ESP32. So I needed to link static ".a" library to my arduino sketch.
Is it possible? Where should I place the ".a" file and which files should I edit to be able link this file to my sketch.

In short, how can one link a static libraries to project?

sysizlayan, 2018-05-23:
I have used the methods mentioned in
https://github.com/arduino/Arduino/issues/4336
Changed, platform.txt and libraries required files as said in documents in BSEC.
Arduino builder is also changed to PR219 version, this allows arduino to be able link precompiled libraries.

closed 2018-05-23 by sysizlayan

Anupam7219, 2025-04-05:
Please find simple steps for using custom build ESP32 Static lib(.a) file in Arduino IDE.

1.	Open ld_libs file in “C:\Users\XXY\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-632e0c2a\esp32s3\flags”

2.	Add your static lib file name in file (Example: Hellolib)

[image]

3.	Copy custom buid external lib file in “C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-632e0c2a\esp32s3\lib” location.

[image]

Build project from IDE.