espressif/arduino-esp32 #1427: spi timing

idoroseman, 2018-05-23:
I'm connecting the esp32 to CR95HF via the spi port. 
i've noticed there's a bit missing on every transaction. so i dug into the manuals and found there's a cure by changing the SPI_MISO_DELAY_MODE to 2
sure enougth i've added the following line into esp32-hal-spi.c @ spiStopBus()
`    _spi->dev->ctrl2.miso_delay_mode = 2;`
and problem was solved. for future compateblity, i wanted to move the fix to my code and not leave it in the driver code.
i saw you can export spi_t by using SPI.bus() but i get an error
` error: invalid use of incomplete type 'spi_t {aka struct spi_struct_t}'`
any idea how to fix that ?
thanks

stickbreaker, 2018-05-23:
@idoroseman on the incomplete type: verify where `spi_struct_t` is defined.  I have found that the espressif style is to obfuscate HAL structures,  In the `.h` of a library a structure is incompletely defined as:
```
struct hidden_struct_t;
typedef struct hidden_struct_t hidden_t;
```
then in the `.c` the structure is actually fleshed out.
```
struct hidden_struct_t {
    uint8_t super_secret;
    uint32_t dangerous;
};
```

This allows a HAL function to be passed a complex structure, and simultaneously preventing direct modification of the structure elements by the app.

if you need direct access to elements of the structure, copy the definition from the `.c` into your sketch.

Chuck.

idoroseman, 2018-05-23:
thanks @stickbreaker it did the trick
for future reference i added the following at the begining of the code

    #include "soc/spi_struct.h"
    struct spi_struct_t {
        spi_dev_t * dev;
    #if !CONFIG_DISABLE_HAL_LOCKS
        xSemaphoreHandle lock;
    #endif
        uint8_t num;
    };

and then i added to setup()

    SPI.begin();
    spi_t * _spi;
    _spi= SPI.bus();
    _spi->dev->ctrl2.miso_delay_mode = 2;

before closing this, maybe there should be a nicer way to change registers inside the driver (but this is beyond my scope)

closed 2019-08-15 by stale[bot]