espressif/arduino-esp32 #1437: About WROOM32/WROOM32D board list

asetyde, 2018-05-25:
I think is useful for producer like me has a ESP-WROOM-32/D/U option on board list, is more correct to identify most common chip of this esp32.

Thanks Alex, if admin read it can be closed.

lbernstone, 2018-05-26:
esp-wroom-32d/u are modules, not development boards.  The variants included in the boards list have different pins exposed, flash configuration or led lights, or something else unique.  A basic esp-wroom-32 device on a pcb with all the pins exposed will work fine using the generic esp32 dev board.

asetyde, 2018-10-02:
> esp-wroom-32d/u are modules, not development boards. The variants included in the boards list have different pins exposed, flash configuration or led lights, or something else unique. A basic esp-wroom-32 device on a pcb with all the pins exposed will work fine using the generic esp32 dev board.

It's not true, some pins of WROOM 32 has spi flash memory link, but generic esp323 chip not has always. Warning !

closed 2018-10-02 by asetyde