espressif/arduino-esp32 #1439: Read html file from sd card

DianaSalgado, 2018-05-25:
Hi people. I'm working on a project that requires opening a html file from sd card. The board that I'm working on is ESP32-EVB Olimex. 
After looking up, I found some exaples that are very interesting but I keep having some erros and I don't know what to do. 
Can someone help me?

Here is the code I have:
#include "FS.h"
#include "SD_MMC.h"
#include <OneWire.h>
#include <DallasTemperature.h>
#include <WiFi.h>
#include <ESP32WebServer.h>
#include <ESPmDNS.h>
#include <WiFiClient.h>

/*-----------------------------------------------VARIÁVEIS GLOBAIS------------------------------------------------------------*/
int sensor = 5;

OneWire oneWire(sensor);            /*criação de uma estação para comunicar com todos os sensores ligados a este porto*/

DallasTemperature sensor_temperatura(&oneWire);       /*Passa a referência da estação criada para a Dallas Temperature*/

const char* nome= "Huawei P8 Lite";
const char* password= "fiat500bygucci";

ESP32WebServer server(80);

/*-----------------------------------------------ADICIONAR INFORMAÇÃO AO FICHEIRO------------------------------------------------------------*/
//void appendFile(fs::FS &fs, const char * path, const char * message){
//    Serial.printf("Appending to file: %s\n", path);
//
//    File file = fs.open(path, FILE_APPEND);
//    if(!file){
//        Serial.println("Failed to open file for appending");
//        return;
//    }
//    if(file.print(message)){
//        Serial.println("Message appended");
//    } else {
//        Serial.println("Append failed");
//    }
//}

/*-----------------------------------------------SETUP------------------------------------------------------------*/
void setup(){
    pinMode(sensor, INPUT);
    Serial.begin(115200);

    /**********************************************INICIO WIFI**************************************************/
    WiFi.begin(nome, password);              /*Conecção do ESP ao WiFi*/
  
    while(WiFi.status() != WL_CONNECTED)   /*Ciclo while() para verificar se a conecção foi feita. Vai verificar se o método status do WiFi é igual ao WL_CONNECTED*/
    {
        delay(500);   /*O delay é utilizado para que haja espera por resposta externa*/
        Serial.println("A conectar...");
    }

    Serial.print("Conectado a: ");
    Serial.println(nome);
    Serial.print("Endereço IP: ");
    Serial.println(WiFi.localIP());

    if(MDNS.begin("esp32")){
      Serial.println("Resposta de MDNS iniciada!");
    }
    
    

    server.on("/", handleRoot(SD_MMC, "/index.html"));
    server.onNotFound(handleNotFound);
    server.begin();

    Serial.println("HTTP server started");
    Serial.print("Initializing SD card...");
    /**********************************************INICIO SENSOR TEMPERATURA**************************************************/
    sensor_temperatura.begin();
    /**********************************************INICIO SD CARD**************************************************/
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    Serial.println("Inicialização do cartão feita com sucesso!");
    
    uint8_t cardType = SD_MMC.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD_MMC card attached");
        return;
    }

    Serial.print("SD_MMC Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

   uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
   Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);

}

void loop(){
    server.handleClient();     
  /**********************************************LEITURA SENSOR TEMPERATURA**************************************************/
  Serial.print("À espera da leitura da temperatura...");
  sensor_temperatura.requestTemperatures();             /*Vai pedir a leitura global dos sensores utilizados*/
  Serial.print("A temperatura é: ");

  Serial.println(sensor_temperatura.getTempCByIndex(0));     /*Tendo em conta que posso ter mais do que um sensor ligado
                                       ao mesmo pino o 0 serve para indicar que quero a leitura o primeiro IC do pino*/

  /**********************************************ESCRITA PARA SD CARD**************************************************/
  //char sensortemp[32];
  //float leitura;
  //leitura = sensor_temperatura.getTempCByIndex(0);
  //dtostrf(leitura, 8, 2, sensortemp);
  //appendFile(SD_MMC, "/sensortemp.txt",sensortemp);
  //appendFile(SD_MMC, "/sensortemp.txt", "           ");
  //readFile(SD_MMC, "/index.html");
  
  //delay(10000000);

 
       
}

/*-----------------------------------------------FUNÇÃO PARA LIDAR COM SERVIDOR------------------------------------------------------------*/
void handleRoot(fs::FS &fs, const char * path){
  File file = fs.open("index.html");
  if(file){
    size_t sent = server.streamFile(file,"text/html");
    file.close();
  } else{
    Serial.println("Erro na abertura do ficheiro!");
  }
}

/*-----------------------------------------------SERVIDOR NÃO ENCONTRADO------------------------------------------------------------*/
void handleNotFound(){
  String message = "Ficheiro não encontrado\n\n";
  server.send(404, "text/plain", message);
}

And these is the error that is shown when I try to compile:
C:\Users\Diana\Documents\Arduino\SD_CARD\SD_CARD.ino: In function 'void setup()':

SD_CARD:150: error: invalid use of void expression

     server.on("/", handleRoot(SD_MMC, "/index.html"));

                                                     ^

exit status 1
invalid use of void expression

lbernstone, 2018-05-25:
You really should post this wherever you got the ESP32WebServer.h from.  However, what server.on wants is just a callback to the function.  You cannot send parameters this way.

DianaSalgado, 2018-05-26:
Thank's for your answer. I followed your advice and I already posted this in the place where I got ESP32WebServer.h
About server.on , I realized that he only wants a callback to the function. But, when I put:
void handleRoot(){
  File file = fs.open("index.html");
  if(file){
    size_t sent = server.streamFile(file,"text/html");
    file.close();
  } else{
    Serial.println("Erro na abertura do ficheiro!");
  }
}

and:   server.on("/", handleRoot);

Arduino IDE gives me this error: 
C:\Users\Diana\Documents\Arduino\SD_CARD\SD_CARD.ino: In function 'void handleRoot()':

SD_CARD:229: error: expected primary-expression before '.' token

   File file = fs.open("index.html");

                 ^

exit status 1
expected primary-expression before '.' token

lbernstone, 2018-05-26:
This is not a programming tutorial site.  Your issue is not a problem with the system code, but with your code.  Please close the issue and look through the examples in esp8266 (https://github.com/esp8266/Arduino/blob/master/libraries/SD/examples/ReadWrite/ReadWrite.ino) for simpler global objects.

rel1ct, 2018-05-26:
File file = fs.open("/index.html");

closed 2018-05-26 by DianaSalgado