espressif/arduino-esp32 #1442: Issue with onewire since commit 7aa8fc4

CelliesProjects, 2018-05-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			7aa8fc4b26dfb66de46b51f6baae3f33367e4c6b
IDE name:					        Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
Since 7aa8fc4b26dfb66de46b51f6baae3f33367e4c6b my upload script does something strange.
This is the script:
````bash
echo Compiling and flashing aquacontrol32 version: $(git describe --tags --always)
echo We are on branch: $(git branch | grep \* | cut -d ' ' -f2)
echo
echo "const char * sketchVersion = \"$(git describe --tags --always)\";" > gitTagVersion.h
~/arduino-1.8.5/arduino --upload test.ino --pref custom_DebugLevel=esp32_none --port /dev/ttyUSB0
rm gitTagVersion.h
````
Basically it provides a version number in my sketch, based on current git tag and commit.
But if I use this script with an ESP32 Arduino core after 7aa8fc4b26dfb66de46b51f6baae3f33367e4c6b my one-wire ds18b20 sensors don't get recognized.
If I flash from the Arduino IDE after 7aa8fc4b26dfb66de46b51f6baae3f33367e4c6b I can use my sensors, but flashing with the script above results in no Dallas sensors found and no error messages.

Anybody else having issues with onewire since this commit?
Or can somebody shed some light on why this does not work?

me-no-dev, 2018-05-26:
why would you have issues with that commit if you are not using that board?

CelliesProjects, 2018-05-27:
I discovered that my compile script worked by sheer luck. I should have setup the compile for a specific device. The Arduino IDE commandline probably defaulted to the OLIMEX/ESP32-GATEWAY device.

Anyway, the problem is solved by adding a specific board parameter to the Arduino commandline.

For reference:
````bash
echo Compiling and flashing aquacontrol32 version: $(git describe --tags --always)
echo We are on branch: $(git branch | grep \* | cut -d ' ' -f2)
echo
echo "const char * sketchVersion = \"$(git describe --tags --always)\";" > gitTagVersion.h
~/arduino-1.8.5/arduino --board espressif:esp32:esp32 --upload test.ino --pref custom_DebugLevel=esp32_none --port /dev/ttyUSB0
rm gitTagVersion.h
````
The missing part was the `--board espressif:esp32:esp32` parameter.

closed 2018-05-27 by CelliesProjects