espressif/arduino-esp32 #1446: ESP-WROOM-32 ADC2 with WiFi

MeisterQ, 2018-05-28:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

Im measuring a voltage on GPIO 26 (ADC2) and want to transmit it to thinspeak.

My way is simple:
-> Wake up
-> Measure voltage
-> Get Sensordata from a I2C Sensor
-> Connect to WiFi
-> Send data
-> Sleep 10minutes

In this way, after i press a reset button, its measuring my voltage correctly. After the 10minutes of deepsleep, it shows me only "8.2" (Its my maximum voltage range). So. i thought after a deepsleep the ESP32 is resetting itself? seems like its not?

Connection to wifi like this:

  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  WiFi.config(ip, gateway, subnet);

Going to sleep like this:

ESP.deepSleep(sleepTime * 1000000);

Is there a way to disable wifi completely? Thanks.

lbernstone, 2018-05-29:
deepsleep is definitely not the same as a reset. Unfortunately, esp_wifi_deinit does not yet work.  When it does, your code will magically work :) . Until then, you will have to use ADC1 (pins 32-39).

MeisterQ, 2018-05-29:
But after deepsleep the ESP is doing a reset, right?

Maybe i could do a "ESP.reset();" at the end of my code? Would it fix it temporary?

When "esp_wifi_deinit" will work?

closed 2019-08-15 by stale[bot]