espressif/arduino-esp32 #1449: Recommended method to add Arduino libraries using arduino-eps32 as component to esp-idf

PreferHardware, 2018-05-30:
Using arduino-esp32 as a component to esp-idf in Eclipse seems to work well.  I installed it exactly as the README suggests and it compiled first time.

What is the recommended way to add Arduino libraries and ensure that make knows what to do with them?

Many thanks.

copercini, 2018-05-30:
I'm not sure what is the better way, but the easier way is just put your libs on `ProjectFolder\components\arduino\libraries` and that's all

PreferHardware, 2018-06-02:
Thanks, that works.  I was hoping that there was an established method to include libraries, but keeping them separate from the arduino core.

lbernstone, 2018-06-02:
You can softlink in order to keep some logical separation.  git won't complain quite as much.  Eclipse may have some other way to include directories into your project.

dannybackx, 2018-07-17:
I tried several of the variables described in the documentation (chapter 4 "API Guides" -> "Using the Build System" if you look at the pdf).
I've added several arduino libraries as libraries/ so libraries/TFT_eSPI , libraries/pubsubclient etc .
I'm looking for a way to : 
- get that to work
- make their build use include files from main/ 

This would keep things much more nicely separated ....

Does anyone know how to do this ?

Workarounds like
ln -s `pwd`/libraries/* components/arduino/libraries
are possible but not clean

me-no-dev, 2018-07-18:
you can add the libs to your components folder and then add one file in each lib so the build will pick them up. Here is an example that has the source files in src folder (adjust as necessary) https://github.com/me-no-dev/AsyncTCP/blob/master/component.mk

dannybackx, 2018-07-19:
Thanks.
More trial and error got me with this (main/component.mk) which appears to work.
Anything I'm still doing wrong ?

```
COMPONENT_SRCDIRS += libraries \
        libraries/TFT_eSPI \
        libraries/arduinojson \
        libraries/pubsubclient \
        libraries/pubsubclient/src \
        libraries/rcswitch \
        libraries/rfid \
        libraries/TFT_eSPI \
        libraries/Timezone \
        main .

COMPONENT_ADD_INCLUDEDIRS := libraries \
        libraries/arduinojson \
        libraries/pubsubclient \
        libraries/pubsubclient/src \
        libraries/rcswitch \
        libraries/rfid \
        libraries/TFT_eSPI \
        libraries/Timezone \
        main .
```

Tuet, 2018-07-26:
Hi,
I added a libraries folder below the main folder and copied the libraries in it. Then I populated the main/component.mk with the following to add all library contents (but as all source code is included, you have to remove examples and other source-code typically not used in the library)

```
ARDUINO_ULIBRARIES_LIST := $(patsubst $(COMPONENT_PATH)/libraries/%,%,$(wildcard $(COMPONENT_PATH)/libraries/*))
ARDUINO_SINGLE_ULIBRARY_FILES = $(patsubst $(COMPONENT_PATH)/%,%,$(sort $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/*/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/*/*/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/*/*/*/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/*/*/*/*/*/))))
ARDUINO_USER_LIBS := $(foreach MODULE,$(ARDUINO_ULIBRARIES_LIST), $(ARDUINO_SINGLE_ULIBRARY_FILES))

COMPONENT_ADD_INCLUDEDIRS := $(ARDUINO_USER_LIBS) main .
COMPONENT_SRCDIRS := $(ARDUINO_USER_LIBS) main .
```

For me this works for standard Arduino libraries. I derived this from the component.mk of the arduino-folder. Maybe you can change it to only include the /src/-folder stuff from the libraries, however, not all libraries are using this new convention.

closed 2019-08-15 by stale[bot]

allex1978, 2019-11-03:
апаук

> Hi,
> I added a libraries folder below the main folder and copied the libraries in it. Then I populated the `main/component.mk` with the following below to add all library contents (all source files in library main folder and all source files in library/src/ and all sub-folders of library/src/).
> 
> ```
> ARDUINO_ULIBRARIES_LIST := $(patsubst $(COMPONENT_PATH)/libraries/%,%,$(wildcard $(COMPONENT_PATH)/libraries/*))
> ARDUINO_SINGLE_ULIBRARY_FILES = $(patsubst $(COMPONENT_PATH)/%,%,$(sort $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/*)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/src/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/src/*/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/src/*/*/*/)) $(dir $(wildcard $(COMPONENT_PATH)/libraries/$(MODULE)/src/*/*/*/*/))))
> ARDUINO_USER_LIBS := $(foreach MODULE,$(ARDUINO_ULIBRARIES_LIST), $(ARDUINO_SINGLE_ULIBRARY_FILES))
> 
> COMPONENT_ADD_INCLUDEDIRS := $(ARDUINO_USER_LIBS) main .
> COMPONENT_SRCDIRS := $(ARDUINO_USER_LIBS) main .
> ```
> 
> For me this works for standard Arduino libraries. I derived this from the component.mk of the arduino-folder.
> 
> Please let us know if this works for you as well!

Hi, this method not working now. How to use something similar with cmake ?

Suraj-singh999898, 2020-03-03:
hi everyone
I also have this problem 
what I have did
git clone > esp-idf v4.0
git clone > Arduino esp32 v4.0
and the rest of the process same as shown in the official guide.

now I want to add some Arduino another library in my Arduino component library
if I give absolute file location in include declaration than it's work.
but in esp idf 3.2.2 it happens automatically 

sorry for my English.
thanks for your effort