espressif/arduino-esp32 #1451: Rebooting ESP32 again and again

MagicOAk, 2018-05-31:
Whenever I am adding WiFi.begin in my code I'm getting this error-

Guru Meditation Error: Core 1 panic'ed (IllegalInstruction)
. Exception was unhandled.
Core 1 register dump:
PC : 0x400d188d PS : 0x00060230 A0 : 0x00000000 A1 : 0x3ffda070
A2 : 0x000003e8 A3 : 0x3ffc38c4 A4 : 0x00000000 A5 : 0x00000000
A6 : 0x00000000 A7 : 0x00000000 A8 : 0x800d188d A9 : 0x3ffda050
A10 : 0x0000000a A11 : 0x4d87aef6 A12 : 0x3ffc4194 A13 : 0x3ffda050
A14 : 0x00003a98 A15 : 0x3ffda030 SAR : 0x0000000a EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000 LBEG : 0x4000c2e0 LEND : 0x4000c2f6 LCOUNT : 0xffffffff

Backtrace: 0x400d188d:0x3ffda070 0x7ffffffd:0x3ffda0b0

Rebooting...
ets Jun 8 2016 00:22:57

lbernstone, 2018-05-31:
Please use the exception decoder: https://github.com/espressif/arduino-esp32#decoding-exceptions
Please post minimal code that causes the issue.
WiFi.begin gets used quite thoroughly by almost every ESP32 user.

techievex, 2019-06-05:
I am having almost the same problem, so didn't open new issue and wrote my issue over here.  Please have a look into it and let me know what can be done to solve the issue. 

### Hardware:
Board:						ESP32 Dev Module (breakout board)
Core Installation version:			1.0.1-git (downloaded through git gui) 
IDE name:					Arduino IDE
Flash Frequency:				40Mhz
PSRAM enabled:                               no
Upload Speed:					115200
Computer OS:                                  Windows 7

--------------------------------------------------------------------------------------------------
### Description:
The system rebooting continuously by providing following output at serial port when the serial monitor is opened.

ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x00080000,len:5864
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x00080000,len:5864
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x00000000,len:5864
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x00080000,len:5864
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
[... 218 lines cut ...]
A2      : 0x0000002e  A3      : 0x0000000e  A4      : 0x0000f01d  A5      : 0x00000000  
A6      : 0x4011bba4  A7      : 0x00000000  A8      : 0x800d8367  A9      : 0x3ffe3ae0  
A10     : 0x0000f01d  A11     : 0x00000000  A12     : 0xffffffff  A13     : 0x00000020  
A14     : 0x0000001f  A15     : 0x00000002  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000f01d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d84e4:0x3ffe3b20 0x400d85b5:0x3ffe3b60 0x4011bc3e:0x3ffe3b90 0x4011bcfb:0x3ffe3bb0 0x400d999d:0x3ffe3bd0 0x40082b01:0x3ffe3bf0 0x40082d90:0x3ffe3c20 0x40078f0b:0x3ffe3c40 0x40078f71:0x3ffe3c70 0x40078f7c:0x3ffe3ca0 0x40079145:0x3ffe3cc0 0x400806ce:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57
---------------------------------------------------------------------------------------------------

Here is the decoded file of Guru Meditation Error which was obtained when Core Debug Mode was set to Debug:

PC: 0x400d83b4: esp_intr_alloc_intrstatus at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 600
EXCVADDR: 0x00000000

Decoding stack results
0x400d83b4: esp_intr_alloc_intrstatus at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 600
0x400d85b5: esp_intr_alloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 686
0x4011bc3e: rtc_isr_ensure_installed at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/rtc_module.c line 1989
0x4011bcfb: rtc_isr_register at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/rtc_module.c line 2002
0x400d999d: esp_brownout_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/brownout.c line 59
0x40082b01: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 325
0x40082d90: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 234

lbernstone, 2019-06-05:
You are having power issues.  Get a better power supply.

techievex, 2019-06-06:
I did try it with external power supply(3.3V from AMS117) and the code still shows me the same error.

atanisoft, 2019-06-06:
@techievex does that provide at least 600mA at 3.3v? If not, that is why you are still seeing brownout errors. The esp32 requires a lot of current when the WiFi system initializes.

techievex, 2019-06-06:
@atanisoft ya the datasheet for AMS1117 says it can provide 800mA.

atanisoft, 2019-06-06:
Is that a sustained 800mA? It is possible the esp32 is attempting to pull more current, do you have a way to test the current load by the esp32?

techievex, 2019-06-06:
yes i checked it via multi-meter, the overall current consumption shown is just 0.07A when the code is been executed.

chegewara, 2019-06-06:
Faulty usb cable?

techievex, 2019-06-06:
@chegewara nope, I changed it twice and the result is still the same.

stickbreaker, 2019-06-06:
@techievex a multi-meter will not respond fast enough to see the current spikes when WiFi is active. verify your wiring.  add a 1000uf cap as near as you can to the ESP32 

Chuck.

techievex, 2019-06-07:
@stickbreaker I tired your suggestion but still the result is same.

chegewara, 2019-06-07:
New esp32 board?

techievex, 2019-06-07:
@chegewara Just tried it with another ESP, it's still showing the same error

chegewara, 2019-06-07:
If its not esp32 board then it is either usb cable or power supply, there is no other option for brownout (or you are very unlucky and have 2 esp32 boards broken).

techievex, 2019-06-08:
I am not sure if it's broken, because it can run other codes which do not need wifi.
For now, I haven't introduced any other peripheral. 

P.S basically this isn't a development board, I have made the breakout board and placed the ESP32-Wroom-32 chip on it.

atanisoft, 2019-06-08:
If it only fails when wifi is enabled then it is a power supply issue. The WiFi system does require a lot of current when starting up, the datasheet specifies at least 600mA for the power supply. Very likely if you provide closer to 1A it will work fine

techievex, 2019-06-14:
@atanisoft I recently purchased a 3.3V Hi-link power supply which is capable of providing 1.5A output. But even using this, the output still shows a brownout error. I think I will have to use another breakout board to test the esp32.

techievex, 2019-06-14:
ok, so I have tested it with new breakout board. The brownout message is gone but instead guru Meditation error is shown:

Error Message:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4011c121  PS      : 0x00060034  A0      : 0x40081778  A1      : 0x3ffbe0a0  
A2      : 0x3ffaff7c  A3      : 0x00000080  A4      : 0x800892a6  A5      : 0x4008b354  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8011c0f8  A9      : 0x3ffbe070  
A10     : 0x00000001  A11     : 0x3ffb5724  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4011c121

Backtrace: 0x4011c121:0x3ffbe0a0 0x40081775:0x3ffbe1c0 0x4000bfed:0x3ffb52b0 0x40089e65:0x3ffb52c0 0x4008a404:0x3ffb52e0 0x400828ed:0x3ffb5300 0x400822e6:0x3ffb5320 0x400dd839:0x3ffb5390 0x400dc2c1:0x3ffb53b0 0x400dc901:0x3ffb53d0 0x400db3ad:0x3ffb5450 0x400db45c:0x3ffb54d0 0x400da526:0x3ffb5530 0x400daa59:0x3ffb5570 0x4010e7a5:0x3ffb5590 0x4010e953:0x3ffb55e0 0x4010c35c:0x3ffb5600 0x4010d45a:0x3ffb5630 0x4008f758:0x3ffb5650 0x400885dd:0x3ffb5690

Core 1 register dump:
PC      : 0x40139ce6  PS      : 0x00060c34  A0      : 0x800d81ca  A1      : 0x3ffbc570  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060620  A7      : 0x00000000  A8      : 0x00060023  A9      : 0x3ffbc668  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x800887f0  A13     : 0x3ffbc470  
A14     : 0x00000000  A15     : 0x3ffbc260  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40139ce6:0x3ffbc570 0x400d81c7:0x3ffbc590 0x4008a4dd:0x3ffbc5b0 0x400885dd:0x3ffbc5d0

===============================================================
And here is the decoded file of the error which I got:
PC: 0x4011c121: rtc_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/rtc_module.c line 1972
EXCVADDR: 0x00000000

Decoding stack results

0x4011c121: rtc_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/rtc_module.c line 1972
0x40089e65: vTaskExitCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 4304
0x4008a404: vTaskPrioritySet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 1770
0x400828ed: spi_flash_disable_interrupts_caches_and_other_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 116
0x400822e6: spi_flash_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 155
0x400dd839: nvs::nvs_flash_read(unsigned int, void*, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_ops.cpp line 70
0x400dc2c1: nvs::Page::readEntry(unsigned int, nvs::Item&) const at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp line 717
0x400dc901: nvs::Page::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int, unsigned char, nvs::VerOffset) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp line 288
0x400db3ad: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp line 430
0x400db45c: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp line 455
0x400da526: nvs_get_str_or_blob(nvs_handle, nvs::ItemType, char const*, void*, size_t*) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp line 515
0x400daa59: nvs_get_blob(nvs_handle, char const*, void*, size_t*) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp line 525
0x400885dd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

PC: 0x40139ce6: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c line 487
EXCVADDR: 0x00000000

Decoding stack results
0x40139ce6: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c line 487
0x400d81c7: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 63
0x4008a4dd: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3412
0x400885dd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

lbernstone, 2019-06-14:
Looks like your nvs or flash might be damaged.  Try wiping the device by running `python <esp32>/tools/esptool.py erase_flash` from a command prompt.

techievex, 2019-07-03:
Hello everyone, 
Thanks you so much for your suggestions. Really appreciate it!!
The problem was not with PSU, but the issue was how it was given to the ESP. So I made a new PCB and placed all the component on the board including the power source. Now the ESP is working perfectly without any brownout error.
Thanks again! I hope this issue can be closed.

closed 2019-07-04 by deventank7

Murugesh-Hobbyist, 2024-03-14:
Hi Im faced the same issue and below are the decoded details.
can you please help me to fix?

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420021cf  PS      : 0x00060830  A0      : 0x8200207c  A1      : 0x3fcebdf0  
A2      : 0x003c0301  A3      : 0x3fc95392  A4      : 0x00000003  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x820024a4  A9      : 0x3fcebdd0  
A10     : 0x3fc91e7c  A11     : 0x3c03302c  A12     : 0x00000002  A13     : 0xffffffff  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x003c030d  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x420021cc:0x3fcebdf0 0x42002079:0x3fcebe10 0x420020ab:0x3fcebe30 0x4200215a:0x3fcebe50 0x42001b9d:0x3fcebe70 0x42001c9e:0x3fcebe90 0x42003f2d:0x3fcebeb0

ELF file SHA256: a29005a4d446cba5
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403770f0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

Converted data:

PC: 0x420021cf: Adafruit_I2CDevice::begin(bool) at C:\Users\Vasanth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire\src/Wire.h line 92
EXCVADDR: 0x003c030d

Decoding stack results
0x420021cc: Adafruit_I2CDevice::begin(bool) at C:\Users\Vasanth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire\src/Wire.h line 92
0x42002079: Adafruit_ADS1X15::writeRegister(unsigned char, unsigned short) at C:\Users\Vasanth\Documents\Arduino\libraries\Adafruit_ADS1X15\Adafruit_ADS1X15.cpp line 383
0x420020ab: Adafruit_ADS1X15::startADCReading(unsigned short, bool) at C:\Users\Vasanth\Documents\Arduino\libraries\Adafruit_ADS1X15\Adafruit_ADS1X15.cpp line 344
0x42001b9d: readADC() at C:\Users\Vasanth\Desktop\11_Un_Activated_Dual_Core_7inch/11_Un_Activated_Dual_Core_7inch.ino line 178
0x42001c9e: readADC() at C:\Users\Vasanth\Desktop\11_Un_Activated_Dual_Core_7inch/11_Un_Activated_Dual_Core_7inch.ino line 209
0x42003f2d: uartFlushTxOnly at C:\Users\Vasanth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\cores\esp32\esp32-hal-uart.c line 455

chegewara, 2024-03-14:
Its not the same.
You have issue with your code related to `readADC()` probably.

Murugesh-Hobbyist, 2024-03-14:
ok please let me know whether i need to create a new issue or can you help here itself....
Because I couldn't figure this thing out.

zzff-sys, 2024-05-15:
hi  @chegewara 
I run into the same problem when I use watchdog interrupts. When the watchdog times out, enter the interrupt function esp_task_wdt_isr_user_handler() to perform the led flashing, but the phenomenon is that the led light and the system immediately restart. The code and error reports are as follows。
`void esp_task_wdt_isr_user_handler(void)
{
       while (1){
        esp_rom_printf("entry isr\n");
        blink_led();
        esp_rom_printf("exit isr\n");
       }
        esp_rom_printf("exit isr 1\n");

}`
> I (284) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (294) main_task: Returned from app_main()
E (5294) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5294) task_wdt:  - task (CPU 0)
E (5294) task_wdt: Tasks currently running:
E (5294) task_wdt: CPU 0: IDLE
entry esp_task_wdt_isr_user_handler
entry isr
entry led_strip_refresh
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
MEPC    : 0x40806b78  RA      : 0x40807756  SP      : 0x4080cab0  GP      : 0x4080b750
0x40806b78: vListInsert at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/list.c:183 (discriminator 1)
0x40807756: vTaskPlaceOnEventList at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/tasks.c:3700

TP      : 0x40807148  T0      : 0x00000018  T1      : 0x00000019  T2      : 0x00098003
0x40807148: vTaskDelete at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/tasks.c:1310 (discriminator 1)

S0/FP   : 0xffffffff  S1      : 0x00000000  A0      : 0x40810f00  A1      : 0x40810ab0
A2      : 0x00000000  A3      : 0x00000019  A4      : 0x00000019  A5      : 0x40810ab0
A6      : 0x00000001  A7      : 0x00000019  S2      : 0x00000001  S3      : 0x4080cb0c
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x60006400  T4      : 0x00000001  T5      : 0x00000000  T6      : 0x40811058
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000018  MTVAL   : 0x000043d8
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4080cab0: 0x40810d10 0x00000000 0x40810edc 0x40806168 0x00000000 0x00000000 0x00000000 0xffffffff
0x40806168: xQueueReceive at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/queue.c:1655

4080cad0: 0x00000000 0x40810d10 0x00000000 0x000001f5 0x00000000 0x00000000 0x00000000 0x00000001
4080caf0: 0xffffffff 0x00000000 0x40810d10 0x4200b558 0x00000000 0x00000000 0x00000000 0x00000000
0x4200b558: rmt_tx_wait_all_done at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/driver/rmt/rmt_tx.c:554 (discriminator 1)

4080cb10: 0x000014ae 0x00000000 0x00000000 0x00000001 0x000014ae 0x40810ce4 0x00000000 0x420074d6
0x420074d6: led_strip_rmt_refresh at D:/code/5.2.1/blink/managed_components/espressif__led_strip/src/led_strip_rmt_dev.c:75

4080cb30: 0x00000001 0x00000000 0x00000000 0x00000000 0x00000010 0x00000000 0x40810ce4 0x42007242
0x42007242: led_strip_refresh at D:/code/5.2.1/blink/managed_components/espressif__led_strip/src/led_strip_api.c:83

4080cb50: 0x4201c9a8 0x0000000a 0x4080d000 0x420070c8 0x00000001 0x00000000 0x00000001 0x420071b6
0x420070c8: blink_led at D:/code/5.2.1/blink/main/blink_example_main.c:51
0x420071b6: esp_task_wdt_isr_user_handler at D:/code/5.2.1/blink/main/blink_example_main.c:120 (discriminator 1)

4080cb70: 0x00000000 0x4200719e 0xa0000000 0x4200d24e 0x00001881 0x4080cbb0 0x00000001 0x00000001
0x4200719e: esp_task_wdt_isr_user_handler at D:/code/5.2.1/blink/main/blink_example_main.c:115
0x4200d24e: task_wdt_isr at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system/task_wdt/task_wdt.c:512

4080cb90: 0x00001881 0x8000000c 0x00000008 0x4080a096 0x00000000 0x00000000 0x00000000 0x408001e6
0x4080a096: _global_interrupt_handler at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/riscv/interrupt.c:107
0x408001e6: _interrupt_handler at ??:?

4080cbb0: 0x6000a000 0x40804d72 0x40804d7e 0x00000000 0x4080cbbc 0x00000000 0x00000000 0x00000000
0x40804d72: systimer_ticks_to_us at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support/port/esp32c6/systimer.c:16
[... 19 lines cut ...]

4080ccf0: 0x4200d1ae 0x00000000 0x40800d40 0x40810fe8 0x00000000 0x00000000 0x00000000 0x00000000
0x4200d1ae: task_wdt_isr at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system/task_wdt/task_wdt.c:482
0x40800d40: shared_intr_isr at D:/Espressif521/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support/intr_alloc.c:435

4080cd10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080cd30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080cd50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080cd70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080cd90: 0x00000001 0x00000009 0x00000000 0x00400000 0x00400000 0x00028000 0x003d8000 0x0000001b
4080cdb0: 0x00000001 0x00000000 0x4080cdb4 0x40810f48 0x40810fa0 0x00000001 0x00000000 0x00000000
[... cut ...]
4080ce30: 0x00000000 0x00000000 0x00000000 0x4080f268 0x00000000 0x00000000 0x00000000 0x00000000
4080ce50: 0x12121212 0x00000000 0x00000000 0x4080f2c0 0x00000000 0x00000000 0x4080df44 0x00000000
4080ce70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080ce90: 0x00000000 0x00000004 0x00000000 0x00000000 0x00000000 0x00000000 0x4080ce88 0x42010db2
0x42010db2: __sread at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:37

ELF file SHA256: a54aa1a10

chegewara, 2024-05-15:
```
esp_task_wdt_isr_user_handler -> blink_led -> led_strip_refresh -> rmt_tx_wait_all_done -> crash
```
that should explain why it crash (wait in function name)

frrhhhann, 2024-07-29:
hi all, can anyone help me with this issue

13:54:52.301 -> 
13:54:52.301 -> ELF file SHA256: 8bfad1fe0ec960ad
13:54:52.301 -> 
13:54:52.301 -> Rebooting...
13:54:52.301 -> ESP-ROM:esp32c3-api1-20210207
13:54:52.301 -> Build:Feb  7 2021
13:54:52.301 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
13:54:52.301 -> Saved PC:0x40381c52
13:54:52.301 -> SPIWP:0xee
13:54:52.301 -> mode:DIO, clock div:1
13:54:52.301 -> load:0x3fcd5810,len:0x438
13:54:52.301 -> load:0x403cc710,len:0x90c
13:54:52.301 -> load:0x403ce710,len:0x2624
13:54:52.301 -> entry 0x403cc710
13:54:55.577 -> Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
13:54:55.577 -> 
13:54:55.577 -> Core  0 register dump:
13:54:55.577 -> MEPC    : 0x42010698  RA      : 0x42010620  SP      : 0x3fc99d60  GP      : 0x3fc8c000  
13:54:55.577 -> TP      : 0x3fc85dfc  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x0000000e  
13:54:55.577 -> S0/FP   : 0x00000000  S1      : 0x3fc8de88  A0      : 0x3fc8de88  A1      : 0x00000020  
13:54:55.577 -> A2      : 0x00000000  A3      : 0x646e7553  A4      : 0x646e7553  A5      : 0x00000000  
13:54:55.577 -> A6      : 0x00000000  A7      : 0x00000003  S2      : 0x00000040  S3      : 0x00000020  
13:54:55.577 -> S4      : 0x00000000  S5      : 0x646e7553  S6      : 0x00000040  S7      : 0x00000001  
13:54:55.577 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000008  S11     : 0x00000000  
13:54:55.577 -> T3      : 0x00000078  T4      : 0x00000001  T5      : 0x3fc99c70  T6      : 0x00000001  
13:54:55.577 -> MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x646e7553  
13:54:55.577 -> MHARTID : 0x00000000  
13:54:55.577 -> 
13:54:55.577 -> Stack memory:
13:54:55.577 -> 3fc99d60: 0x00000005 0x00000001 0x3fc8bc60 0x4201ba08 0x3fc9293a 0x00000017 0x3fc91000 0x3fc929f1
13:54:55.577 -> 3fc99d80: 0x00000000 0x00000000 0x00000000 0x3fc93000 0x00010001 0x3fc8b92c 0x3fc9293a 0x3c061000
13:54:55.577 -> 3fc99da0: 0x00000020 0x00000017 0x3fc92a30 0x4200e5a0 0x00000000 0x00000000 0x3fc9a378 0x3fc93000
13:54:55.577 -> 3fc99dc0: 0x00000000 0x00000000 0x0008002a 0x44414f4c 0x00474e49 0x00000000 0x87c8bc40 0x6c87153b
13:54:55.577 -> 3fc99de0: 0x00000000 0x00000000 0x3fc91000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99e00: 0x00000000 0x00000000 0x3fc91000 0x3fc9293a 0x08000000 0x3fc8e000 0x3fc92a30 0x4200e776
13:54:55.577 -> 3fc99e20: 0x00000000 0x00000000 0x00000000 0x00005000 0x08000000 0x0001c000 0x3fc91000 0x4200e828
13:54:55.577 -> 3fc99e40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc93000 0x42019000 0x4201c21e
13:54:55.577 -> 3fc99e60: 0x00000000 0x00000000 0x00000000 0x4038777e 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99e80: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
13:54:55.577 -> 3fc99ea0: 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x3fc99cb0 0x00000b0e 0x3fc91e9c 0x3fc91e9c
13:54:55.577 -> 3fc99ec0: 0x3fc99eb0 0x3fc91e94 0x00000018 0x3fc9a308 0x3fc9a308 0x3fc99eb0 0x00000000 0x00000001
13:54:55.577 -> 3fc99ee0: 0x3fc97ea0 0x706f6f6c 0x6b736154 0x5a490d00 0x00fd7026 0x00000000 0x3fc99e90 0x00000001
13:54:55.577 -> 3fc99f00: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc936b4 0x3fc9371c 0x3fc93784 0x00000000
13:54:55.577 -> 3fc99f20: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4203eb72 0x00000000 0x00000000
13:54:55.577 -> 3fc99f40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99f60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99f80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99fa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99fc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc99fe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc9a000: 0x03000000 0xbaad5678 0x00000060 0xabba1234 0x00000054 0x00000000 0x3fc9a014 0x00000000
13:54:55.577 -> 3fc9a020: 0x00000000 0x00000000 0x3fc9a02c 0xffffffff 0x3fc9a02c 0x3fc9a02c 0x00000000 0x3fc9a040
13:54:55.577 -> 3fc9a040: 0xffffffff 0x3fc9a040 0x3fc9a040 0x00000001 0x00000001 0x00000000 0xff00ffff 0x00000000
13:54:55.577 -> 3fc9a060: 0xb33fffff 0x00000000 0xbaad5678 0x0000008c 0xabba1234 0x00000080 0x29ecf3bf 0xc0b7c92f
13:54:55.577 -> 3fc9a080: 0x4b0968c1 0x22a57444 0x80157514 0x928df6ac 0x9d03fe6f 0x09dccb1f 0xba3f36ca 0x6e43b986
13:54:55.577 -> 3fc9a0a0: 0x50220bc3 0x6891f716 0x2c421a88 0x3093b4b4 0xea0b6d7e 0xecf877ea 0x9e8aa2b8 0xb6d75e1b
13:54:55.577 -> 3fc9a0c0: 0x020d2844 0x5ae74b84 0x49626c7c 0xd2422ed5 0xc1e96ab0 0x8fe857e9 0xb497395f 0xb2df51eb
13:54:55.577 -> 3fc9a0e0: 0x4582a012 0x24018682 0x00000290 0x6135ca9b 0x73e45eec 0xe7f62206 0xbaad5678 0x0000008c
13:54:55.577 -> 3fc9a100: 0xabba1234 0x00000080 0x00000040 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc9a120: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
13:54:55.577 -> 3fc9a140: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x07000000 0x04040404

atanisoft, 2024-07-29:
@frrhhhann please open a new issue with a test case that reproduces this specific issue. Most likely it will be something in your code that is causing this but it is hard to tell for certain from what you have posted. Please also be sure to decode the backtrace and enable core debug logging at the most verbose setting when you file your new issue.

awong1900, 2024-07-30:
When using wifi or the espnow protocol, the original motherboard will reboot. I encountered the same problem before and finally reported it to the hardware vendor. They confirmed that there is some peak current when the wifi is turned on, and the inferior motherboards on the market cannot provide enough power. I ended up updating with a larger peak power supply to fix the problem.