espressif/arduino-esp32 #1453: 30% growth of .elf file after update espressif32 platform v0.12.0 -> 1.0.0

cyberman54, 2018-05-31:
### Hardware:
Board:							ESP32generic and Heltec Lora32
Core Installation/update date:                30.5.2018	
IDE name:						PlatformIO with Espressif32 1.0.0

### Description:

Yesterday the espressif32 platform was upgraded from v0.12.0 to v1.0.0 in PlatformIO.

I have code that compiled with previous v0.12.0 to 1,060,542 bytes firmware.elf
After ugrade to v1.0.0 the same code compiles a 1,390,381 bytes firmware.elf - which is too large to fit in the 1,25MB of an ESP32.

This ~30% growth of the binary. Too much to crunch this under 1,25MB by refactoring the code.

What can i do to shrink the firmware.elf?

### Sketch:
see [repo here](https://github.com/cyberman54/ESP32-Paxcounter)

lbernstone, 2018-05-31:
What does this have to do with arduino-esp32?  https://github.com/platformio/platform-espressif32/issues

cyberman54, 2018-05-31:
To be honest: i don't know. I posted this in PlatformIO Community, and there i was told to raise an issue here.

negativekelvin, 2018-06-01:
https://github.com/espressif/arduino-esp32/issues/1367

closed 2018-11-20 by cyberman54