espressif/arduino-esp32 #1456: When wifi gets disconnected after some time other task gets slow

MagicOAk, 2018-06-02:
When wifi and mqtt connection is proper that time both the task are working but when I unplug the LAN cable of router it still works but after 2min my 1st task i.e LED blinking gets slow.

#include <WiFi.h>
#include <PubSubClient.h>

const char* mqttServer = "mymqttserver";
const int mqttPort = mymqttport;
WiFiClient wifiClient;
PubSubClient client(wifiClient);

#define led1 12
#define led2 13
const char* ssid     = "";
const char* password = "";

void setup() {
pinMode(led1,OUTPUT);
pinMode(led2,OUTPUT);
Serial.begin(115200);

 WiFi.begin(ssid, password);
 while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");
    
     xTaskCreatePinnedToCore(task1,"task1",1000,NULL,1,NULL,1); 
     xTaskCreatePinnedToCore(task2,"task2",1000,NULL,1,NULL,0); 
}

void loop()
{
   reconnect();
}

void reconnect(){
   client.setServer(mqttServer, mqttPort);
    while (!client.connected()) {
    Serial.println("Connecting to MQTT...");
     Serial.println("connected");  

    } else {
    
      Serial.print("failed with state ");
      Serial.print(client.state());
    }
  }
  
  void task1(void *pvParameters)
{
   while (1) {

   digitalWrite(led1,HIGH);
   digitalWrite(led2,HIGH);
   client.publish("abcd", "led1");
   delay(1000);
   digitalWrite(led2,LOW);
   digitalWrite(led1,LOW);
   client.publish("abcd", "led2");
   delay(1000);
  }
}

void task2(void *pvParameters)
{
   while (1) {
    if(WiFi.status() != WL_CONNECTED)
      { 
       WiFi.begin(ssid, password);
         while (WiFi.status() != WL_CONNECTED) {
          delay(500);
          Serial.println("Connecting to WiFi..");
          }
   }
    client.publish("abcd", "wifi connected");
    delay(5000);
   }
    
}

stickbreaker, 2018-06-02:
@magicoak I don't think mqtt is thread safe.

You are using `client()` in 3 different threads.

Chuck.

MagicOAk, 2018-06-04:
Thanks, it got solved...

closed 2018-06-04 by MagicOAk