espressif/arduino-esp32 #1457: Whenever I put MQTT logic in a task, it restarts. If I put it in a void loop(), it works fine

MagicIoT, 2018-06-02:
In this, I used FreeRTOS for different task in different core(s). But whenever I put MQTT logic in any of the task, it continuously reboots the board. The error is mentioned after the code

When the MQTT code is placed in the void loop(), it works perfectly. Please guide me.

#if CONFIG_FREERTOS_UNICORE
#endif
#include <DNSServer.h>
#if defined(ESP8266)
#include <ESP8266WebServer.h>
#else
#include <WiFi.h>
#endif
#include <WiFiManager.h>  
#include <PubSubClient.h>

const char* ssid = "Cohive-Meeting Room";
const char* password =  "cohive@dny";
const char* mqttServer = "iot.eclipse.org";
const int mqttPort = 1883;
const char* mqttUser = "YourMqttUser";
const char* mqttPassword = "YourMqttUserPassword";

WiFiClient wifiClient;
PubSubClient client(wifiClient);

void setup() {

  Serial.begin(115200);
  WiFi.begin(ssid, password);
   while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");
 xTaskCreatePinnedToCore(task1,"task1",1000,NULL,1,NULL,1); 
}
void loop() 
{
   //nothing
   
   }
   
void task1(void *pvParameters)
{
   while (1) 
   {
     client.setServer(mqttServer, mqttPort);
  //client.setCallback(callback);

  while (!client.connected()) {
    Serial.println("Connecting to MQTT...");

    if (client.connect("ESP8266Client" )) {

      Serial.println("connected");  

    } else {

      Serial.print("failed with state ");
      Serial.print(client.state());
      delay(2000);

    }
  }

  client.publish("abcd", "Hello from ESP8266");
  //client.subscribe("esp/test");
   }
}

ERROR --------------------------------------------------------------------------------------------------------
Connecting to WiFi..
Connecting to WiFi..
Connected to the WiFi network
Connecting to MQTT...
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (task1) 
Core 1 register dump:
PC      : 0x40081659  PS      : 0x00060336  A0      : 0x00060330  A1      : 0x3ffd98f0  
A2      : 0x00000000  A3      : 0x00060323  A4      : 0x00060320  A5      : 0x3ffd258c  
A6      : 0x3ffd9a50  A7      : 0x00000000  A8      : 0x80082a90  A9      : 0x3ffd9980  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060123  A15     : 0x00000004  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40081659:0x3ffd98f0 0x0006032d:0x3ffd99c0 0x4008a991:0x3ffd99e0 0x400f8f62:0x3ffd9a20 0x400e9b28:0x3ffd9a50 0x400f93fd:0x3ffd9a80 0x400f968e:0x3ffd9aa0 0x400e8a4e:0x3ffd9ae0 0x400e972d:0x3ffd9b20 0x400d195a:0x3ffd9b40 0x400d15db:0x3ffd9b80 0x400d22ce:0x3ffd9bb0 0x400d244a:0x3ffd9bf0 0x400d1445:0x3ffd9c20

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
Connecting to WiFi..
Connecting to WiFi..

copercini, 2018-06-02:
it seems to be 2 problems:

1- you should put some delay inside the task loops, otherwise the system tasks will not have a chance to run and this will cause problems, your `loop()` and `while(1)` seems to run without any delay in some cases

2- out os stack on task1:
` xTaskCreatePinnedToCore(task1,"task1",1000,NULL,1,NULL,1);`
1000 seems too low for your code

MagicIoT, 2018-06-04:
I made the above changes, but still not working.

MagicIoT, 2018-06-04:
.

closed 2018-06-04 by MagicIoT