espressif/arduino-esp32 #1458: How to pair with ESP32 Bluetooth device using passkey/PIN from smartphone?

avinashs24, 2018-06-02:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:			01/JUN/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I am using the SerialToSerialBT example. I am able to scan and discover the Bluetooth device from my smartphone. But the pairing is without any security/passcode. It simply pairs immediately. How to make the pairing using PIN/Passkey entry?

### Sketch:
```cpp

#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  delay(5000);
  Serial.println("Starting..");
  SerialBT.begin("MyESP32"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
  
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Messages:
```
Starting..
The device started, now you can pair it with bluetooth!

```

copercini, 2018-06-02:
It depends of some IDF/Bluetooth core functions that aren't available yet... https://github.com/espressif/esp-idf/issues/1541

once it is released, it will be implemented here as well

maddmaxamofo, 2018-08-21:
I am also working on a Bluetooth device that needs this feature.  Is there an estimate on its release date?

nayan27, 2018-09-05:
@copercini feature was implemented in commit https://github.com/espressif/esp-idf/commit/8571170abe3b2b7e5344e9603dadf2e6f74916e9

Any idea on when it could be available for esp-idf ?

Thank you

pir0c0pter0, 2018-10-22:
would be great to have this feature.

mpbejo, 2018-11-16:
Hi,
Any idea on when it could be available for Arduino ?
Thank You

chegewara, 2018-11-16:
I know its not the best solution, but you can try to use esp-idf functions in your app. Just saying.

zdclaudy, 2019-03-28:
Anything more on this? It's a feature i am needing badly.

vcmorini, 2019-05-23:
hello, anyway to pair ble programming esp32 with arduino ide? thx a lot!

chegewara, 2019-05-23:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/StaticPIN/StaticPIN.ino

HarpreetHeera, 2019-12-18:
No help. 
I am using Lolin D32 with Arduino IDE. I have tried all the ways many examples, still serial terminal app is connecting to esp32 without facing any passkey authentication in bluetooth classic mode.

Looking forward for support.

Thanks

chegewara, 2019-12-19:
There is no option to build classic bluetooth with static key pairing (or any key pairing). It is because it requires to change settings in menuconfig. Classic bluetooth can be configured to use SSP (secure single pairing) or legacy pairing which is controlled in menuconfig.
Only option is to use arduino as component.

HarpreetHeera, 2019-12-19:
Hi everyone,
Thanks to all.
I have successfully implemented pass key aka pin code authentication functionality in my lolin 32 using Arduino as IDE in classic Bluetooth mode for sending data over serial Bluetooth profile.
What I did is as following.
1. I uninstall latest version of of espressif esp32 Arduino 1.0.4 release.
2. Install espressif/ Arduino ver. 1.0.1 from board manager.
3. Select board.
4. Download disable_ssp.zip from https://github.com/espressif/arduino-esp32/issues/2320#issuecomment-477288545
5. Follow the readme instruction available in above zip file.
6. Upload the compiled program.

Open Bluetooth setting in my phone , unpair esp32 device from my old paired list. Rescan for new device then click on the my esp32  device ... Here m the miracle happened .... THE PIN CODE AUTHENTICATION APPEARED ....   to pair my Bluetooth esp32 device.... 

Thanks for all brilliant people who saved my hard work done by providing this solution.

Regards
Harpreet Singh

chegewara, 2019-12-19:
Thanks for sharing @HarpreetHeera 
As you said, you are using arduino-esp32 v1.0.1, which is very old version and probably at the time it was compiled option i mentioned earlier was not implemented in esp-idf, but glad you achieved what you are looking for.

HarpreetHeera, 2019-12-19:
> Thanks for sharing @HarpreetHeera
> As you said, you are using arduino-esp32 v1.0.1, which is very old version and probably at the time it was compiled option i mentioned earlier was not implemented in esp-idf, but glad you achieved what you are looking for.

Thanks Chegewara ...
 for your support and valuable response.
Have a great time.

choichangjun, 2020-02-06:
Hi guys. I found some code and I uploaded my repository.
come and try my code. it might be worked !
It should Arduino IDF and ESP-IDF installed

https://github.com/choichangjun/ESP32_arduino

Flashbulbtech, 2020-02-09:
> Hi guys. I found some code and I uploaded my repository.
> come and try my code. it might be worked !
> It should be Arduino IDF and ESP-IDF installed
> No need to downgrade version
> https://github.com/choichangjun/ESP32_arduino

Its wonderfully working. Thanks

chegewara, 2020-02-09:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/StaticPIN/StaticPIN.ino

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLESecurity.cpp#L67

Its here over a year, and its working, just for some reasons is not updated in arduino-essp32.

BTW this issue was about PIN in classic BT, not BLE

HarpreetHeera, 2020-02-09:
Thanks Chegewara, 
Now we have working pin key authentication in latest version of Arduino-Esp32.

Thanks for your efforts. Great to know. Now we are  able to do more stuff on latest release .

Thanks again.

HorstBaerbel, 2020-02-20:
So... I've Bluetooth classic serial on the ESP32 and pairing does not work for me with Ubuntu 19.10. The ESP32 always connects and directly disconnects again. Is this the same problem? What can I do about it?

Vish1998, 2020-03-11:
> Thanks Chegewara,
> Now we have working pin key authentication in latest version of Arduino-Esp32.
> Thanks for your efforts. Great to know. Now we are  able to do more stuff on latest release .
> Thanks again.

sir, can you please provide me the code to set password for esp-32 classic Bluetooth.

HarpreetHeera, 2020-03-11:
> > Thanks Chegewara,
> > Now we have working pin key authentication in latest version of Arduino-Esp32.
> > Thanks for your efforts. Great to know. Now we are  able to do more stuff on latest release .
> > Thanks again.
> 
> sir, can you please provide me the code to set password for esp-32 classic Bluetooth.

What specific problem are you facing? Need little explanation.

Vish1998, 2020-03-11:
> What specific problem are you facing? Need little explanation.

I am unable to set a paring password for esp32 board. The board is automatically getting connected to mobile phone without asking for any passkey.

Vish1998, 2020-03-13:
I am in need of help regarding setting passkey for ESP-32 classic Bluetooth please somebody help me out.

chegewara, 2020-03-13:
> I am in need of help regarding setting passkey for ESP-32 classic Bluetooth please somebody help me out.

You need to use arduino as component and to change some settings in bluetooth menuconfig or rebuild  bluetooth library in esp-idf with new settings and replace the one in arduino.

Luis-Ferreira, 2020-03-17:
Hi,I´m also needing to set a passkey to classic Bluetooth .Use arduino as a component?How can we do that?I´m learning ,please help!Thanks.

chegewara, 2020-03-17:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

Luis-Ferreira, 2020-03-17:
> https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

Thanks for the answer.But is it possible to make it easier?How can a simple thing like this give so much work.I don´t know how to do it .Can you help?

Vish1998, 2020-03-17:
> > https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
> 
> Thanks for the answer.But is it possible to make it easier?How can a simple thing like this give so much work.I don´t know how to do it .Can you help?

I am also in need of it very much. Please help me out.

HarpreetHeera, 2020-03-17:
If you are still not able to figure out with latest update.
Then try this.

 https://github.com/espressif/arduino-esp32/issues/1458#issuecomment-567611934

Luis-Ferreira, 2020-03-18:
> If you are still not able to figure out with latest update.
> Then try this.
> 
> [#1458 (comment)](https://github.com/espressif/arduino-esp32/issues/1458#issuecomment-567611934)

Thanks for reply ,but I´m designing a commercial product with possible future updates so I can´t depend on older releases of arduino.I want something that I can work in the future .

Vish1998, 2020-03-18:
> Thanks for reply ,but I´m designing a commercial product with possible future updates so I can´t depend on older releases of arduino.I want something that I can work in the future .

Yes sir, I am also doing the similar thing please help us out in easiest way.
Thank you

OlivierOld, 2020-04-01:
I'm new to ESP32. I used to work with Arduino Bluno Mega, which is a Mega with a BT BLE built in.
I succeed to exchange over BT with Androïd App spécialy designed, but never to go trought the Androïd BT standard pairing process to exchange with existing Apps. I understand this standard pairing can be done with ESP32 (please confirm my understanding). And by the way, Flashbulbtech, as you are very knowledgeable, do you know if a solution exists for the Bluno board? Thank you in advance.

Sule-2002, 2020-05-28:
Hello I am new on ESP32,
I am working with nrf52832 bluetooth modules . I prepared static passkey "123456" on nrf modules I can connect on nrf connect app after I write static passkey .Now I am trying to connect with ESP32 DEV Module .Is there any example Arduino code for ESP32 for pairing my nrf module with passkey? Can you advise me any example?

chegewara, 2020-05-28:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_client/BLE_client_passkey/BLE_client_passkey.ino#L21

Sule-2002, 2020-05-31:
hello @chegewara ,
 I can connect on nrf connect app after I write static passkey with my nrf module but ESP32 looking can not connect with passkey .   I tried example that you advised me above, I write nrf52832 service uuid , my nrf ble password and I build successfully after I opened the serial port ,I saw this messages
I am new on ESP32 Coding can you tell me what should I do for connecting with passkey nrf52832?

If I remove the password setting part in the nrf code and then try to connect esp32, I can see that the connection is established immediately. not connecting when I enable encryption in nrf code 

BLE Advertised Device found: Name: NoQey, Address: cd:59:6d:14:44:fa, serviceUUID: f3641400-00b0-4240-ba50-05ca45bf8abc
Found our device!  address: Forming a connection to cd:59:6d:14:44:fa
 - Created client
 - Connected to server
Failed to find our service UUID: f3641400-00b0-4240-ba50-05ca45bf8abc
We have failed to connect to the server; there is nothin more we will do.

chegewara, 2020-05-31:
You have to add this to your code too to tell nRF you are able to input numbers:
```
BLESecurity *pSecurity = new BLESecurity();
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY);
pSecurity->setCapability(ESP_IO_CAP_KBDISP);
```

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/security/SampleServer_authentication_passkey.cpp#L101

Sule-2002, 2020-05-31:
BLESecurity *pSecurity = new BLESecurity();
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY);
pSecurity->setCapability(ESP_IO_CAP_KBDISP);

@chegewara 
this code part that you said me already exist in my arduino  code you advised me above. Altough this part exist in code , I can not connect to esp32.    
I couldn't understand what the problem was

parklemon99, 2020-09-28:
> ### Hardware:
> Board: ESP-WROOM-32
> Core Installation/update date: 01/JUN/2018
> IDE name: Arduino IDE
> Flash Frequency: 80Mhz
> Upload Speed: 921600
> 
> ### Description:
> I am using the SerialToSerialBT example. I am able to scan and discover the Bluetooth device from my smartphone. But the pairing is without any security/passcode. It simply pairs immediately. How to make the pairing using PIN/Passkey entry?
> 
> ### Sketch:
> ```c++
> #include "BluetoothSerial.h"
> 
> #if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
> #error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
> #endif
> 
> BluetoothSerial SerialBT;
> 
> void setup() {
>   Serial.begin(115200);
>   delay(5000);
>   Serial.println("Starting..");
>   SerialBT.begin("MyESP32"); //Bluetooth device name
>   Serial.println("The device started, now you can pair it with bluetooth!");
>   
> }
> 
> void loop() {
>   if (Serial.available()) {
>     SerialBT.write(Serial.read());
>   }
>   if (SerialBT.available()) {
>     Serial.write(SerialBT.read());
>   }
>   delay(20);
> }
> ```
> 
> ### Debug Messages:
> ```
> Starting..
> The device started, now you can pair it with bluetooth!
> ```

Hi, is there any arduino example on how to pair bluetooth with PIN?

hulktech, 2020-12-14:
Hello ,chegewara,
can you please give us an example on how to use arduino as component and to change some settings in bluetooth menuconfig or rebuild bluetooth library in esp-idf with new settings and replace the one in arduino? The only think we want is to add PIN to Classic BT without downgrade to 1.0.1

severinbeauvais, 2021-02-15:
This is part of the solution:
```
  //
  // init Bluetooth
  //
  SerialBT.enableSSP();
  SerialBT.setPin("918780");
  if (!SerialBT.begin(BT_LOCAL_NAME)) {
    Serial.println("Bluetooth init failed!");
    while (true) {
      delay(1000); // don't proceed, loop forever
    }
  }
  Serial.println("Bluetooth initialized!");
```

I'm still trying to figure out how to sync the code between device and phone...

severinbeauvais, 2021-02-15:
I found this but it's not quite working either...
https://github.com/choichangjun/ESP32_arduino/blob/master/ESP32_Arduino_paring_Key.ino

lbernstone, 2021-02-15:
https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Secure_Client/NimBLE_Secure_Client.ino

severinbeauvais, 2021-02-15:
> https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Secure_Client/NimBLE_Secure_Client.ino

This is BLE not BL Classic. But thanks! I may use it anyway... :)

ingsoc, 2021-04-15:
Is this ever going to be addressed ?

Not being able to set a pass code makes it not usable for a large number of applications :(

podaen, 2021-04-27:
I want to setup a pin too. Did you tried this for an incomming connection on classic bluetooth over the AT-command.

[Documentation AT-commands](https://docs.espressif.com/projects/esp-at/en/release-v2.1.0.0_esp8266/AT_Command_Set/BT_AT_Commands.html#cmd-BTSPPCONN)

podaen, 2021-04-27:
Sorry, It's not the latest version.

> I want to setup a pin too. Did you tried this for an incomming connection on classic bluetooth over the AT-command.
> 
> [Documentation AT-commands](https://docs.espressif.com/projects/esp-at/en/release-v2.1.0.0_esp8266/AT_Command_Set/BT_AT_Commands.html#cmd-BTSPPCONN)

This is...

[Documentation AT-commands](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_use_ESP_AT_Classic_Bluetooth.html)

podaen, 2021-04-28:
I could not access the AT-command over serial monitor in the Arduino IDE. It could maybe be possible by using the UART pins of the bluetooth. To do that you need to config ESP-IDF with python. 

So I followed the instruction here and it worked.

podaen, 2021-04-28:
And I reflash the firmware with 2.1 too.

ingsoc, 2021-04-28:
@podaen Hi, Could you provide a few more detail on how you managed to achieve this ?

Cheers

podaen, 2021-04-29:
The one I flashed doesn't solve the issue here. Just for you to know. Anyway you can downloaded the "downoad tool" and "bin file" from espressif server, unzip it and direct the tool to the bin files. I flashed it on byte 0x1000, 0x8000 and 0x10000. It could be different from ESP to ESP. Google is your friend.

podaen, 2021-05-02:
I found a way to set a fix pin (for what I have test it now) with the last SDK. 

I did set the SPP (serial port protocol) to true in the menuconfig with python, but it can be already enabled default. I am not sure, after a lot of flashing and reconfiguration. The Arduino precompiled code does support SPP also. 

I test it on multiple devices and it worked but now I found out it doesn't  work with ios like HC05.

[Fix pin python code](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor/main/example_spp_acceptor_demo.c)

podaen, 2021-05-04:
note: In the sdkconfig files the secure simple pairing (SPP) needs to be NOT set. The arduino sdkconfig file that's generated is set for other reasons and can not be edit. So it cannot be done by this libary or any other libary without addapting other sdk files.

VojtechBartoska, 2021-05-06:
@avinashs24 You can take a look on these examples:

- https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor/ESP32_SSP.md
- https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor

podaen, 2021-05-06:
In my eyes secure simple pairing is interesting when you have the capibility too enter  a key in both devices. I could be wrong... I m connecting with an android to esp without screen or keyboard.

closed 2021-05-21 by VojtechBartoska

alfonso18martin, 2021-11-18:
alguien lo pudo hacer????.... el codigo pin

SuGlider, 2022-02-10:
@alfonso18martin 

Try calling ```setPin()``` before ```begin()```.

``` dart
void setup() {
  Serial.begin(115200);
  SerialBT.setPin(pin);    // setPin() shall come before begin()
  SerialBT.begin("ESPBT", true); 
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  
  connected = SerialBT.connect(address);
  
  if(connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while(!SerialBT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
  }
}
```

PilnyTomas, 2023-07-14:
> @alfonso18martin
>
> Try calling `setPin()` before `begin()`.
>
> ```dart
> void setup() {
>   Serial.begin(115200);
>   SerialBT.setPin(pin);    // setPin() shall come before begin()
>   SerialBT.begin("ESPBT", [...]

`setPin` is checking to make sure it was called **after** the begin
https://github.com/espressif/arduino-esp32/blob/855f5261913c74e9f4f99fc0de184c847cc4db59/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L915C1-L915C1

RMU56, 2023-12-21:
> ### Hardware:
> Board: ESP-WROOM-32 Core Installation/update date: 01/JUN/2018 IDE name: Arduino IDE Flash Frequency: 80Mhz Upload Speed: 921600
>
> ### Description:
> I am using the SerialToSerialBT [...]

I have successfully connected using password 123456 and sending continuous data 0 to 9 in loop and can visible in BLE scanner app in R mode. You can only Read data if your connection is successful. 
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "0000FFE0-0000-1000-8000-00805F9B34FB"
#define CHARACTERISTIC_UUID "0000FFE1-0000-1000-8000-00805F9B34FB"

class MySecurity : public BLESecurityCallbacks {
  uint32_t onPassKeyRequest() {
    ESP_LOGI(LOG_TAG, "PassKeyRequest");
    return 123456;
  }

  void onPassKeyNotify(uint32_t pass_key) {
    ESP_LOGI(LOG_TAG, "The passkey Notify number:%d", pass_key);
  }

  bool onConfirmPIN(uint32_t pass_key) {
    ESP_LOGI(LOG_TAG, "The passkey YES/NO number:%d", pass_key);
    vTaskDelay(5000);
    return true;
  }

  bool onSecurityRequest() {
    ESP_LOGI(LOG_TAG, "SecurityRequest");
    return true;
  }

  void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl) {
    ESP_LOGI(LOG_TAG, "Starting BLE work!");
  }
};

class MyCallbacks : public BLECharacteristicCallbacks {
  void onRead(BLECharacteristic *pCharacteristic) {
    if (pCharacteristic->getValue().length() > 0) {
      Serial.println("Read request received!");
      // You can add your custom read handling code here
    }
  }
};

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;

int counter = 0;

void setup() {
  Serial.begin(115200);

  BLEDevice::init("ESP32");

  BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);

  BLEDevice::setSecurityCallbacks(new MySecurity());

  pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
      CHARACTERISTIC_UUID,
      BLECharacteristic::PROPERTY_READ |
          BLECharacteristic::PROPERTY_WRITE);
  pCharacteristic->setValue("Hello World");
  pCharacteristic->setCallbacks(new MyCallbacks());
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  BLESecurity *pSecurity = new BLESecurity();
  uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
  uint32_t passkey = 123456;
  uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
  esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);
  pSecurity->setCapability(ESP_IO_CAP_OUT);
  pSecurity->setKeySize(16);
  esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
  pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
  esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));
  Serial.println("Characteristic defined! Now you can read it on your phone!");
}

void loop() {
  // Increment the counter from 0 to 9
  pCharacteristic->setValue(String(counter).c_str());
  pCharacteristic->notify();

  Serial.print("Sending value: ");
  Serial.println(counter);

  delay(1000); // Adjust the delay based on your application requirements

  // Reset counter when it reaches 9
  if (counter == 9) {
    counter = 0;
  } else {
    counter++;
  }
  
  delay(2000);
}