espressif/arduino-esp32 #1459: BME280 & I2C

victorsanchis, 2018-06-02:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:	ESPWROOM32						
Core Installation/update date:			April2018
IDE name:							Arduino IDE1.8.5 
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hello to anybody who can help me!
I recently got an ESPWROOM32. I've tried it with different analog/digital sensors (DHT22 -with its dedicated "esp" library-, MQ135, DS18B20...) with no excesive problems. Now I'm trying the BME280 through I2C (3.3V / Ground/ SCL / SDA) and it seems impossible to get any data, independently of the program/library used (and also the 0x76 or 0x77). 
I've read about some troubles with ESP32, its I2C, Wire.h library, and some sensors... 
So, to be sure, I've finally scanned the I2C connections with 'I2C scan' program. It returns the following, depending if the BME280 is connected (pins 21/SDA & 22/SCL): "127 I2C devices found" (!) 

Scanning I2C Addresses
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 
20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 
30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 
40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 
50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 
60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 
70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E Scan Completed, 127 I2C Devices found.

...if there is nothing connected to ESP32 I  receive

[esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 70
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 71
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 72
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 73
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 74
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 75
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 76
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 77
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 78
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 79
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 7a
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 7b
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 7c
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 7d
.. [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 7e
.. Scan Completed, 0 I2C Devices found.

Is there any way to find out if there is a problem with this sensor? Even if there is a problem with the I2C pins (not 21-22 but others), I thought they should be indicated with "I2C_Scan" but unfortunately this is not the case... So this sensor is really drivin' me crazy... 

Thank you in advance for your help!
Victor

stickbreaker, 2018-06-02:
@victorsanchis try using the [stickbreaker-i2c](https://github.com/espressif/arduino-esp32/tree/stickbreaker-i2c) branch instead of the [Master](https://github.com/espressif/arduino-esp32/tree/master) branch.

See if that makes a difference for you.

Chuck.

victorsanchis, 2018-06-02:
Thank you very much, Chuck!
I will, and let you know tomorrow.

Victor

victorsanchis, 2018-06-03:
One question comes to my mind **@stickbreaker**  : do you recommend (should I) save the whole original Master folder in another location? Just in case stickbreaker-i2c doesn't solve the problems: I wouldn't like to lose all the sketches I developed before...
Thank you in advance!

stickbreaker, 2018-06-03:
@victorsanchis you can just copy the effected files.  From the [stickbreaker-i2c](https://github.com/espressif/arduino-esp32/tree/stickbreaker-i2c) branch:

cores/esp32/esp32-hal-i2c.h
cores/esp32/esp32-hal-i2c.c
cores/esp32/esp32-hal-log.h
libraries/Wire/src/Wire.h
libraries/Wire/src/Wire.cpp

Just copy them over the same files from [Master](https://github.com/espressif/arduino-esp32)

Chuck.

victorsanchis, 2018-06-03:
Ufff.. what a relief! Thanks, Chuck!

victorsanchis, 2018-06-03:
Done.
Well, at least now it doesn't mention "errors" 'though no I2C device is found (with BME280 connected as I2C).
Based on this info (below), I'm not going to spend more time on it. I'll have to return this sensor...

[W][esp32-hal-i2c.c:1092] i2cCheckLineState(): invalid state sda=1, scl=0

[E][esp32-hal-i2c.c:1108] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init

 Scanning I2C Addresses
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. Scan Completed, 0 I2C Devices found.

Thanks for all the help!

stickbreaker, 2018-06-03:
@victorsanchis If you're satisfied, close this issue.

Chuck.

closed 2018-06-04 by victorsanchis