espressif/arduino-esp32 #1460: Probelm softAp with windows

vidaxdu40, 2018-06-04:
Hello everyone, I have a problem. i have a esp32 sparkfun thing board and i have configurated a softAP with arduino. My problem is that i can connect a station on the AP with an iPhone, macbook but with my PC on windows 10 and a windows tab on windows 8 the connection doesn’t work.
Do you have a solution please ?

timkoers, 2018-06-09:
Could you describe what is going wrong? Is it not connect all, or is there no op.
Windows should connect fine when there is no internet connection on the AP, but you might have to tell Windows to connect first.

lbernstone, 2018-06-09:
https://en.wikiversity.org/wiki/Wireshark/DHCP

vidaxdu40, 2018-06-11:
hello,
I discovered sometime I can connect the pc to esp32.
When the connexion work i have this on wireshark 
[image]

but when the connexion fail i have nothing

lbernstone, 2018-06-11:
If the client never asks for an address, the server can't give it one :smile: .

vidaxdu40, 2018-06-12:
Yes of course ;) but what is strange is that even if I don't have communication shown on wireshark, when the connexion fail i have the SYSTEM_EVENT_AP_START event in the esp32

vidaxdu40, 2018-06-12:
https://drive.google.com/open?id=1EjPk04hRMx9Wp94K9GDWjN0v6LvA0MOB

closed 2019-08-15 by stale[bot]