espressif/arduino-esp32 #1463: problem while reconnecting WiFi and MQTT

MagicIoT, 2018-06-04:
### Hardware:
Board:							ESP32 Dev Module
IDE name:					         Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:				         115200

### Description:
The very first time, ESP32 starts its connection with MQTT and WiFi but as soon as the internet is not present it never gets reconnected automatically.

#include <WiFiManager.h>
#include <PubSubClient.h>
#define led1 12
#define led2 13
const char* ssid = "************";
const char* password = "************";
const char* mqttServer = "************.************.************";
const int mqttPort = ************;

WiFiClient wifiClient;
PubSubClient client(wifiClient);

void setup() 
{
pinMode(led1,OUTPUT);
pinMode(led2,OUTPUT);
Serial.begin(115200);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) 
{
delay(500);
Serial.println("Connecting to WiFi..");
}
    Serial.println("Connected to the WiFi network");
    
    xTaskCreatePinnedToCore(task1,"task1",2000,NULL,1,NULL,1);
    xTaskCreatePinnedToCore(task2,"task2",10000,NULL,1,NULL,0);

}
void loop()
{
  
  delay(1000);
}
void connect_Wifi(){
  while (1) {
    if(WiFi.status() != WL_CONNECTED)
    { 
       WiFi.begin(ssid, password);
         while (WiFi.status() != WL_CONNECTED) {
          delay(500);
          Serial.println("Connecting to WiFi..");
          }
   }
}
}
void connect_MQTT(){
 client.setServer(mqttServer, mqttPort);
  while (!client.connected()) {
      Serial.println("Connecting to MQTT...");
      if (client.connect("ESP8266Client" )) {
        Serial.println("connected");  
      } else {
        Serial.print("failed with state ");
        Serial.print(client.state());
        delay(1000);
    }
  }
}

void task2(void *pvParameters)

{
 
  if(WiFi.status() != WL_CONNECTED){
      connect_Wifi();
   }

  if(!client.connected()){
    connect_MQTT();
  }
   client.publish("abcd", "sensor data");
   Serial.println("Data published on MQTT");
   delay(5000);
}

void task1(void *pvParameters)
{
while (1){
    digitalWrite(led1,HIGH);
   delay(500);
  digitalWrite(led1,LOW);
   delay(500);
   Serial.println("light on off");

 
}
   delay(500);

}

timkoers, 2018-06-09:
Is the loop continuing?

helmut64, 2018-06-16:
My understanding is that that Arduino libraries, the IDF libs and the compiler runtime library is are thread save. Therefore threads must be used with care.

You need to develop the Wifi connect asynchrony in the Arduino main thread, after WiFi is up and running you can initialze the MQTT.

vanThijs, 2018-07-16:
I had a lot of issues with the pubsubclient as well, but setting the wifi mode solved these for me: 
place "WiFi.mode(WIFI_STA);"  in your setup() routine, and you should be good to go.

closed 2019-08-15 by stale[bot]